CertainLach/VivePro2-Linux-Driver #1: Using lensServer

CandyGumdrop, 2022-01-23:
Hi.  First of all, I'm really impressed with the work you've done on getting the Vive Pro 2 to work on Linux.  Thank you very much.

Thanks to your kernel patches, along with this proxy driver, I've been able to somewhat use my headset under Linux.  Though it's worth noting I've had to change the following two things to get it to work:

- Change `gcc` to `g++` in the Makefile.
- Change the 2448 by 1224 resolution in driver.cpp to 3680 x 1836.  For some reason vrcompositor fails to enable direct mode unless set to this specific resolution in my case.  Perhaps something in the VIVE Console software configures what the headset advertises?

Now the main issue I'm having now is that I'm not understanding how exactly to use your newly added lens-server.exe to correct the lens distortion.  Right now I get the "fish-eye" effect.  I managed to build lens-server, but just running it on its own under Wine doesn't seem to do anything.  Should the proxy driver.cpp be somehow passing data into stdin of this program?  Are there some updates to driver.cpp you haven't yet committed to this repo?

CertainLach, 2022-01-23:
Hi

Yeah, driver.cpp should communicate with it, but currently there is performance issues, which makes it fail to start often (Due to image loading timeouts)

When it starts, everything is fine though

Implementation is on `lens-client` (uploaded now)

Currently i am looking at using `https://github.com/taviso/loadlibrary` instead of stdio communication

CandyGumdrop, 2022-01-23:
Thanks for the update.  Tried the new lens-client branch, however now I'm getting a messed up image:

[image]

Could this be because I changed the resolution to 3680x1836?  Does config.json need changing now?  From where did you obtain this file?

CertainLach, 2022-01-23:
As i saw - config.json doesn't change much between changing display resolution, i extracted it from somewhere in C:\ProgramData (Search for `vive_lh/*/Config.json` file, then extract some field from it)

I think you should change resolution here instead: https://github.com/CertainLach/SteamVR-Proxy-Driver/blob/master/lensServer/src/main.rs#L192

CandyGumdrop, 2022-01-23:
I found my own Config.json from Windows just to double-check, and I've also tried every combination I could of resolutions in both driver.cpp and in main.rs, however I am still getting this same garbled output.  Even using all the default values from the repo (which renders to my desktop instead of using direct mode).

Could it be that the stdout stream in main.rs is somehow mangling the data?  I'm not too familiar with Rust, however I know that using stdout in C on Windows may sometimes convert `\n` characters to `\r\n`, unless stdout is opened in "binary" mode.  Also, could it be something to do with how the Rust docs state this?:

> When operating in a console, the Windows implementation of this stream does not support non-UTF-8 byte sequences. Attempting to write bytes that are not valid UTF-8 will return an error.

If this is something to do with the stdout stream getting mangled, then at least switching to loadlibrary instead of an external process would fix this issue.

CertainLach, 2022-01-24:
There was multiple issues with stdio communication, which i have encountered only twice, and though it was errors due to timeouts.

Seems like steamvr doesn't cares about latency, and 200ms of waiting to convert all uvs is acceptable for me, so i think `loadlibrary` may be postponed

I think everything should be fine on master branch

CertainLach, 2022-01-25:
As for resolution not changing issue - make sure Vulkan WSI is working for you (i.e detects displays) in `~/.local/share/Steam/logs/vrcompositor.txt`
This resolution is hardcoded for now, as it is most likely works out of the box, higher resolutions may cause insufficient bandwidth error/kernel panic in DSC handling code (On kernel 5.16, panic is fixed in 5.17)

Idk how xrandr does it, but for me it drops some modes for some reason
Vulkan WSI on other hand detects them all, and even works on wayland

CandyGumdrop, 2022-01-25:
Lens distortion is working great now since the latest updates.  Thanks for finally making it possible to use the VIVE Pro 2 on Linux!

As for getting the lower resolution to work, I'm not sure what's going on exactly.  This is what I see in the vrcompositor logs with the resolution set to 2448x1224:

```
Tue Jan 25 2022 19:08:53.599014 - Looking for direct display through Vulkan WSI
Tue Jan 25 2022 19:08:53.650402 -  - Vulkan output 0: 2560x1080 resolution, 670x280 dimensions - AOC 2963 (HDMI-0)
Tue Jan 25 2022 19:08:53.650422 -  - Vulkan output 1: 3680x1836 resolution, 0x0 dimensions - HTC Corporation VIVE Pro 2 (DP-4)
Tue Jan 25 2022 19:08:53.650430 - Tried to find direct display through Vulkan WSI: (nil)
Tue Jan 25 2022 19:08:53.650438 - CHmdWindowSDL: Failed to create direct mode surface
```

And here's what I see with the resolution set to 3680x1836:

```
Tue Jan 25 2022 19:16:29.278468 - Looking for direct display through Vulkan WSI
Tue Jan 25 2022 19:16:29.332075 -  - Vulkan output 0: 2560x1080 resolution, 670x280 dimensions - AOC 2963 (HDMI-0)
Tue Jan 25 2022 19:16:29.332094 -  - Vulkan output 1: 3680x1836 resolution, 0x0 dimensions - HTC Corporation VIVE Pro 2 (DP-4)
Tue Jan 25 2022 19:16:29.332103 - Tried to find direct display through Vulkan WSI: 0x557c0da5eb30
Tue Jan 25 2022 19:16:29.336055 - Trying to match desired rate of 90.000000Hz.
Tue Jan 25 2022 19:16:29.336070 - 4 modes on display:
Tue Jan 25 2022 19:16:29.336080 -  - 0: 2448x1224@90.034004Hz
Tue Jan 25 2022 19:16:29.336088 -  - 1: 2448x1224@120.046005Hz
Tue Jan 25 2022 19:16:29.336096 -  - 2: 3264x1632@89.999001Hz
Tue Jan 25 2022 19:16:29.336104 -  - 3: 3680x1836@90.023003Hz
Tue Jan 25 2022 19:16:29.336112 - Selected mode 3.
Tue Jan 25 2022 19:16:29.338295 - Direct mode surface: 0x557c0de0f790
```

In either case, it also reports `(nil)` for RandR.

So far I haven't seen any kernel panic using Linux 5.10.

CertainLach, 2022-01-29:
Seems like you have only one preferred mode, are you just rebooted from windows for that?
Can you please share output of `xrandr` command?
For me, HMD reports all resolutions as preferred, and steamvr can use any of them

You can try to use vivectl (Now uploaded to this repo) to make HMD change reported modes

CertainLach, 2022-03-06:
- Config.json is now being read on driver startup
- HMD mode is now automatically reset
- lens-server encoding issue is solved

Thus i'm closing this issue, as no more work left to do

closed 2022-03-06 by CertainLach