CertainLach/VivePro2-Linux-Driver #2: Troubleshooting

nsf, 2022-01-30:
Trying to figure out how to make it work. I'll just describe everything I did. I have HTC Vive Pro 2 with lighthouse base stations 1.0 (from old vive).

Compiled a custom kernel with patches (based on archlinux 5.16.4). This seems to be working. I can see `non-desktop` flag in xrandr, as well as it showing the right mode: 
```
...
	non-desktop: 1
...
  4896x2448 (0x59) 1543.471MHz +HSync +VSync +preferred
        h: width  4896 start 4946 end 4950 total 4996 skew    0 clock 308.94KHz
        v: height 2448 start 3412 end 3414 total 3432           clock  90.02Hz
...
```
Seems like after either firmware update or reconfiguration on windows, the display begins to show a different resolution than what it was initially I plugged it in. But anyways, reading edid with edid-decode also shows that single resolution there. So far so good.

Compiled the driver, compiled the lensServer (after installing the right rust compiler targets and mingw suite). Copied everything from "asset" dir to "dist" dir, also copied lensServer binary into it. Did `make enable`. Trying to run steamvr via:
```
~/.local/share/Steam/steamapps/common/SteamVR/bin/vrmonitor.sh
```
I actually don't know the right way to start steamvr without steam. But running this script seems to be working normally. And this is what I see in the terminal (I removed excessive `\n`):

```
VR Server (v1634602223)
Using breakpad crash handler
Setting breakpad minidump AppID = 250820
Forcing breakpad minidump interfaces to load
Looking up breakpad interfaces from steamclient
Calling BreakpadMiniDumpSystemInit
PROXY HmdDriverFactory(268): loading library from /home/nsf/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64
PROXY HmdDriverFactory(275): initialized real factory
PROXY HmdDriverFactory(278): starting lens server from /home/nsf/tmp/VivePro2-Linux-Driver/lensServer/dist
PROXY HmdDriverFactory(296): testing lens server
PROXY HmdDriverFactory(284): hello from lens process
PROXY HmdDriverFactory(289): executing [/usr/bin/wine /home/nsf/tmp/VivePro2-Linux-Driver/lensServer/dist/lens-server.exe]
hello from lens server
server ready
PROXY HmdDriverFactory(305): request completed, assuming lens server is fine
PROXY HmdDriverFactory(312): server tracked init
PROXY Init(231): Init()
PROXY GetGenericInterface(209): GetGenericInterface(IVRServerDriverHost_006)
PROXY GetGenericInterface(209): GetGenericInterface(IVRSettings_003)
PROXY GetGenericInterface(209): GetGenericInterface(IVRProperties_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRDriverLog_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRDriverManager_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRResources_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRRenderModels_006)
PROXY GetGenericInterface(209): GetGenericInterface(IVRRenderModelsInternal_XXX)
PROXY GetGenericInterface(209): GetGenericInterface(IVRResources_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRDriverManager_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRSettings_003)
PROXY GetGenericInterface(209): GetGenericInterface(IVRSettingsInternal_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRPaths_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRPathsInternal_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRProperties_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRPropertiesInternal_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRServer_XXX)
PROXY GetGenericInterface(209): GetGenericInterface(IVRSystemLayerInternal_XXX)
PROXY GetGenericInterface(209): GetGenericInterface(IVRClientInternal_XXX)
PROXY GetGenericInterface(209): GetGenericInterface(LocalizationManager)
PROXY GetGenericInterface(209): GetGenericInterface(IVRCompositorSystemInternal_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRInput_010)
PROXY GetGenericInterface(209): GetGenericInterface(IVRInputInternal_002)
PROXY GetGenericInterface(209): GetGenericInterface(IVRChaperone_004)
PROXY GetGenericInterface(209): GetGenericInterface(IVRChaperoneSetup_006)
PROXY GetGenericInterface(209): GetGenericInterface(IVRChaperoneInternal_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRApplications_007)
PROXY GetGenericInterface(209): GetGenericInterface(IVRSystem_022)
PROXY GetGenericInterface(209): GetGenericInterface(IVRMailbox_002)
PROXY GetGenericInterface(209): GetGenericInterface(IVRDebug_001)
PROXY GetGenericInterface(209): GetGenericInterface(IVRBlockQueue_004)
PROXY GetGenericInterface(209): GetGenericInterface(IVRControlPanel_006)
[... 56 lines cut ...]
PROXY GetInterfaceVersions(246): done
PROXY GetDriverHandle(219): GetDriverHandle()

Initialize() is failed!!!
STEAM_RUNTIME_HEAVY: ./steam-runtime-heavy
Using breakpad crash handler
Setting breakpad minidump AppID = 250820
Forcing breakpad minidump interfaces to load
Looking up breakpad interfaces from steamclient
Calling BreakpadMiniDumpSystemInit
Using breakpad crash handler
Setting breakpad minidump AppID = 250820
Forcing breakpad minidump interfaces to load
Looking up breakpad interfaces from steamclient
Calling BreakpadMiniDumpSystemInit
Using breakpad crash handler
Setting breakpad minidump AppID = 250820
Forcing breakpad minidump interfaces to load
Looking up breakpad interfaces from steamclient
Calling BreakpadMiniDumpSystemInit
```

From what I can read from it, it seems to be showing some life signs, but then it fails. No clue what to try next. If anyone has any ideas, would be nice to hear it. The steamvr ui shows 214 error (Missing Driver).

And btw, **HUGE** thanks for all the effort to make it run on linux.

CertainLach, 2022-01-30:
> PROXY GetDriverHandle(219): GetDriverHandle()
>
> Initialize() is failed!!!

I think there should be error about not found display in `~/.local/share/Steam/logs/vrcompositor.txt`

Currently proxy driver doesn't reports correct HMD resolution to steam, instead resolution is hardcoded here:
https://github.com/CertainLach/VivePro2-Linux-Driver/blob/master/driver.cpp#L12-L13

nsf, 2022-01-30:
> I think there should be error about not found display in ~/.local/share/Steam/logs/vrcompositor.txt
Sadly I don't have any log entries there. The last one is from Jan 28 (this is me using htc vive 1). Seems like that thing isn't even starting yet.

However I see some log message in vrserver.txt.

```
Sun Jan 30 2022 16:48:27.550699 - Driver 'lighthouse' started activation of tracked device with serial number 'LHR-4C7F9EA6'
Sun Jan 30 2022 16:48:27.551304 - lighthouse: Attempting HID Open IMU: 9D9CD576DC
Sun Jan 30 2022 16:48:27.606035 - lighthouse: Lighthouse IMU HID opened
Sun Jan 30 2022 16:48:27.620977 - lighthouse: Attempting HID Open IMU: 7829524E84
Sun Jan 30 2022 16:48:27.621996 -   driver lighthouse implements interfaces IVRSettings_003 ITrackedDeviceServerDriver_005 IVRDisplayComponent_002 IVRDriverDirectModeComponent_007 IVRCameraComponent_003 IServerTrackedDeviceProvider_004 IVRWatchdog>
Sun Jan 30 2022 16:48:27.622025 - Loaded server driver lighthouse (IServerTrackedDeviceProvider_004) from /home/nsf/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/driver_lighthouse.so
Sun Jan 30 2022 16:48:27.622768 - Could not activate pending device LHR-4C7F9EA6: VRInitError_Driver_PeerDriverNotInstalled
Sun Jan 30 2022 16:48:27.622791 - Driver 'lighthouse' finished adding tracked device with serial number 'LHR-4C7F9EA6'
Sun Jan 30 2022 16:48:27.622803 - Driver lighthouse has no suitable devices.
```

Could this be related to the fact that I'm using Vive Pro 2 with old lighthouses?

CertainLach, 2022-01-30:
Ah, have you installed Vive Console (https://store.steampowered.com/app/1635730/VIVE_Console_for_SteamVR/)?
I still didn't understood its role on linux, but for some reason it is needed

nsf, 2022-01-30:
Not on linux no, will try that out. Thanks.

nsf, 2022-01-30:
Okay, this helps quite a bit. But something isn't right. However I can see a path towards making it working now.

If I try to run it with original resolution from driver.cpp (2448x1224), it runs, but... on a wrong display. It actually renders VR picture (correctly lighthouse tracked) but on a main monitor. I tried compiling with 4896x2448 res, but it fails to run in that case.

I'll try to figure it out. Something isn't connecting there with displays and the vrcompositor.

If I run with 4896x2448 compiled, it finds the right device:
```
Sun Jan 30 2022 17:18:47.224619 - Looking for direct display through RandR
Sun Jan 30 2022 17:18:47.224627 -  - Root 0x6ca
Sun Jan 30 2022 17:18:47.228810 -    - Output 0x54 - 12 modes, 1 preferred
Sun Jan 30 2022 17:18:47.228833 -      - Mode 0 0x59 4896x2448 (looking for 4896x2448)
Sun Jan 30 2022 17:18:47.228842 -        - Found matching output 84
Sun Jan 30 2022 17:18:47.228850 - Found candidate direct display as RandR output 0x54
Sun Jan 30 2022 17:18:47.236854 - Tried to find direct display through RandR: 0x560ef0960e70
Sun Jan 30 2022 17:18:47.236867 - Trying to match desired rate of 90.000000Hz.
Sun Jan 30 2022 17:18:47.236874 - 12 modes on display:
Sun Jan 30 2022 17:18:47.236882 -  - 0: 4896x2448@90.018005Hz
Sun Jan 30 2022 17:18:47.236889 -  - 1: 1920x1200@90.018005Hz
Sun Jan 30 2022 17:18:47.236895 -  - 2: 1920x1080@90.018005Hz
Sun Jan 30 2022 17:18:47.236901 -  - 3: 1600x1200@90.018005Hz
Sun Jan 30 2022 17:18:47.236908 -  - 4: 1680x1050@90.018005Hz
Sun Jan 30 2022 17:18:47.236914 -  - 5: 1280x1024@90.018005Hz
Sun Jan 30 2022 17:18:47.236920 -  - 6: 1440x900@90.018005Hz
Sun Jan 30 2022 17:18:47.236926 -  - 7: 1280x800@90.018005Hz
Sun Jan 30 2022 17:18:47.236932 -  - 8: 1280x720@90.018005Hz
Sun Jan 30 2022 17:18:47.236939 -  - 9: 1024x768@90.018005Hz
Sun Jan 30 2022 17:18:47.236945 -  - 10: 800x600@90.018005Hz
Sun Jan 30 2022 17:18:47.236951 -  - 11: 640x480@90.018005Hz
Sun Jan 30 2022 17:18:47.236962 - Selected mode 0.
Sun Jan 30 2022 17:18:47.245141 - Direct mode surface: 0x560ef0b7f740
```
But then fails somehow. I see some errors like so:
```
Sun Jan 30 2022 17:18:47.527936 - Creating static resources for tracked camera
Sun Jan 30 2022 17:18:47.528455 - Failed to get tracked camera recommended undistortion dimensions!
Sun Jan 30 2022 17:18:47.528465 - Tracked Camera: Failed to create static GPU resources.
Sun Jan 30 2022 17:18:47.528470 - D3D11 Camera Initialization failure.
Sun Jan 30 2022 17:18:47.528474 - Done initializing CGraphicsDevice resources
Sun Jan 30 2022 17:18:47.528481 - Loading MC resources
Sun Jan 30 2022 17:18:47.528489 - LoadMCImage: Error querying image
Sun Jan 30 2022 17:18:47.528545 - Distort initialized
```
Might be wrong config.json? Btw I didn't found mine on windows installation.

However, as I said, it runs fine with 2448x1224 resolution, but renders on my main monitor, it obviously fails to find the right randr output:
```
Sun Jan 30 2022 17:23:15.847394 - Looking for direct display through RandR
Sun Jan 30 2022 17:23:15.847401 -  - Root 0x6ca
Sun Jan 30 2022 17:23:15.851479 -    - Output 0x54 - 12 modes, 1 preferred
Sun Jan 30 2022 17:23:15.851490 -      - Mode 0 0x59 4896x2448 (looking for 2448x1224)
Sun Jan 30 2022 17:23:15.851515 -    - Output 0x55 - 0 modes, 0 preferred
Sun Jan 30 2022 17:23:15.851538 -    - Output 0x56 - 27 modes, 2 preferred
Sun Jan 30 2022 17:23:15.851545 -      - Mode 0 0x65 2560x1440 (looking for 2448x1224)
Sun Jan 30 2022 17:23:15.851552 -      - Mode 1 0x66 2560x1440 (looking for 2448x1224)
Sun Jan 30 2022 17:23:15.851574 -    - Output 0x57 - 0 modes, 0 preferred
Sun Jan 30 2022 17:23:15.851582 - Tried to find direct display through RandR: (nil)
Sun Jan 30 2022 17:23:15.851588 - Looking for direct display through Vulkan WSI
Sun Jan 30 2022 17:23:15.851595 - Tried to find direct display through Vulkan WSI: (nil)
Sun Jan 30 2022 17:23:15.851601 - CHmdWindowSDL: Failed to create direct mode surface
```

But it's something! I'll continue trying things out, will drop a note if I make it work.

CertainLach, 2022-01-30:
You can try to change HMD mode via provided vivectl

nsf, 2022-01-30:
Tried changing mode via vivectl, changing mode worked, but this didn't help with the device. Something strange happens with xrandr and on steamvr init. This is how my xrandr output looks when SteamVR is closed:
```
Screen 0: minimum 320 x 200, current 2560 x 1440, maximum 16384 x 16384
DisplayPort-0 disconnected (normal left inverted right x axis y axis)
   2448x1224     90.03 +
   1920x1200     90.03
   1920x1080     90.03
   1600x1200     90.03
   1680x1050     90.03
   1280x1024     90.03
   1440x900      90.03
   1280x800      90.03
   1280x720      90.03
   1024x768      90.03
   800x600       90.03
   640x480       90.03
DisplayPort-1 disconnected (normal left inverted right x axis y axis)
DisplayPort-2 connected primary 2560x1440+0+0 (normal left inverted right x axis y axis) 697mm x 392mm
   2560x1440    164.96 + 143.97 + 120.00*   99.95    59.95
   1920x1200    164.96
   1920x1080    120.00   100.00   119.88    60.00    60.00    50.00    59.94
   1600x1200    164.96
   1680x1050    164.96
   1280x1024    164.96
   1440x900     164.96
   1280x800     164.96
   1280x720      60.00    50.00    59.94
   1024x768      60.00
   800x600       60.32
   720x480       60.00    59.94
   640x480       60.00    59.94
HDMI-A-0 disconnected (normal left inverted right x axis y axis)
```
But then if I start SteamVR, suddenly HMD migrates to HDMI and appears disconnected (obviously):
```
Screen 0: minimum 320 x 200, current 2560 x 1440, maximum 16384 x 16384
DisplayPort-0 disconnected
DisplayPort-1 disconnected
DisplayPort-2 connected primary 2560x1440+0+0 697mm x 392mm
   2560x1440    164.96 + 143.97 + 120.00*   99.95    59.95
   1920x1200    164.96
   1920x1080    120.00   100.00   119.88    60.00    60.00    50.00    59.94
   1600x1200    164.96
   1680x1050    164.96
   1280x1024    164.96
   1440x900     164.96
   1280x800     164.96
   1280x720      60.00    50.00    59.94
   1024x768      60.00
   800x600       60.32
   720x480       60.00    59.94
   640x480       60.00    59.94
HDMI-A-0 disconnected
  2448x1224 (0x785) 432.201MHz +HSync +VSync
        h: width  2448 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height 1224 start 1864 end 1866 total 1884           clock  90.03Hz
  1920x1200 (0x786) 432.201MHz +HSync +VSync
        h: width  1920 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height 1200 start 1864 end 1866 total 1884           clock  90.03Hz
  1920x1080 (0x787) 432.201MHz +HSync +VSync
        h: width  1920 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height 1080 start 1864 end 1866 total 1884           clock  90.03Hz
  1600x1200 (0x788) 432.201MHz +HSync +VSync
        h: width  1600 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height 1200 start 1864 end 1866 total 1884           clock  90.03Hz
  1680x1050 (0x789) 432.201MHz +HSync +VSync
        h: width  1680 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height 1050 start 1864 end 1866 total 1884           clock  90.03Hz
  1280x1024 (0x78a) 432.201MHz +HSync +VSync
        h: width  1280 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height 1024 start 1864 end 1866 total 1884           clock  90.03Hz
  1440x900 (0x78b) 432.201MHz +HSync +VSync
        h: width  1440 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height  900 start 1864 end 1866 total 1884           clock  90.03Hz
  1280x800 (0x78c) 432.201MHz +HSync +VSync
        h: width  1280 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height  800 start 1864 end 1866 total 1884           clock  90.03Hz
  1280x720 (0x78d) 432.201MHz +HSync +VSync
        h: width  1280 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height  720 start 1864 end 1866 total 1884           clock  90.03Hz
  1024x768 (0x78e) 432.201MHz +HSync +VSync
        h: width  1024 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height  768 start 1864 end 1866 total 1884           clock  90.03Hz
  800x600 (0x78f) 432.201MHz +HSync +VSync
        h: width   800 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height  600 start 1864 end 1866 total 1884           clock  90.03Hz
  640x480 (0x790) 432.201MHz +HSync +VSync
        h: width   640 start 2498 end 2502 total 2548 skew    0 clock 169.62KHz
        v: height  480 start 1864 end 1866 total 1884           clock  90.03Hz
```

Maybe some traces of old HTC Vive, I was connecting it via HDMI. I might actually try migrating HTC Vive to display port (now I have the cables, thanks to Vive Pro 2) and then switch it back to Pro 2, might help.

CertainLach, 2022-01-30:
While image is displayed, it migrates to hdmi for me too (rx6900), this is not a problem, but some software (?) quirk

nsf, 2022-01-30:
Ha-ha, it worked! Connected old HTC Vive via DP and now vive pro 2 shows some image. Well okay, seems like it begins to work. I still see some minor eye misalignment or something like that. Could be the fact that I'm using your config.json. Or maybe some IPD issue. But yeah, overall it's up and running!

Thanks a lot.

CertainLach, 2022-01-30:
This config is somehow extracted from HMD, but this is some complex code required, and i can't yet correctly load mcu.bin in ida pro (this is plain cortex-m0 firmware with some prefixed data, yet i can't find exception vector (file offset) and thus base address)

Windows driver stores it in ProgramData (search for vive_lh/*/Config.json)

nsf, 2022-01-30:
Some more news from me. Found the config here:
```
cat ./ProgramData/HTC/ViveSoftware/ViveVRRuntime/config/vive_lh/0833ea6ec169/Config.json | jq .inhouse_lens_correction > /tmp/1.json
```

It helped with colors. There was an issue with colors I didn't mention. Well, obviously. Lense distortion causes chromatic aberration.

But what helped me with "eye misalignment" is switching INVERT_MATRIX to false. I'm not really sure 100% about this. But it works with "false" just fine.

Couldn't make it work with highest resolution. I think I saw a bug you reported somewhere on mesa (https://gitlab.freedesktop.org/mesa/mesa/-/issues/5898). I also have radeon 6900 xt btw. In my case it sounds similar. On highest res it renders memory garbage into headset display. Doesn't crash the system though. But it's not like I tried running it for a while. Only tried two resolutions: r2448x1224f90, r4896x2448f90. IIRC on high resolution it might get into the area of DP's display stream compression (DSC). Although also I recall it was only about 120hz, but maybe 90hz too. Might be related.

CertainLach, 2022-01-30:
Hmm, seems like there is multiple hardware revisions?
Can you share your config.json?

Yes, with high resolution DSC is required, and there is some timings bug in amdgpu driver i think (not related to mesa issue), as sometimes i see garbage via DSC too

nsf, 2022-01-30:
Here's my config (you can use https://nosmileface.dev/jsondiff/ to compare them):
```
{
  "left": {
    "center": {
      "left": {
        "x": 93,
        "y": 1
      },
      "right": {
        "x": 0,
        "y": 0
      }
    },
    "create_time": "2021-08-16 15:26:03",
    "distortion_coeff": {
      "left": {
        "blue": {
          "k0": 1,
          "k1": 0.20094309747219086,
          "k10": -61544020,
          "k11": 97593032,
          "k12": -62529424,
          "k2": 0.7919728755950928,
          "k3": 7.519835948944092,
          "k4": 222.94488525390625,
          "k5": -5567.43359375,
          "k6": 41956.546875,
          "k7": 73406.2578125,
          "k8": -2987257.75,
          "k9": 19558274,
          "p1": 0.010867365635931492,
          "p2": -0.3462744653224945,
          "p3": 5.641901969909668,
          "p4": 0.0028013635892421007,
          "p5": -0.02303699031472206,
          "p6": 1.0176193714141846,
          "s1": 0.028603898361325264,
          "s2": -0.4545656442642212,
          "s3": 0.008520251139998436,
          "s4": -0.1748131662607193
        },
        "green": {
          "k0": 1,
          "k1": -0.6975202560424805,
          "k10": 20702402,
          "k11": -22706338,
          "k12": 9770733,
          "k2": 26.42713737487793,
          "k3": -387.66558837890625,
          "k4": 3495.500732421875,
          "k5": -19086.93359375,
          "k6": 84521.5703125,
          "k7": -472601.125,
          "k8": 2639537.5,
          "k9": -9747468,
          "p1": 0.00828350055962801,
          "p2": -0.21645258367061615,
          "p3": 3.1165502071380615,
          "p4": 0.003026689635589719,
          "p5": -0.024105601012706757,
          "p6": 0.9394453167915344,
          "s1": 0.015528388321399689,
          "s2": -0.21336869895458221,
          "s3": 0.006977192126214504,
          "s4": -0.1447151154279709
        },
        "red": {
          "k0": 1,
          "k1": -1.1382137537002563,
[... 358 lines cut ...]
          1.2421778440475464,
          0.008169935084879398
        ],
        [
          0,
          0,
          -1
        ]
      ]
    },
    "model_type": "strengthen_high_order",
    "normalizedRadius": 1860.4783935546875,
    "resolution": {
      "height": 2448,
      "width": 2448
    },
    "scale": {
      "left": 0,
      "right": 0.45793330669403076
    },
    "scale_ratio": 2,
    "version": "1.10"
  }
}
```

CertainLach, 2022-01-30:
Yep, with your config matrix is already inverted, so `INVERT_MATRIX` is not required

Lens center is also moved, so probally some changes at driver side also needed, as currently it uses default position, which is same as mine `config.json` (except `y`, but it already looks fine to me)

CertainLach, 2022-02-01:
Seems like DSC is toggleable, added this toggle (and 2 others) to `mode` command: https://github.com/CertainLach/VivePro2-Linux-Driver/commit/e66d29429d0051351889841d52baa7cf24cbfcaf

This doesn't seems to work, however)

nsf, 2022-02-02:
Yeah, I keep getting the same garbage output on r4896x2448f90. Also doesn't seem like those mode flags make a difference. Maybe they were used for debugging during development or something.

CertainLach, 2022-02-02:
DSC flag adds additional section to edid DisplayId
This section doesn't seems to be read by kernel
Maybe there is some non-default DSC configuration is used, and kernel NEEDS to respect this EDID section?..

nsf, 2022-02-02:
Yes, except I saw this section before you added flags feature. It appears when you switch hmd to 5k resolution.

r2448x1224f90 mode (no flags):
```
Block 1, DisplayID Extension Block:
  Version: 2.0
  Extension Count: 0
  Display Product Primary Use Case: Head-mounted Virtual Reality (VR) display
  Video Timing Modes Type 7 - Detailed Timings Data Block:
    DTD:  2448x1224   90.033772 Hz   1:1    169.624 kHz    432.201000 MHz (aspect 1:1, no 3D stereo, preferred)
      Modeline "2448x1224_90.03" 432.201  2448 2498 2502 2548  1224 1864 1866 1884  +HSync +VSync
  Checksum: 0xb3
Checksum: 0x90
```
r4896x2448f90 mode (no flags):
<pre><code>Block 1, DisplayID Extension Block:
  Version: 2.0
  Extension Count: 0
  Display Product Primary Use Case: Head-mounted Virtual Reality (VR) display
  Video Timing Modes Type 7 - Detailed Timings Data Block:
    <b>These timings support DSC pass-through</b>
    DTD:  4896x2448   90.017877 Hz   1:1    308.941 kHz   1543.471000 MHz (aspect 1:1, no 3D stereo, preferred)
      Modeline "4896x2448_90.02" 1543.471  4896 4946 4950 4996  2448 3412 3414 3432  +HSync +VSync
  Vendor-Specific Data Block (0x7e) (VESA), OUI 3A-02-92:
    Data Structure Type: DP
    Default Colorspace and EOTF Handling: Native as specified in the Display Parameters DB
    Number of Pixels in Hor Pix Cnt Overlapping an Adjacent Panel: 0
    Multi-SST Operation: Not Supported
    <b>Pass through timing's target DSC bits per pixel: 8.0000</b>
  Checksum: 0xf1
Checksum: 0x90</code></pre>

I saw those DSC-related bits as well and yes, grepped the kernel code, it doesn't seem to be using it. But also I haven't found anything related to those. It mentions some DSC "pass-through" mode. I don't know how DSC works at all, but since HMD is connected via dongle, it might as well be relevant here. Because there is obviously a question: where DSC decompression happens on dongle or on HMD itself.

When it comes to kernel, the only thing I see there is `aconnector->dsc_settings.dsc_force_disable_passthrough` (commit: https://www.mail-archive.com/amd-gfx@lists.freedesktop.org/msg63971.html), funny enough it's not used anywhere. Maybe in amdgpu-pro somewhere? Or it could be a debug leftover.

Wasn't able to find anything else about DSC "pass-through" mode.

nsf, 2022-02-02:
Also since there are like 5 of us (VR linux users), I'll shamelessly drop a link here, maybe you or somebody else reading this is interested: https://github.com/nsf/vrmp.

Working on a VR media player for linux. It's a very early development version though, but somewhat usable if one can build and run it.

CertainLach, 2022-03-06:
As all problems mentioned in this issue are resolved now
- No automatic detection of HMD mode (Implemented via forced reconfiguration)
- Requirement for manual specification of config (Implemented via fetching it on driver start)
- Matrix inversion (This quirk is detected and automatically applied)
- Highest resolution not working (Implemented via kernel patch)

I am closing it
Feel free to open new issues in case of more broken things :D

closed 2022-03-06 by CertainLach