CertainLach/VivePro2-Linux-Driver #8: Build errors: nix (extra experimental features in use, etc)

h8rt3rmin8r, 2022-08-21:
After installing the recommended 3rd party package manager `nix` ([multi-user installation](https://nixos.org/manual/nix/stable/installation/multi-user.html)) and subsequently running the build command (`nix build .`), the following error was presented:
```bash
error: experimental Nix feature 'nix-command' is disabled; use '--extra-experimental-features nix-command' to override
```
Taking the error message into account, I tried a modified version of the initial build command to satisfy the `nix-command` requirement:
```bash
nix --extra-experimental-features nix-command build .
```
This modified build command then produced a second error message:
```bash
error: experimental Nix feature 'flakes' is disabled; use '--extra-experimental-features flakes' to override
```
In an attempt to satisfy the issues in this error, I then attempted a third build command to satisfy the `flakes` requirement:
```bash
nix --extra-experimental-features nix-command --extra-experimental-features flakes build .
```
This modified build command then produced yet another error message:
```bash
error: flake 'path:/home/h8rt3rmin8r/data/vive/VivePro2-Linux-Driver-master' does not provide attribute 'packages.x86_64-linux.default' or 'defaultPackage.x86_64-linux'
```

Perhaps these issues are coming from my use of Ubuntu instead of some other system. I'm not sure. Either way I am now at a bit of an impasse - the only alternative installation method referenced includes commands involving `pacman` (so intended for Arch-based systems). Is it therefore safe to assume this `VivePro2-Linux-Driver` repository does not currently support installation on Ubuntu? If so, support for Ubuntu would be very much appreciated as setting up a dual boot just to use my new Vive Pro 2 would be quite unfortunate (I've not used Windows in over 10 years now and I'd hate to re-enter the Matrix again lol).

For what it's worth, here's some system info:
```
$> neofetch --package_managers on --os_arch on --speed_type bios --cpu_speed on --refresh_rate on --gpu_brand on --gpu_type all
            .-/+oossssoo+/-.               h8rt3rmin8r@novx-core 
        `:+ssssssssssssssssss+:`           --------------------- 
      -+ssssssssssssssssssyyssss+-         OS: Ubuntu 20.04.4 LTS x86_64 
    .ossssssssssssssssssdMMMNysssso.       Shell: bash 5.0.17 
   /ssssssssssshdmmNNmmyNMMMMhssssss/      CPU: Intel i9-10920X (24) @ 4.800GHz 
  +ssssssssshmydMMMMMMMNddddyssssssss+     GPU: NVIDIA 65:00.0 NVIDIA Corporation Device 2203 
 /sssssssshNMMMyhhyyyyhmNMMMNhssssssss/    Memory: 6499MiB / 257435MiB (2%) 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Packages: 3545 (dpkg), 47 (nix-default), 29 (snap) 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Disk (/): 467G / 915G (54%) 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   Disk (hotwire): 131G / 19T (1%) 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   Disk (SG-10000-F): 4.6T / 9.1T (53%) 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Disk (SG-10000-A): 6.6T / 9.1T (77%) 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.
 /sssssssshNMMMyhhyyyyhdNMMMNhssssssss/
  +sssssssssdmydMMMMMMMMddddyssssssss+
   /ssssssssssshdmNNNNmyNMMMMhssssss/
    .ossssssssssssssssssdMMMNysssso.
      -+sssssssssssssssssyyyssss+-
        `:+ssssssssssssssssss+:`
            .-/+oossssoo+/-.
```
```
$> glxinfo -B
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 24564 MB
    Total available memory: 24564 MB
    Currently available dedicated video memory: 23472 MB
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 3090 Ti/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 515.65.01
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6.0 NVIDIA 515.65.01
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)

OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 515.65.01
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

CertainLach, 2022-08-21:
Yes, `--extra-experimental-features` is required, unless you use unstable nix branch. I should mention this in README.md

Other error (`does not provide attribute 'packages.x86_64-linux.default'`) is caused by missing package name
In `.#driver-proxy-release` the `#driver-proxy-release` part is not a comment, this is a attribute of driver release package name

So, fixed command is:
```shell
nix --extra-experimental-features nix-command,flakes build .#driver-proxy-release
```

h8rt3rmin8r, 2022-08-22:
Perfect. Thank you for the information. Especially the bit about the `.#driver-proxy-release` not being a comment. My brain lives in the world of `bash` a lot so when I saw the hash symbol in there I instantly thought that was a comment so it was very helpful when you cleared that up for me.

Initial results seem to be working. The build process is taking a long time to complete though. Probably normal.

closed 2022-08-22 by h8rt3rmin8r