CertainLach/VivePro2-Linux-Driver #9: Suggestion: Alternative Build Process (etc)

h8rt3rmin8r, 2022-08-23:
# Situation

After receiving your timely and useful help related to the build process with `nix`, the build process completed successfully (total process time was around 90 minutes). The final output of this build process was a new linked directory in the project root called `result`. This new directory link points to the following location:
```
/nix/store/zspbnaxp2dsc7v2f37szg2fjbf9yv7kb-vivepro2-driver-proxy-release-0.1.0
```
This `result` location contains exactly seven files and two directories that inhabit around 86.6 MB of storage space. These items are arranged as follows:

[image]

If I am correctly understanding the installation instructions provided in the repository `README.md`, the next step will be to run the `install.sh` script found inside of the output `result` location.

# Thoughts In Hindsight

The use of [nix](https://nixos.org/) to perform the build process strikes me as a bit excessive (just speaking as an end user here). It works. It definitely works. But I've never used it before (had not even heard of it before, actually) and can't imagine myself needing to use it again anytime in the foreseeable future. Despite these facts, I'm left with over **532,000** mysterious files inhabiting a new `nix` directory at the root of my system. Upon completion of this build process, the first thing many users (myself included) will set out to do is to discover how to fully purge `nix` from their systems.

In summary, it would really help to get this excellent creation into the hands of users if the building process could be done with something a bit more common or conventional. I'm not hardly the developer that you are so I may just be asking for something beyond the scope of what is realistic - but at least I can try `¯\_(ツ)_/¯`

I admire the fantastic work you're doing with this project and, although I've gone about the optional process of building the driver instead of getting a pre-built package via sponsoring the project, I'm still seriously considering sponsoring the project anyway. The world needs this driver and the folks up at HTC need to start realizing that some of the most innovative and creative power users of their products are working exclusively on Linux ;-)

CertainLach, 2022-08-23:
Most users should not build this project from the source, but instead, just donate on Patreon for binary release access.
However, I'm leaving the option to get binaries for free, If the user is ok with setting up the development environment I prefer.

As for development - nix is the clearest way to get things done, and the `/nix` directory is not a problem, once you learn about gc roots, and `nix-collect-garbage` command.
Some cleanup will be needed in any case, either `docker prune`, or manual deinstallation of rust, clang, MinGW, rust MinGW toolchain, specific glibc versions, and some other things.

The build process is slow because it needs to rebuild some of the dependencies for support of older Ubuntu, i.e glibc, some windows libraries for lens-proxy, and all of its dependencies/dependents. After that, later rebuilds will be much faster.

The solution for this problem would be using shared binary caches (at localhost cache is used already between rebuilds), either official (https://cache.nixos.org/), or custom (https://cachix.org)
It is also possible to throw some of the compatibility stuff, but it is not required anyway for development, as you don't need release package to test changes in proxy driver, and standard dev shell is pretty lean already

h8rt3rmin8r, 2022-08-23:
I'll try to read up some more on gc roots and nix in general - I have lots to learn, for sure.

On a side note - the install script worked. The only issue was:
```
./install.sh: 3: set: Illegal option -o pipefail
```
I overcame this by invoking the script directly via `bash`. (See the following screenshot for the full input/output)

[image]

Also, I finally went for it and sponsored the project. Keep up the great work!

closed 2022-08-23 by h8rt3rmin8r

CertainLach, 2022-08-23:
Yep, I fixed `pipefail` in https://github.com/CertainLach/VivePro2-Linux-Driver/commit/02b25f136a5d3ad8d4fd4f4108592d285ebf49c5, will release binaries later

I would make binary releases free, once (when?) I will make this driver work in less hacky ways. Currently it will break randomly if build system is not compatible with the one used to build steamvr, and the best way to ensure this - build binaries by myself in completely reproducible way, and nix helps a lot in this task.
And then, if something is broken - I can help in chat, and it won't be free.

This also will help a lot once I release wireless support, I only hope no hardware modifications will be required, as they are required on my dev environment.