CertainLach/VivePro2-Linux-Driver #11: What's the difference between Patreon code and this GitHub code?

PARKBONG, 2022-08-31:
Hello, I'm a beginner at Linux, so I have a few questions.

1. Does your code work in Ubuntu 20.04?
2. What's the difference between Patreon code and this GitHub code? As I am having trouble using Nix, I'm willing to pay if the Patreon code provides a simpler installation. 
3. I would appreciate it if you could share the video that works on Ubuntu 20.04. (With Vive Pro 2 & SteamVR)

Thank you.

h8rt3rmin8r, 2022-08-31:
> Does your code work in Ubuntu 20.04?

I managed to successfully build the driver on Ubuntu 20.04. Bear in mind, this build process can take around 90 minutes if you have 24 CPU cores etc (that was my experience, anyway). 

> ...I am having trouble using Nix, I'm willing to pay if the Patreon code provides a simpler installation.

About Nix ... so um ... it's large. The installation process makes the total footprint of nix explode to over 500,000 files on your system but the developer told me it's necessary because that's his toolchain - it's how he wants to develop this thing so if we want to access this driver for free then deal with it (fair enough I suppose). To make sure nix is installed correctly, I'd advise you use the multi-user installation method (referenced on the NixOS page [here](https://nixos.org/download.html)).

> What's the difference between Patreon code and this GitHub code?

Nothing, as far as I can tell. Basically what you're getting by going the Patreon route is the time (and complexity) you avoid by NOT compiling the driver yourself via nix.

**IN SUMMARY**

Since you're new to Linux systems then you might benefit from the experience of compiling the driver via `nix`. That said, it makes perfect sense if you don't want to bother. Either way, I'd consider sponsoring this project because it is quite literally the only one of its kind at the moment (basically it's a big deal). And in case you're wondering - no, I'm not getting paid to promote this project and have nothing to gain by you sponsoring it - and yes, I sponsored this project after compiling the driver manually.

I hope this information helps you out a little bit.

CertainLach, 2022-08-31:
1. It should, I haven't tested it personally, but I have fixed some ubuntu related issues, and will fix more, if there is
2. No difference, on patreon I publish binaries built using nix

Also, it is possible to build this project without using nix, but it may, or may not work.
I don't recommend going this way, but I have third-party instruction linked in readme.

closed 2022-09-08 by PARKBONG