CertainLach/VivePro2-Linux-Driver #12: Vertical distance between eyes

galister, 2022-08-31:
Hi, I'm having an issue where there's a significant vertical difference between the eyes, as if there was a vertical component to the IPD.

In an upright orientation, the right eye is lower, while the left eye is higher. It's not pleasant to look at :sweat_smile: 

The best way I can explain is imagine tilting your head to the right by ~45 degrees, and then using the positions of your eyes from that pose for the upright position. At least this is what it looks like to me. The images are in the correct orientation.

This does not occur when using with Windows.

What I've tried so far:
- SteamVR beta / stable / linux_v1.14
- forcing settings via steamvr.vrsettings, like `ipd`, `ipdOffset`
- replace the `LibLensDistortion.dll` from the Vive Console (in case there has been an update)
- `lighthouse_console.exe downloadconfig` under Windows and put that as config.json under the Linux SteamVR's config/lighthouse.

CertainLach, 2022-08-31:
Ehh?.. Are u sure eyes aren't just switched left-right? :D
Because I can't imagine what even can lead to this result for now

galister, 2022-08-31:
Can't really tell. How would you go about un-switching the eyes? force a negative `ipd` value?

CertainLach, 2022-08-31:
No, changing `ipd` won't help
You need to toggle this condition to make it work:
https://github.com/CertainLach/VivePro2-Linux-Driver/blob/master/bin/driver-proxy/src/hmd.rs#L77
```diff
- if eEye == EVREye::Eye_Left
+ if eEye != EVREye::Eye_Left
```

(If you are patron, then you can write in messages on patreon, and I'll build driver with this change for you)

galister, 2022-09-07:
No luck, the vertical offset is still very much there, except things are way trippier now. I'll try and poke around more when I have time. Would welcome any tips though, if you can think of anything.

This HMD has actually had both LCD panels replaced through RMA, so that might have something to do with it.

galister, 2022-09-07:
I tried changing the calls to `lens.matrix_needs_inversion()` to flat `false`, and the distortion issue is gone.

https://github.com/CertainLach/VivePro2-Linux-Driver/blob/master/bin/driver-proxy/src/hmd.rs#L108

https://github.com/CertainLach/VivePro2-Linux-Driver/blob/master/bin/driver-proxy/src/hmd.rs#L127

However, now I only see back faces, and things look really weird. But at least it's comfortable to look at.

CertainLach, 2022-09-07:
Oh, then your image was upside down, because matrix inversion swaps faces and makes image upside-down!

CertainLach, 2022-09-07:
Can you share logs and Config.json from original driver running on windows? Maybe at RMA they have placed LCD panels other way, and there is some changed config fields to look at
Logs and Config.json should be located in C:/ProgramData/HTC

CertainLach, 2022-09-07:
Meanwhile, you can only change GetProjectionRaw, and then everything should be ok for you

CertainLach, 2022-09-07:
It is hard to understand logic regarding inversions in the original driver and mine matrix_needs_inversion function only implements how it should look in my opinion, not the correct code for handling this

galister, 2022-09-08:
[WaveConsole_20220906_140452_38472.txt](https://github.com/CertainLach/VivePro2-Linux-Driver/files/9526733/WaveConsole_20220906_140452_38472.txt)
[Config.json.txt](https://github.com/CertainLach/VivePro2-Linux-Driver/files/9526738/Config.json.txt)

Here's the config. Vastly different than the one I got via `lighthouse_console.exe downloadconfig`

CertainLach, 2022-09-08:
I'm not sure, what `lighthouse_console.exe` downloads, but I think it returns valve config (https://github.com/CertainLach/VivePro2-Linux-Driver/blob/master/crates/vive-hid/src/lib.rs#L100-L133), and not htc one (https://github.com/CertainLach/VivePro2-Linux-Driver/blob/master/crates/vive-hid/src/lib.rs#L251-L287)

htc config contains everything from valve config, except some fields, plus lens configuration
I think valve config exists for backwards compatibility (But this compatibility is not implemented on the steamvr side)

galister, 2022-09-18:
I've tried changing the inversions with all kinds of combinations, and i either get the front faces culled (sometimes with usable eye positions) or the eye positions with the vertical difference :(

CertainLach, 2022-10-22:
Can you capture the image from headset, so I can make sure I understand what's happening, and maybe provide some solution?

galister, 2022-10-22:
It's being RMA'd for the screen (for the 2nd time), I'm getting it back next week. Will let you know if it's still an issue, and try to get a mirror capture or the sorts.

closed 2022-10-24 by galister