CertainLach/VivePro2-Linux-Driver #14: HMD display are not working

PARKBONG, 2022-09-13:
Hi. After the Steam update (Not SteamVR), HMD works wierd.

They seem to create a new display that occupy my monitor and send the HMD display data to the new display window.

1. After the launch SteamVR, a previously unseen window is created on my monitor. It is impossible to resize, and it is like a single display.

2. `display VR view` shows black screen, and also my HMD displays are same.

3. Even though I disconnected the DP cable of HMD, 
 3.1. SteamVR still recognizes that the HMD is connected
 3.2. HMD pose tracking is wokring
[image]

4. I got a new `Headset Notice` -- SteamVR has Detected your headset running as a monitor. For best results, please switch to Direct Display Mode. But after changing the setting, It wasn't help.

5.  Reinstall Steam VR & Vivepro2-linux-driver doesn't help. Could you tell me which settings I should change?

Thanks!

PARKBONG, 2022-09-13:
Umm, Just **reboot the ubuntu and check the cable connection solved the problem..** haha

I will close this issue three days after. I always appreciate your support!

And I have another question, Do you know how to access the application when I stuck in the `Next Up`?
[image]

CertainLach, 2022-09-13:
Seems like this is an application error, the driver can't affect the launching of steam apps

closed 2022-09-14 by PARKBONG