CertainLach/VivePro2-Linux-Driver #15: No vision on HMD

johnsmitth, 2022-12-17:
SteamVR sees all the peripheralls. Tracking all seems to work. However "vrcompositor" starts on my desktop and I get no vision to the VivePro2 HMD (LED on the side remains red, showing the display is off).

SteamVR repeatedly throws an error message _"SteamVR has detected your headset running as a monitor. For best results, please switch to Direct Display Mode."_ however clicking _"Enable Direct Display Mode"_ seems to do nothing.

steamvr.vrsettings lists the device:
> "LastKnown" : {
>       "HMDManufacturer" : "HTC",
>       "HMDModel" : "VIVE_Pro 2 MV"
> }
but nothing else about the device's brightness or resolution settings.

Ostensibly the issue appears to be identical to https://github.com/CertainLach/VivePro2-Linux-Driver/issues/14 posted by PARKBONG, however I've restarted the machine multiple times, and when I dual boot into Windows10 everything works fine, so its not a _"cable issue"_

Tried reinstalling the repo, tried deleting the SteamVR USB drivers and starting over (an issue I'd previously encountered on Windows). Nothing seems to work.

System:
Arch Linux + Gnome43.2
Ryzen 3700X + GTX1080

Let me know if there are any diagnostic tests you want me to perform. I can even shred the drive and reinstall Arch if you want. Its a pain, but not as much of a pain as having to boot into Windows to to VR stuff...

CertainLach, 2022-12-17:
Have you installed the kernel patches?

johnsmitth, 2022-12-17:
I'm assuming so. I ran the nix command as per the installation instructions in the README.md:

```sh
# Note: the #driver-proxy-release part of this command is not a comment, it's just github syntax highlighter is wrong
nix build --extra-experimental-features nix-command --extra-experimental-features flakes .#driver-proxy-release
```

 As I said, the tracking and everything is working correctly. Just no vision on the HMD.

CertainLach, 2022-12-17:
This is not enough, you should also patch your kernel: https://github.com/CertainLach/VivePro2-Linux-Driver#required-kernel-patches

There is build script for building patched kernel for archlinux using docker: https://github.com/CertainLach/VivePro2-Linux-Driver/blob/master/kernel-patches/build-archlinux.sh

johnsmitth, 2022-12-17:
Ah, ok. Had to run the script in `result/install.sh`

The instructions in the README.md give the impression that the `./install.sh` command is specific to the precompiled patreon download. It's not immediately clear from the instructions that it also applies to the git repo verson, or where the install script is located.

Kindof working now, although now I'm encountering a new problem. Two steps forward, one step back. But I'll open a new issue for it.

closed 2022-12-17 by johnsmitth

johnsmitth, 2022-12-17:
FYI, the docker build script fails. I'm getting some vision to the HMD though so I guess something worked.

`
Hunk #1 succeeded at 6289 (offset 12 lines).
Hunk #2 succeeded at 6312 (offset 12 lines).
Hunk #3 FAILED at 6376.
1 out of 3 hunks FAILED -- saving rejects to file drivers/gpu/drm/drm_edid.c.rej
patching file include/drm/drm_connector.h
Hunk #1 FAILED at 656.
1 out of 1 hunk FAILED -- saving rejects to file include/drm/drm_connector.h.rej
patching file include/drm/drm_displayid.h
==> ERROR: A failure occurred in prepare().
    Aborting...
The command '/bin/sh -c cd kernel && makepkg --holdver' returned a non-zero code: 4
Unable to find image 'proxy-driver-archlinux-kernel:latest' locally

docker: Error response from daemon: pull access denied for proxy-driver-archlinux-kernel, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
Unable to find image 'proxy-driver-archlinux-kernel:latest' locally
docker: Error response from daemon: pull access denied for proxy-driver-archlinux-kernel, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
`

CertainLach, 2022-12-17:
Well, some patches needs to be updated

CertainLach, 2022-12-17:
Most critical patch was merged in 5.18, but one is still required to use max resolution:

> Support fixed DSC BPP rate https://lore.kernel.org/linux-kernel/20220220151940.58327-1-iam@lach.pw/, https://lore.kernel.org/linux-kernel/20220220151940.58327-2-iam@lach.pw/ - not yet merged, currently planning to propose it to 5.19, without this patch highest resolution modes will not work

johnsmitth, 2022-12-17:
This might be what is causing my new issue then. I've tried changing the `steamvr.vrsettings` file. The only resolution I get vision from is #0 and its scrambled (almost like its rendering half the left image to the right eye, and vice-versa) and there are rainbow coloured bars down the right side in both eyes.

#1 #2 #3 is no vision to the HMD

#4 and #5 opens the vrcompositor on the desktop like previously.

I'm assuming your precompiled patreon version works out of the box? I might just try that one.

CertainLach, 2022-12-17:
Patreon version is exactly the same as built using nix

johnsmitth, 2022-12-17:
So it doesn't come with the prebuilt kernel patches (one of which appear to have failed to build for me)?

CertainLach, 2022-12-17:
No, I make those things on demand
Here is prebuilt archlinux kernel: https://mega.nz/folder/6nAV1B7K#hnebwYsYQITU9MyzhfrklQ

johnsmitth, 2022-12-18:
I installed the above kernel, and it seems to have bricked my system. Unfortunately I start my nightshift roster in a couple of hours, so I might just have to reload the OS and revist this project again next week.

Thanks for taking the time.

cklarhorst, 2023-01-03:
First of all, I really like this project. ❤️ Thank you @CertainLach.

I have a similar system: Threadripper + 1060 but I'm on gentoo with 6.0.11 + patch 0001 (your repo) and 0003  (from [vive-pro-2-on-linux](https://github.com/santeri3700/vive-pro-2-on-linux))

> The only resolution I get vision from is #0 and its scrambled (almost like its rendering half the left image to the right eye, and vice-versa) and there are rainbow coloured bars down the right side in both eyes.

I had the same problem, but I was able to work around it.
The scrambled version looks like this (only one eye):
[image]

On my System everything depends on the nvidia setting `Option "AllowHMD" "XXX"`.

With  `Option "AllowHMD" "false/no"` (the default) xrandr doesn't report any modes (Output listed as unconnected) and steamvr only finds the vive with resolution mode 0. The image is then scrambled.

With  `Option "AllowHMD" "yes"` xrandr lists all modes except `4896x2448`. The default resolution is `3680x1836`. The
Vive display shows my desktop, but with the help of `nvidia-settings`  the vive monitor can be turned off. After that, steamvr can be used. In mode 0 the image is still scrambled, but mode 3 works perfectly.

I hope that helps anyone.

mwigzell, 2023-01-03:
Hey guys, I hope you don't mind my chiming in here, I have a very similar situation here with standard 6.1.1-arch1-1 kernel. I tried to apply the latency patches from Santeri, but that kernel froze. I have not yet tried kernel patches from CertainLach but I intend to do so. I am running Nvidia GeForce GTX 1660 GPU. I have very similar if not identical artifact in my video after applying driver(s) from CertainLach. I have issues as follows:

1. HMD display as above (using HTC Vive Pro 2)
2. UI issue with SteamVR which seems to intermittently swallow mouse clicks and display black screen. Maybe is due to the USB port I chose for the Vive Link.
3. Valve Base Station 2.0 have conflicting channel 12. I cannot use UI to resolve the issue (it says they are not broadcasting)
4. Valve Index controllers, so far seem to work. 
5. I was able to upgrade firmware on the HMD and controllers, but so far not the base stations.

Apologies for interrupting this thread, but I was so excited to see same issue with the HMD headset