CertainLach/VivePro2-Linux-Driver #16: Issues with HMD (Vive Pro 2)

mwigzell, 2023-01-03:
HI have a very similar situation (https://github.com/CertainLach/VivePro2-Linux-Driver/issues/15) here with standard 6.1.1-arch1-1 kernel. I tried to apply the latency patches from Santeri, but that kernel froze. I have not yet tried kernel patches from CertainLach but I intend to do so. I am running Nvidia GeForce GTX 1660 GPU. I have very similar if not identical artifact in my video after applying driver(s) from CertainLach. I have issues as follows:

1. HMD display as in the thread I linked above.  (using HTC Vive Pro 2) (Nvidia driver 525.60.11) This issue is as in the image, the right fifth of each eye has vertical rainbow lines. Rest of rendering is wrapped around weirdly.
3. UI issue with SteamVR which seems to intermittently swallow mouse clicks and display black screen. Maybe is due to the USB port I chose for the Vive Link.
4. Valve Base Station 2.0 have conflicting channel 12. I cannot use UI to resolve the issue (it says they are not broadcasting)
5. Valve Index controllers, so far seem to work.
6. I was able to upgrade firmware on the HMD and controllers, but so far not the base stations.

cklarhorst, 2023-01-04:
I updated my comment on #15 to be more clear on how I solved the weirdly rendering. Sadly, I don't have any help for the other problems, sorry.

CertainLach, 2023-01-04:
1 - I think nvidia broke something in newest drivers, can you try with driver 520.56? I have only tested nvidia support on this version, and can't test on others due to lack of hardware
2 - SteamVR issue, unactionable in this driver
3 - You can use windows to reconfigure channels, or this android app: https://play.google.com/store/apps/details?id=com.jeroen1602.lighthouse_pm&hl=en&gl=US. Out of scope of this project
5 - SteamVR doesn't support base station firmware update on linux, you should use windows for this. Out of scope of this project

The only thing this project "fixes"/implements on SteamVR side is base station power management

mwigzell, 2023-01-04:
1. Ok, looks like I'll have to create a separate ARCH boot anyway, It'll take some time to get control of the nvidia driver version 520.56. (I don't want to tamper with my rolling update of latest)
2. Ok, I'll plug the SteamVR forums for this one. I just thought you may know what the hell that is. Something dumb I am doing, or everyone has this issue?
3. I was able to run that Android app, but seems only for power management. It does show me that indeed both channels are on 12. Looks like Valve ships their box of 2 base stations configured with same channel. duh! But the Android app won't set the channel AFAIK
4. Cool
5. Ha, maybe I'll have to buy a windows licence and try to put it in a spare partition. I hope that Windows will allow itself to be installed on a partition these days. It always used to try to overwrite my bootstrap

mwigzell, 2023-01-04:
@cklarhorst  Can you spell that out for me? I saw your mention of nvidia-settings, but don't find what you are referring to in /etc/X11/xorg.conf there is no mention of AllowHMD? 
In fact the Device section only contains comments!

CertainLach, 2023-01-04:
2 - It just happens sometimes. SteamVR has bad support for linux, unfortunately, and bugfixes are rare.
3 - Well, seems like I misremembered it having one. Maybe some CLI utilities for base station management have this? 
Anyway, you can use any btle debugging app (I.e https://play.google.com/store/apps/details?id=com.punchthrough.lightblueexplorer, I'm not affiliated with this one) for this, write desired channel id (< 16) to `000015241212EFDE1523785FEABCD124` characteristic

cklarhorst, 2023-01-04:
@mwigzell Sure, the AllowHMD setting wasn't there for me before either. You have to add that explicitly. Make it look like that:
```
Section "Device"
    Identifier     "Device0"
    Driver         "nvidia"
    VendorName     "NVIDIA Corporation"
    BoardName      "GeForce GTX 1060 3GB"
    BusID          "PCI:9:0:0" #09:
    Screen          0
    Option "AllowHMD" "yes"
EndSection
```

CertainLach, 2026-10-01:
Stale

closed 2026-10-01 by CertainLach