CertainLach/VivePro2-Linux-Driver #18: Cosmos Elite support ?

MonsieurLanza, 2023-01-19:
Hello, do you have an idea what amount of effort it would take to get Cosmos Elite working, using this repo as a basis ?

Should I dig in, or ditch the idea straight away ?

I use Arch, btw. ;)
Thanks.

MonsieurLanza, 2023-01-21:
So far, after some battle with kernel versions and config here and on https://github.com/santeri3700/vive-pro-2-on-linux, I was able to get a patched kernel (with the last patches which includes the cosmos ID \o/), and the vivepro2 driver. 

Patches need another update, they won’t work on the latest `linux-drm-misc-next`, I had to get the 2023-01-12 tag. And the provided config failed to boot on my machine.

Steam now displays the VR icon in the toolbar, and on my nvidia box with the `AllowHMD` option set in xorg conf, I can see the render… on my primary display. So I now know the tracking works, and the render seems somewhat correct, I’m not sure about the dimensions. And the system freezes while exiting SteamVR.

But with or without the `AllowHMD` flag, nothing renders on HMD itself.  My guess because the vivepro2 driver does not set the mode properly (only vivepro2 modes seems available), but I now have to find a way to get logs.

CertainLach, 2023-01-21:
For cosmos support, you should at least
1. Set your modes here: https://github.com/CertainLach/VivePro2-Linux-Driver/blob/master/crates/vive-hid/src/lib.rs#L169 (Only one mode is available on cosmos)
2. Disable every writing method in vive-hid crate (Because cosmos has other protocols)

CertainLach, 2023-01-21:
Set of changes, that was applied on v0.3.1-cosmos tag:
https://gist.github.com/CertainLach/888c4958678bee5f1736bc2ac3a5095d

MonsieurLanza, 2023-01-21:
Ok, I had correct modes, and product ID. I did not remove any of the writing, though.

CertainLach, 2023-01-21:
You should remove writings, as cosmos has slightly different protocol, which also lacks some of the vive pro 2 features (I.e mode switching)

MonsieurLanza, 2023-01-22:
Well, stuck on "Permission denied". Maybe stupid, but still…

```bash
[31mERROR[0m [2mdriver_proxy::server::driver_host[0m[2m:[0m failed to p hmd: hid error: hid error: hidapi error: Permission denied
```

[cosmos-log.txt](https://github.com/CertainLach/VivePro2-Linux-Driver/files/10473443/cosmos-log.txt)

CertainLach, 2023-01-22:
chmod a+rw /dev/hidraw*

There is no udev rules for vive cosmos installed

MonsieurLanza, 2023-01-22:
Thanks again.

So it does not work out of the box. No image, and cosmos is led still red.

At least it does not seem to crash all system on exit. Note : station are not going to standby mode, despite driver saying so.

[cosmos-log.txt](https://github.com/CertainLach/VivePro2-Linux-Driver/files/10474154/cosmos-log.txt)

CertainLach, 2023-01-22:
To make stations sleep, you should enable power management for them, driver tells its going to put only managed ones to sleep.

MonsieurLanza, 2023-01-22:
Ok. 

Now, for what I understand, there is probably an « usb switch » to turn on display. 

OpenHMD does this by sending a magic sequence https://github.com/OpenHMD/OpenHMD/blob/master/src/drv_htc_vive/vive.c#L594 , and as the sequence is different for Vive an Vive Pro, I guess it is different for Cosmos. And I guess on Vive Pro 2 setting the display mode does turn the display on. 

If I understand correctly, I just have to find that value in the middle if the flow of USB traffic I just wiresharked, write it and see if something happens.

CertainLach, 2023-01-22:
For Vive Pro 2 nothing is required, the display just starts working as soon as it receives the DisplayPort signal.
There are commands for led control, but they are not mandatory to start displaying an image.

I didn't tough about Vive Cosmos requiring it, but this may be the case

CertainLach, 2023-01-22:
I have looked in EyesReader.dll, and seems like there is some function for managing HMD panel.
Is I can see, report should have 0x782938 as its 1-3 bytes (counting from zero) as part of `HMD_SetFeatures` call, then should contain 0x80 and power state (1 byte) somewhere in data (Too lazy to count bytes, this report is created in `SetPowerState` function), and rest is filled with zeros.

CertainLach, 2023-01-22:
I think you should be able to easly recover all of the protocol messages by using api monitor (http://www.rohitab.com/apimonitor), and capturing HidD_SetFeature calls from the `EyesReader.dll` library, every function there has self-explaining name

MonsieurLanza, 2023-01-22:
Thank you for all that, I’ll have a deeper look when I can get some free time again.

MonsieurLanza, 2023-01-23:
[image]

PanelOnOff \o/

MonsieurLanza, 2023-01-25:
So far I learned a lot, and have a better idea of what I need to do, which is basically write another `ViveDevice` for cosmos, there is too much difference with the vivepro2 one for it to make sense.

I have to find larger chunks of time, as I need to (re)learn things, dust off my little rust knowledge, go deeper in HID, Windows tooling…

MonsieurLanza, 2023-01-27:
@CertainLach do you have the vivepro 2 hid descriptors, so that I can compare ? 

I’m not getting why with the cosmos I get 2 Hid devices with the right signature, and they do have slightly different hid descriptors. HTC’s drivers seem to "talk" to the first one they find, but I’m not certain.

And did I mention that I HATE little endian ? I want my MC68000 amiga back.

CertainLach, 2023-01-27:
> I’m not getting why with the cosmos I get 2 Hid devices with the right signature, and they do have slightly different hid descriptors. HTC’s drivers seem to "talk" to the first one they find, but I’m not certain.

Its the same for Vive Pro 2, one device is for lighthouse driver, and its implements valve protocols (`SteamDevice` struct in this driver), and then there is Vive device itself, which implements HTC protocols (`ViveDevice` struct)

Driver only needs to talk to ViveDevice, SteamDevice is compatible with lighthouse driver, and will work by itself.

There is nothing useful in HID descriptors

CertainLach, 2023-01-27:
Writing captured data didn't work?

MonsieurLanza, 2023-01-27:
I really have 2 ˋViveDevice` "0x0bb4:0x0313" signature, along with 2 `SteamDevice` for lighthouse tracking, when doing a device enumeration.

CertainLach, 2023-01-27:
That's weird, are you sure they are both active on the same time?
That may be possible with SteamDevice, as they are differentiated by serial number, but ViveDevice?..

If there is multiple ViveDevices - check their serials, and use the correct one with `open` method, instead of just calling `open_first`

<details>
<summary>Ok, there is my descriptors.</summary>

ViveDevice:
```
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x09, 0x01,        //   Usage (0x01)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0x81, 0x82,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x85, 0x01,        //   Report ID (1)
0x09, 0x01,        //   Usage (0x01)
0x91, 0x82,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x02,        //   Report ID (2)
0x09, 0x02,        //   Usage (0x02)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0x81, 0x82,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x85, 0x02,        //   Report ID (2)
0x09, 0x02,        //   Usage (0x02)
0x91, 0x82,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x03,        //   Report ID (3)
0x09, 0x00,        //   Usage (0x00)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0x81, 0x82,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x85, 0x04,        //   Report ID (4)
0x09, 0x00,        //   Usage (0x00)
0xB1, 0x82,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0xC0,              // End Collection

// 71 bytes
```

SteamDevice:
```
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0xA1, 0x01,        //   Collection (Application)
0x95, 0x33,        //     Report Count (51)
0x85, 0x20,        //     Report ID (32)
0x09, 0x01,        //     Usage (0x01)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xA1, 0x01,        //   Collection (Application)
0x95, 0x02,        //     Report Count (2)
0x85, 0x01,        //     Report ID (1)
0x09, 0x01,        //     Usage (0x01)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
0xA1, 0x01,        //   Collection (Application)
0x95, 0x2E,        //     Report Count (46)
[... 71 lines cut ...]
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0xA1, 0x01,        //   Collection (Application)
0x95, 0x02,        //     Report Count (2)
0x85, 0x10,        //     Report ID (16)
0x09, 0x01,        //     Usage (0x01)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
0xA1, 0x01,        //   Collection (Application)
0x95, 0x3F,        //     Report Count (63)
0x85, 0x11,        //     Report ID (17)
0x09, 0x01,        //     Usage (0x01)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
0xA1, 0x01,        //   Collection (Application)
0x95, 0x01,        //     Report Count (1)
0x85, 0x12,        //     Report ID (18)
0x09, 0x01,        //     Usage (0x01)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
0xC0,              // End Collection

// 209 bytes
```
</details>

MonsieurLanza, 2023-01-28:
SteamDevice looks the same.

<details>
<summary>Cosmos one is way simpler</summary>
<pre>
# HTC Corporation VIVE COSMOS
# 0x06, 0x00, 0xff,              // Usage Page (Vendor Defined Page 1)  0
# 0x09, 0x01,                    // Usage (Vendor Usage 1)              3
# 0xa1, 0x01,                    // Collection (Application)            5
# 0x15, 0x00,                    //  Logical Minimum (0)                7
# 0x26, 0xff, 0x00,              //  Logical Maximum (255)              9
# 0x75, 0x08,                    //  Report Size (8)                    12
# 0x96, 0x00, 0x04,              //  Report Count (1024)                14
# 0x09, 0x01,                    //  Usage (Vendor Usage 1)             17
# 0x81, 0x02,                    //  Input (Data,Var,Abs)               19
# 0x95, 0x40,                    //  Report Count (64)                  21
# 0x09, 0x01,                    //  Usage (Vendor Usage 1)             23
# 0xb1, 0x02,                    //  Feature (Data,Var,Abs)             25
# 0xc0,                          // End Collection                      27
# 
R: 28 06 00 ff 09 01 a1 01 15 00 26 ff 00 75 08 96 00 04 09 01 81 02 95 40 09 01 b1 02 c0
N: HTC Corporation VIVE COSMOS
I: 3 0bb4 0313
</pre>
</details>
Not even a feature report ID.

Both `/dev/hidraw0` and `/dev/hidraw1` reports the same, and `usbhid-dump` reports 2 devices for 
<details>
<summary>`$ usbhid-dump -d 0bb4:0313`</summary>
<pre>
004:004:001:DESCRIPTOR         1674908243.867237
 06 00 FF 09 01 A1 01 15 00 26 FF 00 75 08 96 00
 04 09 01 81 02 95 40 09 01 B1 02 C0

004:004:000:DESCRIPTOR         1674908243.867394
 06 00 FF 09 01 A1 01 15 00 26 FF 00 75 08 96 00
 04 09 01 81 02 95 40 09 01 B1 02 C0
</pre>
</details>

However `lsusb` reports only one Vive cosmos.

MonsieurLanza, 2023-01-28:
Ok, understood one thing. Everything go through the feature report on Cosmos, even the `read_config`. They could have used http instead of hid without much difference :D

MonsieurLanza, 2023-01-30:
> [image]
> 
> PanelOnOff \o/

The values here are… pointers! Which make little sense at first sight.

CertainLach, 2023-01-30:
There are missing memzeros/wrong array sizes used, it is possible that this is stack values leaked to a SetFeature call.
Example (pseudocode):
```
char report[10];
int somevalue;
report[0] = 0;
...
SetFeature(&report) // where report is expected to be a char[65 - header], so value of somevalue is also included in report value
```

This is fairly common in Vive Pro 2 driver too

MonsieurLanza, 2023-01-30:
I see, this is not rust code ^^. 

Thing is, the start of the report seems invariable whether it is an On or an Off call. There is something I miss here.

BTW, the config file does not fit at all inside the Vive Pro 2 `structs` : https://gist.github.com/MonsieurLanza/583613475093b716f65c32260f8c364a

JJones780, 2023-02-16:
I've got a regular Vive headset and a Cosmos Elite.. trying to get the Cosmos going.
I've recompiled my kernel etc
and followed through on CertainLatch's driver with Cosmos modifications, compiled and installed.
I'm now at the same stage as @MonsieurLanza where I the headset display isn't coming on.

I do get a tracked view when I turn on SteamVR's "Display VR View" which makes it look like it's sooo close to working.

My controllers won't connect/ track though - but work fine when I plug them into the computer with a usb cable.
I also noticed that the single trigger button on the headset won't work ( when trying to cancel a VR startup notification without the vr controllers).  
I have already tried  sudo chmod a+rw  /dev/hidraw*  (and on /dev/usb/hiddev* as well).

p.s. @MonsieurLanza - yeah! Amiga/68k !! ;)

MonsieurLanza, 2023-02-16:
@JJones780 I have made progress on the PR here https://github.com/CertainLach/VivePro2-Linux-Driver/pull/20 . It currently dumps the cosmos config file in the logs. I have some hid dump from windows too, and have to make this in.

It’s close but not so close. The tracking part it similar to other vive/valve lightouse products and work out of the box, but the init of the display is completely different. So the `vive_hid` lib needs specific code, maybe different config struct/handling, as does the hmd init part in `driver_proxy`.

I’ve been away because of work, but ss I plan to use the cosmos to make things with Godot, and given the fact I cannot stand windows except to launch SteamVR from time to time,  I’ll get back at it soon.

Nvidia drivers may part of the problem, I’d like to check with an AMD GPU, but I need to dig in boxes, as it is packed during house renovation.

(This little endian thing makes it stupidly complex to write bytes in memory. I will never understand.)

JJones780, 2023-02-16:
I tried accessing the display via i2c and other methods but also ran into a nvidia bug.  I don't have windows installed at all. I am happy to see I am able to switch between the Original HTC Vive and Cosmos headsets with only the driver file change ( driver_lighthouse_real.so ) which means I'll be able to keep up on the cosmos hacking.

JJones780, 2023-04-29:
@MonsieurLanza  
I saw your USB reports with the HMDPanelOn/Off .... what about an HMDOn/Off or some other overall On/Off or sleep wake command? 
It seems more than just the LCD panel is off.... 
I noticed that, like the display,  the cameras are detected but not enabled.
Also, the HMD's bluetooth is off?  The controllers work via usb but not the HMD's bluetooth.

JJones780, 2023-05-03:
I opened up the system.img file for the headset ( yes it's Android but I'm betting not connected to the display panels other than turning them or maybe only their backlights off and on ) . It provides ADB, HID and UVC  but I don't know of any way to switch it to ADB to connect to the running headset.  Looking through the system.img files to see if I can find any hints as to the hardware/startup/hid settings.  If anyone has ideas to look into please let me know. 
The best bet is probably still the capture of USB HID traffic from a windows machine.

MonsieurLanza, 2023-06-01:
I am sorry I have not been around for some monthes, life has surprises… And I won’t be able to get something done until july, or august. No more room for that during house renovation.

I do not have the problems with controllers/bluetooth you mention. For me, everything works fine, except display. Tracking of the headset and the controllers works, I am able to see it by enabling SteamVR view.

I do have USB hid dumps and API calls. But to get further in, and really try to reproduce the HID exchanges (which are not really HID, but anyway), I have to get the driver to accept the cosmos configuration file, which is quite different from the vive pro 2 one.

Kiuhrly, 2023-09-28:
hi, I'm very interested in getting this working (VR is the only reason I ever have to boot to windows...) so before I start hacking at it, has any progress been made here or elsewhere which hasn't been captured in this thread?

JJones780, 2023-09-30:
Sorry, nothing new to report from me.
Windows USB "feature report" traffic trace is needed.   I was hoping @MonsieurLanza would be finished his house renovations but I know those easily go overtime ;)  
Maybe I'll try digging into the Android code that runs on the headset again and hope the Monsieur appears again soon!

Kiuhrly, 2023-10-07:
In the end I just bought a cheap second hand Vive Pro which isn't really any worse in terms of specs (and it has OLED!) and is supported on linux (in theory). In practice I also haven't been able to get it to work properly because of <https://github.com/ValveSoftware/SteamVR-for-Linux/issues/175>... so that's annoying.

crystalclaw, 2023-12-17:
I've got a vive cosmos and both a Linux and windows partition, I'd be happy to provide a feature report but I can't find good documentation on how to go about that. I've used Wireshark for stuff like that before, but I've also run into it missing critical things so I'm not sure if that would be sufficient

JJones780, 2023-12-25:
Anything you can dig up would be handy.  I've still got the Cosmos sitting in storage and would like to mess with it... and hacking out a solution might be fun when time frees up.  I've been meaning to dig through the Cosmos Elite Android ISO some more but getting numbers from the USB communication could help with that too. @MonsieurLanza - did you make those USBHid Setfeature etc  log files accessable here somewhere?

BlueMoonShadow620, 2024-01-02:
Hello, this gonna be a bit off topic here (sorry about that), but im here in search of HTC vive cosmos elite config.json file contence i was modifying some stuff and messed up and don't have a back up. Could i ask for that file here?

JJones780, 2024-01-03:
It's getting a little bit harder to find people running windows?  ;) 
Have you tried on the discord?  https://discord.gg/viverse
The community managers are welcoming - They tried to pass along my request for info  to the engineering dept re: Cosmos Elite on Linux but haven't heard anything back.  An old product on an unsupported platform - so not entirely unexpected.

BlueMoonShadow620, 2024-01-03:
Yeah i have tried the group ill be honest i have checked all corners of internet for this file and now im loosing hope i saw that people here have the cosmos and are not afraid of programing so feels like last dich option to recover a "working" headset with a dead config file

CertainLach, 2024-01-03:
Config files differ between headset, as they contain lens distortion parameters, which are different from unit to unit.
Those parameters are also very precise, and it might even damage your eyes if you try to use the headset with wrong config for a prolonged period of time.

It’s better to try sending it in for repair, because in this case they will be able to calculate the correct parameters for your lenses and assemble the correct config.
They know how to do this because this is what happens when they replace lenses.

BlueMoonShadow620, 2024-01-03:
Found out these files are kept by steam as back ups (snapshot of configs) in a different folder managed to get it out and recover my headset. Vive support actually responded in the viverse after 5 days of silence. so if anything happens to your vr's configs you can find the org files here if OS was not reinstalled : C:\Program Files (x86)\Steam\config\lighthouse
Also @CertainLach yes they differ a bit people say that its not too much i cant say for long term tho they tried it on OG htc and pro

> It's getting a little bit harder to find people running windows? ;) Have you tried on the discord? https://discord.gg/viverse The community managers are welcoming - They tried to pass along my [...]

CertainLach, 2024-01-04:
> its not too much

Well, in Vive Pro 2, they liked to experiment with the different lens configurations, and thus lens distortion driver supports lots of distortion types, and combinations of them. They can even make lens from the two different lenses glued together, and lens distortion config still fits that configuration due to its complexity.

Maybe the lens construction is more uniform for cosmos, but its config format still allows for wild configurations.

MonsieurLanza, 2024-01-10:
Sorry for the silence, quite a bit happened here, which kept me away from computers. I do (did) have config.json and other bits from cosmos, I’ll try to upload them somewhere when I can.

WindowsLonestar, 2024-04-11:
Dang, I was getting really hopeful at the end. Someday I hope to be able to ditch windows, but today is not the day.

LuxLucian, 2024-11-13:
I recently got a cosmos elite, I could try to help with testing on Arch(EndeavorOS) soon. I'll send an update once I get the kernel patches and drivers installed.

LuxLucian, 2024-12-13:
I'm running EndeavorOS and I also have a windows install working, is there anything I can do to help out with this issue? Having to boot up windows for VR is a pain.

JJones780, 2024-12-16:
> I'm running EndeavorOS and I also have a windows install working, is there anything I can do to help out with this issue? Having to boot up windows for VR is a pain.
If you want to try using wireshark to sniff the usb traffic ( or just use https://desowin.org/usbpcap/ ) when running in windows... just until the HMD display turns on ... I'd be interested in having a look.

LuxLucian, 2025-02-11:
> > I'm running EndeavorOS and I also have a windows install working, is there anything I can do to help out with this issue? Having to boot up windows for VR is a pain.
> 
> If you want to try using wireshark to sniff the usb traffic ( or just use https://desowin.org/usbpcap/ ) when running in windows... just until the HMD display turns on ... I'd be interested in having a look at the data.

https://mega.nz/folder/FRVzgRIK#MHem3F1REpuCMT_m0kM7eQ

The Cosmos1 file has me turning the link box on, then starting VR via steam, then exiting VR through steam, then turning link box back off.
Cosmos 2 has the linkbox already on throughout the capture, and I start and exit VR via steam
Github did not let me upload the files so hopefully the megalink is good enough.

JJones780, 2025-03-07:
I had a quick look 2 weeks ago but haven't gotten back to it. I'm interested in sharpening up my skills in this area but in the meantime I'm rushing to get out of the house I just sold and have moved on to a new headset recently so can't make any promises.

Ay1tsMe, 2025-05-30:
I'm in the process of getting my environment setup for troubleshooting. I got windows installed on my system and have captured the USB traffic with wireshark when the hmd turns on but i don't know what I'm looking for to get it to work.

I also tried installing the drivers from https://github.com/CertainLach/VivePro2-Linux-Driver/pull/20 but I think the PR is too old because when I try to build automically using this command it fails:
```
# Note: the #driver-proxy-release part of this command is not a comment, it's just github syntax highlighter is wrong
nix build --extra-experimental-features nix-command --extra-experimental-features flakes .#driver-proxy-release
```
 And manually using these instructions https://github.com/santeri3700/vive-pro-2-on-linux/blob/master/STEAMVR.md , I don't have a `driver_lighthouse.so` file to copy.

I also tried patching my kernal from this repo using these instructions but that also failed to build as well https://github.com/santeri3700/vive-pro-2-on-linux/blob/master/KERNEL.md

I'm gonna try install the drivers again today but if I can't get it installed on my system, then it would be a dead end for me. I'd be able to help with logging the headset through windows but someone would need to edit the driver code.

Ay1tsMe, 2025-06-12:
I've made some progress in this PR: https://github.com/Ay1tsMe/VivePro2-Linux-Driver

I was able to compile the drivers and copy the code that @MonsieurLanza had written in his PR and now have gotten to this stage. All devices seem to show up in steamvr but it gets stuck on Waiting... and it says failed to connect to headset display. Below are the vrserver.txt logs. It seems to be failing on these error lines:
```
Thu Jun 12 2025 19:42:25.299738 [Info] - lighthouse: ERROR GetProjectionRaw{eEye=Eye_Left pfLeft=0x7f5f1d8c00cc pfRight=0x7f5f1d8c00d0 pfTop=0x7f5f1d8c00d4 pfBottom=0x7f5f1d8c00d8}: driver_proxy::driver::hmd: failed: lens error: io error: failed to fill whole buffer
Thu Jun 12 2025 19:42:25.299806 [Info] - lighthouse: ERROR GetProjectionRaw{eEye=Eye_Right pfLeft=0x7f5f1d8c0140 pfRight=0x7f5f1d8c0144 pfTop=0x7f5f1d8c0148 pfBottom=0x7f5f1d8c014c}: driver_proxy::driver::hmd: failed: lens error: io error: Broken pipe (os error 32)
```

[vrserver.txt](https://github.com/user-attachments/files/20708374/vrserver.txt)

[image]

Maybe @CertainLach would know a bit more about what would be causing this error and how to fix it?

voxel-public, 2025-07-19:
> if I undertsood correctly we need someone that would be able to reverse engineer on windows what initialize this using WireShark? My knowledge on this field is quite lacking, but if there's anything that can be provided from Linux I would happily oblige.

That was my hope too. My understanding was that there are 'magic' sequences to power on the display, sent over usb HID, as per https://github.com/OpenHMD/OpenHMD/blob/85075b0c7e3c723ded2577edb79d00ee11aac339/src/drv_htc_vive/magic.h#L12

In my post with usb traffic dumps attached https://github.com/CertainLach/VivePro2-Linux-Driver/issues/18#issuecomment-2961096738 there are some similar sequences, but whether or not they're actually doing anything related to enabling the display on the cosmos elite is beyond my ability to test.

LunyaticCat, 2025-07-19:
> [@LunyaticCat](https://github.com/LunyaticCat) where you able to get "movement" when using this PR https://github.com/Ay1tsMe/VivePro2-Linux-Driver ? For example, on the VR View you can see that the headset and controllers are able to move but no video is being sent to the display.
> 
> I still get Failed to connect to headset display and I see no movement on my VR view. Ive sort of hit a roadblock. [@CertainLach](https://github.com/CertainLach) suggests a proton issue but I wasn't able to get it to work with the fix. Ill make another crack at it when I get more time.

Well, I do get an error at some point, but here is a recording of what I get : 
https://lunaticcat.me/assets/videos/vr_test.mp4

Ay1tsMe, 2025-07-19:
Hmm interesting. I was never able to get to that stage so I assumed it was the code but this is more reassuring. I'll have to work out how to get my proton working. I am using arch with KDE Wayland so maybe Wayland is the issue? I'll have to look into it more.

Once I get to your stage I'll try figure out how to turn on the display.

LunyaticCat, 2025-07-19:
> Hmm interesting. I was never able to get to that stage so I assumed it was the code but this is more reassuring. I'll have to work out how to get my proton working. I am using arch with KDE Wayland [...]

I can't say for sure Wayland is the only issue, but from what I read there's currently no way to run it with Wayland, I do get a black output similar to yours if I run steamVR using it.

LuxLucian, 2026-02-02:
Any interest in this project still? I did grab my headset for a decent price and I've been thinking of selling it since the IPD is a bit too high for me.

JJones780, 2026-02-20:
I kept one base station with the idea of maybe looking into this again or at least having a working VR kit to sell/give away.  Moved a few times and dropped the ball. 
My last notes involved looking at the executable for the headset I believe - maybe it'll give someone else an idea? 
I had run the Vive Installer using wine and put it in the ViveCosmosElite dir:

```
cd "ViveCosmosElite/drive_c/Program Files (x86)/VIVE/Updater/App/update_tool/ROM/HMD"

simg2img oem.img Cosmos_uncompressed_oem.img
sudo mount -o loop Cosmos_uncompressed_oem.img vc/

it looks like oem has the goods!?

other notes:
readelf -a vc/bin/dp

readelf -a -W vc/bin/horusd      <-- lots here

FAIL (power=%d) 
CMD: HORUS_CMD_WORKSTATE, VALUE: %d -> %d
workstate : SUSPEND (%d) -> READY (%d) : auto power on HMD
power status is already off, ignore workstate change
CMD: HORUS_CMD_UVC, VALUE: %d
CMD: HORUS_CMD_VIDEO_STATUS_UPDATE
```

animatek38, 2026-03-08:
> > > hey, not sure about the prerequisite I checked while trying to make it work, but along the way and by feeding it the activation part of htc_vive_cosmos_elite_activation_deactivation.txt sent by [@voxel-public](https://github.com/voxel-public) , I managed to turn the lenses on, yipee
> > 
> > 
> > This is exciting! Can you describe how you 'feed' the usb hid commands to the headset? Is this something we can try to reproduce?
> 
> quite literally sent your file to the headset. I also followed aadrick guide written above, but I'm still struggling with wayland right now. if someone is on an OS allowing the usage of x11, I think they should be able to use their headset 🤔
> 
> import hid
> import time
> import re
> import sys
> 
> VID = 0x0bb4
> PID_CONTROL = 0x0313
> 
> 
> def send_payload(interface_h, payload):
>     report = bytearray([0x00]) + payload
> 
>     if len(report) < 65:
>         report += bytearray(65 - len(report))
>     elif len(report) > 65:
>         report = report[:65]
> 
>     for _ in range(5):
>         if interface_h.send_feature_report(report) >= 0:
>             time.sleep(0.02)
>             return True
>     return False
> 
> 
> def main():
>     file_name = "htc_vive_cosmos_elite_activation_deactivation.txt"
>     try:
>         with open(file_name, "r") as f:
>             pcap_text = f.read()
>     except FileNotFoundError:
>         print(f"Error: Could not find '{file_name}'.")
>         sys.exit(1)
> 
>     pattern = re.compile(r'bRequest: SET_REPORT \(0x09\).*?wIndex: (\d+).*?Data Fragment: ([0-9a-fA-F]+)', re.DOTALL)
>     matches = pattern.findall(pcap_text)
> 
>     raw_sequence = [(int(w), bytes.fromhex(d)) for w, d in matches]
>     activation_sequence = raw_sequence[:106]
>     deactivation_sequence = raw_sequence[106:]
> 
>     devices_found = hid.enumerate(VID, PID_CONTROL)
>     if not devices_found:
>         print("No HTC Vive Cosmos Elite devices found.")
>         sys.exit(1)
> 
>     interfaces = {}
>     for dev in devices_found:
>         if 'interface_number' in dev:
>             iface_num = dev['interface_number']
>             try:
>                 h = hid.device()
>                 h.open_path(dev['path'])
>                 h.set_nonblocking(1)
>                 interfaces[iface_num] = h
>             except Exception:
>                 pass
> 
>     if not interfaces:
>         print("Could not hook into the headset interfaces.")
>         sys.exit(1)
> 
>     print("\nSending activation sequence...")
> 
>     for interface_num, payload in activation_sequence:
>         if interface_num in interfaces:
>             send_payload(interfaces[interface_num], payload)
> 
>     print("Headset is awake")
> 
>     input("\nPress Enter to send Deactivation Sequence and turn off...\n")
> 
>     print("\nShutting down headset...")
> 
>     for interface_num, payload in deactivation_sequence:
>         if interface_num in interfaces:
>             send_payload(interfaces[interface_num], payload)
> 
>     for h in interfaces.values():
>         h.close()
> 
> 
> if __name__ == "__main__":
>     main()

I just tried it on hyprland and got my headset to show my desktop

LunyaticCat, 2026-03-10:
> would this work for the vive cosmos? 
> i have a vive cosmos and would be willing to try this but from what i can tell, is that this is for the vive cosmos elite.
> (would appreciate some help if possible, i would say i'm still quite new to linux - i know how to do some stuff)
> also have you managed to get this working with playing any vr games yet?
> 
> my guess is that there would have to be some changes but it would be mostly the same? I'm not sure but I'm also not sure where to start with this.
> i do also have a windows 10 partition on my laptop for vr as i haven't found much about support for the vive cosmos(not elite) on linux.

Iirc the regular cosmos is quite different, using the cameras instead of base station.
But I think some tools already have it supported on Linux

[1 more reply]