CertainLach/VivePro2-Linux-Driver #22: Sewer rules need update after steamvr 1.26

MilheiroSantos, 2023-07-28:
After updating to steamvr 1.26, I'm no longer able to run the proxy driver. I suspect it's the sewer rules.

Here is the output when I try to run the install:
```
= Moving original driver
= Patching real driver
# Driver requirement check (before 1.25.7)
source pattern not found, file is already patched?
# Driver requirement check (before 1.26.1)
source pattern not found, file is already patched?
# Driver requirement check
source pattern not found, file is already patched?
one or more rules failed
= Overriding current driver
= Updating proxy server
= Testing proxy server
Wine found at wine
```

Here's part of the steamVR output with the error I'm getting: 
```
 INFO driver_proxy::server::driver_context: get generic interface "IVRIOBuffer_002"
 INFO driver_proxy::server::driver_host: added tracked device: "LHR-A292DEE3" (TrackedDeviceClass_HMD)
 INFO lens_client: using lens server at "/home/HOME/.steam/debian-installation/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/lens-server/lens-server.exe"
 INFO lens_client: trying wine64 as wine
 INFO lens_client: trying wine as wine
 INFO lens_server: hello from lens server
 INFO lens_server: dll path: "Z:\\home\\HOME\\.steam\\debian-installation\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bi
n\\linux64\\lens-server\\LibLensDistortion.dll"
 INFO lens_server: set config
reaping pid: 21549 -- IPC:CSteamEngin
 INFO driver_proxy::driver::server_tracked_provider: disconnecting from base stations
ERROR lens_server: error=failed to read request
Error: failed to read request
```

Can you double check on your side?

CertainLach, 2023-07-28:
Yes, I'm aware of that, will update rules this weekend

MilheiroSantos, 2023-07-31:
Thank you for your work. In the meantime, I can share my workaround to this issue:

Start Steam with console support; 
```bash
$ steam steam://open/console
```

In the steam console, download the last known working configuration with

```bash
$ download_depot 250820 250823  2218622420329124205
```

Copy the old driver into your steamVR installation

```bash
$ cp $STEAM_PATH/steamapps/content/app_250820/depot_250823/drivers/lighthouse/bin/linux64/driver_lighthouse.so  $STEAM_PATH/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/driver_lighthouse.so
```

Re-run the install script

closed 2023-09-09 by CertainLach