CertainLach/VivePro2-Linux-Driver #23: Failed to connect to headset display.

Yersi88, 2023-09-15:
After installing the driver, I get the following message when I try to start steamvr: (Failed to connect to headset display) 
[image]

The LED on the headset remains red.

CertainLach, 2023-09-15:
1. Please switch to a stable release 1.27 of steamvr, it is newer than the beta 1.27.5.
2. Which compositor do you use? It is running on Xorg/wayland?

Yersi88, 2023-09-15:
Hello, Thank you for the reply! 

1. I switched to stable release, behaviour is the same. 
[image]

2. KDE Plasma on Wayland

CertainLach, 2023-09-19:
1. Which version of kernel do you have? (6.3+ is required, for lower versions you need to recompile kernel with patches in this repo)
2. Which GPU do you use?
3. Can you share logs of vrcompositor? `$HOME/.local/share/Steam/logs/vrcompositor.txt`
4. Have you approved root access to SteamVR first time it was launched? (It asks for it to run setcap SYS_CAP_NICE+ep, and this may not work well in some setups)

Yersi88, 2023-09-19:
Hello again, many thanks for the reply. 

1. 6.5.3-zen1-1-zen 
2. AMD ATI Radeon RX 7900 XTX (Recently upgraded from Nvidia 2080 ti, not sure if that's of any significance, but it's the same system/install) 
3. See downstairs, last line might be of significance: `Tue Sep 19 2023 19:01:02.442920 [Info] - Failed to start compositor: VRInitError_Compositor_CannotDRMLeaseDisplay`
4. Yes 

### vrcompositor.txt: 

```
Tue Sep 19 2023 19:00:39.757121 [Info] - //==============================================================================================
Tue Sep 19 2023 19:00:39.757144 [Info] - ================================================================================================
Tue Sep 19 2023 19:00:39.757156 [Info] - ================================================================================================
Tue Sep 19 2023 19:00:39.757169 [Info] - vrcompositor.real 1.27.5 startup with PID=123515, config=/home/usr/.local/share/Steam/config, runtime=/home/usr/.local/share/Steam/steamapps/common/SteamVR
Tue Sep 19 2023 19:00:39.757179 [Info] - VR compositor 1.27.5 (v1693508082) Mixed starting up
Tue Sep 19 2023 19:00:39.779347 [Info] - CIPCPipe::ConnectPipe(SteamVR_Namespace) attempting connect to /steamvr/SteamVR_Namespace
Tue Sep 19 2023 19:00:39.779385 [Info] - CSharedResourceNamespaceClient::Init(): received namespace data 123347
Tue Sep 19 2023 19:00:39.779404 [Info] - CIPCPipe::CreatePipe(VR_CompositorPipe_123347) bound to /steamvr/VR_CompositorPipe_123347
Tue Sep 19 2023 19:00:39.780018 [Info] - Set fd limit to: 524288
Tue Sep 19 2023 19:00:39.780611 [Info] - [Settings] Load Default Json Settings from /home/usr/.local/share/Steam/steamapps/common/SteamVR/drivers/htc/resources/settings/default.vrsettings
Tue Sep 19 2023 19:00:39.780646 [Info] - [Settings] Load Default Json Settings from /home/usr/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/resources/settings/default.vrsettings
Tue Sep 19 2023 19:00:39.780674 [Info] - [Settings] Load Default Json Settings from /home/usr/.local/share/Steam/steamapps/common/SteamVR/drivers/null/resources/settings/default.vrsettings
Tue Sep 19 2023 19:00:39.780777 [Info] - [Settings] Load Default Json Settings from /home/usr/.local/share/Steam/steamapps/common/SteamVR/resources/settings/default.vrsettings
Tue Sep 19 2023 19:00:39.780971 [Info] - [Settings] Load Json Settings from /home/usr/.local/share/Steam/config/steamvr.vrsettings
Tue Sep 19 2023 19:00:39.781623 [Info] - Events created
Tue Sep 19 2023 19:00:39.782044 [Info] - Found bad mirror window settings: 
Tue Sep 19 2023 19:00:39.782057 [Info] - Creating CHmdWindowSDL!
Tue Sep 19 2023 19:00:39.782088 [Info] - CHmdWindowSDL: Using Wayland
Tue Sep 19 2023 19:00:39.797248 [Info] - WaylandHMDState: Got interface: wp_drm_lease_device_v1
Tue Sep 19 2023 19:00:39.797410 [Info] - WaylandHMDLeaseDevice::LeaseDevice_DrmFD: /dev/dri/card0
Tue Sep 19 2023 19:00:39.797427 [Info] - WaylandHMDLeaseConnector::LeaseConnector_Done
Tue Sep 19 2023 19:00:39.800590 [Info] - WaylandHMDLeaseDevice::LeaseDevice_Done
Tue Sep 19 2023 19:00:39.804662 [Info] - Instance layers available:
Tue Sep 19 2023 19:00:39.804679 [Info] -  - VK_LAYER_ALVR_capture, version 4194372
Tue Sep 19 2023 19:00:39.805181 [Info] - Instance extensions available:
Tue Sep 19 2023 19:00:39.805195 [Info] -  - VK_KHR_device_group_creation, version 1
Tue Sep 19 2023 19:00:39.805204 [Info] -  - VK_KHR_display, version 23
Tue Sep 19 2023 19:00:39.805213 [Info] -  - VK_KHR_external_fence_capabilities, version 1
Tue Sep 19 2023 19:00:39.805222 [Info] -  - VK_KHR_external_memory_capabilities, version 1
Tue Sep 19 2023 19:00:39.805230 [Info] -  - VK_KHR_external_semaphore_capabilities, version 1
Tue Sep 19 2023 19:00:39.805239 [Info] -  - VK_KHR_get_display_properties2, version 1
Tue Sep 19 2023 19:00:39.805247 [Info] -  - VK_KHR_get_physical_device_properties2, version 2
Tue Sep 19 2023 19:00:39.805256 [Info] -  - VK_KHR_get_surface_capabilities2, version 1
Tue Sep 19 2023 19:00:39.805264 [Info] -  - VK_KHR_surface, version 25
Tue Sep 19 2023 19:00:39.805273 [Info] -  - VK_KHR_surface_protected_capabilities, version 1
Tue Sep 19 2023 19:00:39.805281 [Info] -  - VK_KHR_wayland_surface, version 6
Tue Sep 19 2023 19:00:39.805290 [Info] -  - VK_KHR_xcb_surface, version 6
Tue Sep 19 2023 19:00:39.805299 [Info] -  - VK_KHR_xlib_surface, version 6
Tue Sep 19 2023 19:00:39.805307 [Info] -  - VK_EXT_acquire_drm_display, version 1
Tue Sep 19 2023 19:00:39.805316 [Info] -  - VK_EXT_acquire_xlib_display, version 1
Tue Sep 19 2023 19:00:39.805332 [Info] -  - VK_EXT_debug_report, version 10
Tue Sep 19 2023 19:00:39.805341 [Info] -  - VK_EXT_debug_utils, version 2
Tue Sep 19 2023 19:00:39.805350 [Info] -  - VK_EXT_direct_mode_display, version 1
Tue Sep 19 2023 19:00:39.805358 [Info] -  - VK_EXT_display_surface_counter, version 1
Tue Sep 19 2023 19:00:39.805367 [Info] -  - VK_EXT_surface_maintenance1, version 1
Tue Sep 19 2023 19:00:39.805375 [Info] -  - VK_EXT_swapchain_colorspace, version 4
Tue Sep 19 2023 19:00:39.805384 [Info] -  - VK_KHR_portability_enumeration, version 1
Tue Sep 19 2023 19:00:39.805392 [Info] -  - VK_LUNARG_direct_driver_loading, version 1
Tue Sep 19 2023 19:00:39.805404 [Info] - SDL version: 2.28.3
Tue Sep 19 2023 19:00:39.805415 [Info] - Direct mode features: present
Tue Sep 19 2023 19:00:39.805424 [Info] - Requesting 9 instance extensions:
Tue Sep 19 2023 19:00:39.805434 [Info] -  - VK_KHR_surface
[... cut ...]
Tue Sep 19 2023 19:01:02.413515 [Info] - Tried to find direct display through Wayland: (nil)
Tue Sep 19 2023 19:01:02.413528 [Error] - CHmdWindowSDL: Failed to create direct mode surface
Tue Sep 19 2023 19:01:02.413575 [Error] - CHmdWindowSDL: VR requires direct mode.
Tue Sep 19 2023 19:01:02.413677 [Error] - Error making window!
Tue Sep 19 2023 19:01:02.442770 [Info] - Failed to kill gpu-trace
Tue Sep 19 2023 19:01:02.442883 [Info] - Failed to initialize compositor
Tue Sep 19 2023 19:01:02.442920 [Info] - Failed to start compositor: VRInitError_Compositor_CannotDRMLeaseDisplay

```

CertainLach, 2023-09-19:
It looks like it is failing to find the display:
`Tried to find direct display through Wayland: (nil)`

1. Are you using latest version of the driver (v0.6.0)? On v1.27 version of SteamVR, commit d673d02920c350ba8ef7e50dcc20433b1f9eb5e7 is required.

2. If yes, then try to start stream from shell, to be able to see stderr output of SteamVR, this driver logs everything there. (I think I should redirect this output to file :D)

3. Also, have you configured custom display resolution? `vivepro2.resolution` in `steamvr.vrsettings`. If not - then it defaults to the lowest, and everything should work. If you using 4/5 - then try to lower it to 3, modes 4/5 are special, as they require DSC, and support of DSC implementation of Vive Pro 2 is not quite right in Linux, and patches might still be required.

Yersi88, 2023-09-19:
1. Yes 
2. I noticed some ALVR things in the output  I was using ALVR for pico headset, even tho I uninstalled the driver some stuff I guess remained) 
```
/home/marius/.local/share/Steam/steamapps/common/SteamVR/bin/linux64/vrcompositor
ALVR: drm-lease shim loaded
```

3. No, I have not set the resolution. 

I got it in almost working state by nuking the steamvr folder, verifying files and installing the driver again. 
[image]

# New problem: split vision
No more display error, headset light is green, there is picture in headset and tracking works fine. 

However, when I look inside the image is split in two, it looks like two different pictures one on top of another.

CertainLach, 2023-09-19:
> I noticed some ALVR things in the output I was using ALVR for pico headset, even tho I uninstalled the driver some stuff I guess remained)

Both drivers are somewhat invasive, so you can't have them both at the same time.

> New problem: split vision

This is something in SteamVR itself, I get this thing randomly every once in a while, try to restart.

Yersi88, 2023-09-19:
After restarting, no change. SteamVR also fails the first time I open it after restart, works second time around.

james-luther, 2025-06-21:
@Yersi88, I found this issue while troubleshooting my own configuration (which I'm still working on), and figured out the issue with the headset not starting. 

My system:

Arch with Hardened Kernel 6.14.9

AMD 7950X
AMD 7900XTX

I realized the drivers that were installed for Vulkan and the libraries were not the ones required for SteamVR and the driver. I removed the packages `amdvlk` and `lib32-amdvlk`. I then replaced them with `vulkan-radeon` and `lib32-vulkan-radeon`. The headset then started up perfectly.  

I still have an issue with Proton/Wine regarding the Lens distortion helper.

[image]

I'm not sure what's going on there but I will keep working at it. The issue here: https://github.com/CertainLach/VivePro2-Linux-Driver/issues/24. Seems to be a good starting point. 

I know your comment is from a few years ago but in case you or anyone else runs into the same trouble, try changing out the packages. It worked for me :-)

james-luther, 2025-06-21:
Update: I realize that even when running `steam-native` applications are still launched in `steam-runtime`. I found this by adding the following in Launch Options:

`PRESSURE_VESSEL_SHELL=instead %command% --shell=instead`

[image]

This opens up an xterm within the runtime and I can see both `proton` and `wine` are not available. 

[image]

When I attempt to run `steam-native` by setting `STEAM_RUNTIME=0`, `STEAM_RUNTIME_HEAVY=0` I am unable to open the runtime shell but I still am not able to locate wine even though it is setup in the global path in `/usr/bin`. And, when I specify the location with Launch Options it states no file found. I'm on the right path, I just havent found the right piece to this puzzle.

CertainLach, 2025-06-21:
> I realized the drivers that were installed for Vulkan and the libraries were not the ones required for SteamVR and the driver. I removed the packages amdvlk and lib32-amdvlk. I then replaced them with vulkan-radeon and lib32-vulkan-radeon. The headset then started up perfectly.

vulkan-radeon package still doesn't seem to be right package, you should just use mesa on arch linux, it includes everything

Note that proprietary amdgpu-pro driver won't work, if you have installed it for some reason

CertainLach, 2025-06-21:
So, you don't have any proton installed in `$STEAM_COMPAT_INSTALL_PATH/../Proton *`?

james-luther, 2025-06-21:
I have proton installed but it doesn't appear to be in that location... 

[image]

[image]

CertainLach, 2025-06-21:
Try to remove your ".steam/steam/steamapps/compatdata/250820" directory, it is possible your wine prefix is broken for some reason
Then this fix should be applied and work: https://github.com/CertainLach/VivePro2-Linux-Driver/pull/50

james-luther, 2025-06-21:
> Try to remove your ".steam/steam/steamapps/compatdata/250820" directory, it is possible your wine prefix is broken for some reason Then this fix should be applied and work: [#50](https://github.com/CertainLach/VivePro2-Linux-Driver/pull/50)

Thanks for all the help thus far 💓 . I subscribed to your Patreon because of it :-p

Same result with the two windoes and SteamVR error :-(

Here is the log:

```
Sat Jun 21 2025 13:24:57.854003 [Info] - //==============================================================================================
Sat Jun 21 2025 13:24:57.854098 [Info] - ================================================================================================
Sat Jun 21 2025 13:24:57.854120 [Info] - ================================================================================================
Sat Jun 21 2025 13:24:57.854145 [Info] - vrserver 2.11.2 startup with PID=36544, config=/home/colibri/.local/share/Steam/config, runtime=/home/colibri/.local/share/Steam/steamapps/common/SteamVR, arch=linux64
Sat Jun 21 2025 13:24:57.854162 [Info] - [vrserver] watchdogs enabled
Sat Jun 21 2025 13:24:57.854236 [Info] - Found saved crash timestamp time 1750527654 v1749156180 uptime 22 -- Using safe mode
Sat Jun 21 2025 13:24:57.854523 [Info] - CSharedResourceNamespaceServer starting up
Sat Jun 21 2025 13:24:57.854595 [Info] - CIPCPipe::ConnectPipe(SteamVR_Namespace) attempting connect to /steamvr/SteamVR_Namespace
Sat Jun 21 2025 13:24:57.854636 [Info] - Unable to connect to pipe errno=111
Sat Jun 21 2025 13:24:57.854659 [Error] - CSharedResourceNamespaceClient::Init(): failed connect
Sat Jun 21 2025 13:24:57.854673 [Warning] - CSharedResourceNamespaceClient::Init(): still waiting for vrserver to reply with namespace data
Sat Jun 21 2025 13:24:57.856420 [Error] - Failed to set thread 0x6a118882a6c0 priority: set failed for 5: -1: setpriority() failed
Sat Jun 21 2025 13:24:57.856456 [Error] - Failed to set thread priority: setschedparam failed 5: 1
Sat Jun 21 2025 13:24:57.856470 [Error] - Failed to set thread 0x6a118882a6c0 priority: invalid priority 5
Sat Jun 21 2025 13:24:57.856494 [Info] - CIPCPipe::CreatePipe(SteamVR_Namespace) bound to /steamvr/SteamVR_Namespace
Sat Jun 21 2025 13:24:57.856576 [Info] - Set thread 0x6a1185aee6c0 priority to: -15
Sat Jun 21 2025 13:24:57.954762 [Info] - CIPCPipe::ConnectPipe(SteamVR_Namespace) attempting connect to /steamvr/SteamVR_Namespace
Sat Jun 21 2025 13:24:57.954874 [Info] - CSharedResourceNamespaceServer sent namespace data: 36544
Sat Jun 21 2025 13:24:57.954925 [Info] - CSharedResourceNamespaceClient::Init(): received namespace data 36544
Sat Jun 21 2025 13:24:57.954953 [Info] - Self queried namespace: 36544
Sat Jun 21 2025 13:24:57.956658 [Info] - [Init] VR server 2.11.2 (v1749156180) starting up with config=/home/colibri/.local/share/Steam/config
Sat Jun 21 2025 13:24:57.956677 [Info] - [Init] VRServer will wait for VRMonitor before exiting
Sat Jun 21 2025 13:24:57.956740 [Info] - CIPCPipe::CreatePipe(VR_ServerPipe_36544) bound to /steamvr/VR_ServerPipe_36544
Sat Jun 21 2025 13:24:57.957344 [Info] - [Settings] Load Default Json Settings from /home/colibri/.local/share/Steam/steamapps/common/SteamVR/drivers/htc/resources/settings/default.vrsettings
Sat Jun 21 2025 13:24:57.957383 [Info] - [Settings] Load Default Json Settings from /home/colibri/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/resources/settings/default.vrsettings
Sat Jun 21 2025 13:24:57.957415 [Info] - [Settings] Load Default Json Settings from /home/colibri/.local/share/Steam/steamapps/common/SteamVR/drivers/null/resources/settings/default.vrsettings
Sat Jun 21 2025 13:24:57.957456 [Info] - [Settings] Load Default Json Settings from /home/colibri/.local/share/Steam/steamapps/common/SteamVR/drivers/vrlink/resources/settings/default.vrsettings
Sat Jun 21 2025 13:24:57.957574 [Info] - [Settings] Load Default Json Settings from /home/colibri/.local/share/Steam/steamapps/common/SteamVR/resources/settings/default.vrsettings
Sat Jun 21 2025 13:24:57.957831 [Info] - [Settings] Load Json Settings from /home/colibri/.local/share/Steam/config/steamvr.vrsettings
Sat Jun 21 2025 13:24:58.003245 [Info] - CSharedResourceNamespaceServer sent namespace data: 36544
Sat Jun 21 2025 13:24:58.004083 [Info] - [Settings] Load Default Json Settings from /home/colibri/.local/share/Steam/steamapps/common/SteamVR/drivers/vrlink/resources/settings/default.vrsettings
Sat Jun 21 2025 13:24:58.004154 [Info] - [Settings] Load Default Json Settings from /home/colibri/.local/share/Steam/steamapps/common/SteamVR/drivers/htc/resources/settings/default.vrsettings
Sat Jun 21 2025 13:24:58.004193 [Info] - [Settings] Load Default Json Settings from /home/colibri/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/resources/settings/default.vrsettings
Sat Jun 21 2025 13:24:58.004244 [Info] - [Settings] Load Default Json Settings from /home/colibri/.local/share/Steam/steamapps/common/SteamVR/drivers/null/resources/settings/default.vrsettings
Sat Jun 21 2025 13:24:58.004329 [Info] - [Settings] Load Default Json Settings from /home/colibri/.local/share/Steam/steamapps/common/SteamVR/resources/settings/default.vrsettings
Sat Jun 21 2025 13:24:58.004551 [Info] - [Settings] Load Json Settings from /home/colibri/.local/share/Steam/config/steamvr.vrsettings
Sat Jun 21 2025 13:24:58.004599 [Info] - Set fd limit to: 524288
[... cut ...]
Sat Jun 21 2025 13:32:48.086949 [Info] - lighthouse:  INFO lens_client: using lens server at "/home/colibri/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/lens-server/lens-server.exe"
Sat Jun 21 2025 13:32:48.087046 [Info] - lighthouse:  INFO lens_client: trying /home/colibri/.local/share/Steam/steamapps/common/Proton Hotfix/proton as proton
Sat Jun 21 2025 13:33:06.985680 [Info] - Failed Watchdog timeout in thread Connection load_drivers in LookForNewDevices after 21.495186 seconds. Aborting.
```

CertainLach, 2025-06-21:
Unfortunately, I have no idea why proton does that for some people.

There should be no external window for lens-server, driver communicates with it using stdin/stdout
Having it as an external window makes it impossible to communicate using stdio.

Are you building the driver using nix?

james-luther, 2025-06-21:
> Unfortunately, I have no idea why proton does that for some people.
> 
> There should be no external window for lens-server, driver communicates with it using stdin/stdout Having it as an external window makes it impossible to communicate using stdio.
> 
> Are you building the driver using nix?

It is probably something simple that I'm missing. I built the driver manually, then with nix, then just downloaded build from Patreon. All provide the same results.

joni999, 2025-06-22:
Back then what made it work for me was forcing Proton 8 by removing Proton 9. Maybe there is a better way now to just force a specific prefix so you can test it.
https://github.com/CertainLach/VivePro2-Linux-Driver/issues/33#issuecomment-2088761154

james-luther, 2025-06-22:
Same results. I wonder if it has to do with the hardened kernel.

CertainLach, 2025-06-22:
More like proton shenanigans

You can always just specify path to wine by using WINE env var

james-luther, 2025-06-23:
I decided to just blow away my Library folder and have Steam recreate it. This fixed my issues. Thanks!

CertainLach, 2025-06-23:
Just `steamapps/common`, or something more?

james-luther, 2025-06-23:
Initially I tried just removing `.steam` and after there was no change I removed `~/.local/share/Steam` and had it recreated. I'm sure there is a simpler way.