CertainLach/VivePro2-Linux-Driver #24: Failed to connect to headset display SteamVR 2.0.9

Samega7Cattac, 2023-10-31:
I'm having the exact same issue has #23 but since it deviated to another issue and it's using a older version of SteamVR decided to open a new one.
On Arch Linux and using the kernel 6.6 (so no kernel patches required I assume).
Using SteamVR 2.0.9 (tried with stable 2.0.8 and same issue).
I get `Failed to connect to headset display`.
Already enabled direct mode (idk how to check if it enabled correctly).
[image]

# vrcompositor.txt:
```
Tue Oct 31 2023 00:57:09.240188 [Info] - //==============================================================================================
Tue Oct 31 2023 00:57:09.240208 [Info] - ================================================================================================
Tue Oct 31 2023 00:57:09.240214 [Info] - ================================================================================================
Tue Oct 31 2023 00:57:09.240220 [Info] - vrcompositor 2.0.9 startup with PID=29632, config=/home/samega7cattac/.local/share/Steam/config, runtime=/home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR
Tue Oct 31 2023 00:57:09.240224 [Info] - VR compositor 2.0.9 (v1698453907) Mixed starting up
Tue Oct 31 2023 00:57:09.261706 [Info] - CIPCPipe::ConnectPipe(SteamVR_Namespace) attempting connect to /steamvr/SteamVR_Namespace
Tue Oct 31 2023 00:57:09.261724 [Info] - CSharedResourceNamespaceClient::Init(): received namespace data 28937
Tue Oct 31 2023 00:57:09.261733 [Info] - CIPCPipe::CreatePipe(VR_CompositorPipe_28937) bound to /steamvr/VR_CompositorPipe_28937
Tue Oct 31 2023 00:57:09.262090 [Info] - Set fd limit to: 524288
Tue Oct 31 2023 00:57:09.262494 [Info] - [Settings] Load Default Json Settings from /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/drivers/htc/resources/settings/default.vrsettings
Tue Oct 31 2023 00:57:09.262514 [Info] - [Settings] Load Default Json Settings from /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/resources/settings/default.vrsettings
Tue Oct 31 2023 00:57:09.262529 [Info] - [Settings] Load Default Json Settings from /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/drivers/null/resources/settings/default.vrsettings
Tue Oct 31 2023 00:57:09.262542 [Info] - [Settings] Load Default Json Settings from /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/drivers/openhmd/resources/settings/default.vrsettings
Tue Oct 31 2023 00:57:09.262622 [Info] - [Settings] Load Default Json Settings from /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/resources/settings/default.vrsettings
Tue Oct 31 2023 00:57:09.262774 [Info] - [Settings] Load Json Settings from /home/samega7cattac/.local/share/Steam/config/steamvr.vrsettings
Tue Oct 31 2023 00:57:09.263138 [Info] - Events created
Tue Oct 31 2023 00:57:09.263259 [Info] - Found bad mirror window settings: 
Tue Oct 31 2023 00:57:09.263266 [Info] - Creating CHmdWindowSDL!
Tue Oct 31 2023 00:57:09.263281 [Info] - CHmdWindowSDL: Using Wayland
Tue Oct 31 2023 00:57:09.275534 [Info] - WaylandHMDState: Got interface: wp_drm_lease_device_v1
Tue Oct 31 2023 00:57:09.275822 [Info] - WaylandHMDLeaseDevice::LeaseDevice_DrmFD: /dev/dri/card0
Tue Oct 31 2023 00:57:09.275838 [Info] - WaylandHMDLeaseConnector::LeaseConnector_Done
Tue Oct 31 2023 00:57:09.275850 [Info] - WaylandHMDLeaseDevice::LeaseDevice_Done
Tue Oct 31 2023 00:57:09.276181 [Info] - Instance layers available:
Tue Oct 31 2023 00:57:09.276187 [Info] -  - VK_LAYER_VALVE_steam_overlay_32, version 4206799
Tue Oct 31 2023 00:57:09.276191 [Info] -  - VK_LAYER_VALVE_steam_overlay_64, version 4206799
Tue Oct 31 2023 00:57:09.276194 [Info] -  - VK_LAYER_VALVE_steam_fossilize_32, version 4206799
Tue Oct 31 2023 00:57:09.276197 [Info] -  - VK_LAYER_VALVE_steam_fossilize_64, version 4206799
Tue Oct 31 2023 00:57:09.276200 [Info] -  - VK_LAYER_FROG_gamescope_wsi_x86_64, version 4206813
Tue Oct 31 2023 00:57:09.276204 [Info] -  - VK_LAYER_MANGOHUD_overlay_x86_64, version 4206592
Tue Oct 31 2023 00:57:09.276207 [Info] -  - VK_LAYER_MANGOAPP_overlay, version 4206592
Tue Oct 31 2023 00:57:09.276210 [Info] -  - VK_LAYER_MESA_device_select, version 4206803
Tue Oct 31 2023 00:57:09.276213 [Info] -  - VK_LAYER_AMD_switchable_graphics_32, version 4206859
Tue Oct 31 2023 00:57:09.276216 [Info] -  - VK_LAYER_INTEL_nullhw, version 4198473
Tue Oct 31 2023 00:57:09.276219 [Info] -  - VK_LAYER_MESA_overlay, version 4206803
Tue Oct 31 2023 00:57:09.276556 [Info] - Instance extensions available:
Tue Oct 31 2023 00:57:09.276565 [Info] -  - VK_KHR_device_group_creation, version 1
Tue Oct 31 2023 00:57:09.276577 [Info] -  - VK_KHR_display, version 23
Tue Oct 31 2023 00:57:09.276582 [Info] -  - VK_KHR_external_fence_capabilities, version 1
Tue Oct 31 2023 00:57:09.276587 [Info] -  - VK_KHR_external_memory_capabilities, version 1
Tue Oct 31 2023 00:57:09.276590 [Info] -  - VK_KHR_external_semaphore_capabilities, version 1
Tue Oct 31 2023 00:57:09.276593 [Info] -  - VK_KHR_get_display_properties2, version 1
Tue Oct 31 2023 00:57:09.276596 [Info] -  - VK_KHR_get_physical_device_properties2, version 2
Tue Oct 31 2023 00:57:09.276599 [Info] -  - VK_KHR_get_surface_capabilities2, version 1
Tue Oct 31 2023 00:57:09.276603 [Info] -  - VK_KHR_surface, version 25
Tue Oct 31 2023 00:57:09.276606 [Info] -  - VK_KHR_surface_protected_capabilities, version 1
Tue Oct 31 2023 00:57:09.276609 [Info] -  - VK_KHR_wayland_surface, version 6
Tue Oct 31 2023 00:57:09.276611 [Info] -  - VK_KHR_xcb_surface, version 6
Tue Oct 31 2023 00:57:09.276615 [Info] -  - VK_KHR_xlib_surface, version 6
Tue Oct 31 2023 00:57:09.276618 [Info] -  - VK_EXT_acquire_drm_display, version 1
Tue Oct 31 2023 00:57:09.276621 [Info] -  - VK_EXT_acquire_xlib_display, version 1
Tue Oct 31 2023 00:57:09.276624 [Info] -  - VK_EXT_debug_report, version 10
Tue Oct 31 2023 00:57:09.276627 [Info] -  - VK_EXT_debug_utils, version 2
Tue Oct 31 2023 00:57:09.276630 [Info] -  - VK_EXT_direct_mode_display, version 1
Tue Oct 31 2023 00:57:09.276633 [Info] -  - VK_EXT_display_surface_counter, version 1
Tue Oct 31 2023 00:57:09.276636 [Info] -  - VK_EXT_surface_maintenance1, version 1
Tue Oct 31 2023 00:57:09.276639 [Info] -  - VK_EXT_swapchain_colorspace, version 4
Tue Oct 31 2023 00:57:09.276642 [Info] -  - VK_KHR_portability_enumeration, version 1
Tue Oct 31 2023 00:57:09.276645 [Info] -  - VK_LUNARG_direct_driver_loading, version 1
Tue Oct 31 2023 00:57:09.276650 [Info] - SDL version: 2.28.4
Tue Oct 31 2023 00:57:09.276655 [Info] - Direct mode features: present
[... 493 lines cut ...]
      "PowerManagedBaseStations2" : "LHB-69DA65D6:BS2:1,LHB-131E16FF:BS2:1"
   },
   "steam.app.955610" : {
      "disableAsync" : true,
      "resolutionScale" : 100
   },
   "steamvr" : {
      "enableLinuxVulkanAsync" : false,
      "installID" : "10172945583768885362",
      "lastVersionNotice" : "2.0.9",
      "lastVersionNoticeDate" : "1698459667",
      "showAdvancedSettings" : true,
      "showMirrorView" : false,
      "supersampleManualOverride" : true
   },
   "system.generated.openvrbenchmark.exe" : {
      "disableAsync" : true
   },
   "vivepro2" : {
      "brightness" : 130,
      "noiseCancel" : false,
      "resolution" : 3
   }
}
```

CertainLach, 2023-10-31:
This driver wants standalone-installed wine to work:
ERROR driver_proxy::server::driver_host: failed to wrap hmd: lens client error: can't find wine in PATH
This is because there is a lot of complex math code in windows-only library provided by vive (LibLensDistortion.so), and this library is ran using lens-server.exe proxy.

Samega7Cattac, 2023-10-31:
@CertainLach but I have wine-staging installed
Is something missing?

CertainLach, 2023-10-31:
It is available using `wine` command?
If not, you can specify path to wine directly using SteamVR launch args:
`WINE=/path/to/wine %command%`

Samega7Cattac, 2023-10-31:
```
$ whereis wine
wine: /usr/bin/wine /usr/lib/wine /usr/lib32/wine /usr/include/wine /usr/share/wine /usr/share/man/man1/wine.1.gz
$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/bin:/home/samega7cattac/.dotnet/tools:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/opt/rocm-5.7.1/bin:/opt/rocm-5.7.1/hip/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin
```
I will try setting it in the args

Samega7Cattac, 2023-10-31:
@CertainLach ok the issue now seems different
```
$ steam
steam.sh[11437]: Running Steam on arch rolling 64-bit
steam.sh[11437]: STEAM_RUNTIME is enabled automatically
setup.sh[11509]: Steam runtime environment up-to-date!
steam.sh[11437]: Steam client's requirements are satisfied
[2023-10-31 21:29:46] Startup - updater built Oct 25 2023 18:40:00
[2023-10-31 21:29:46] Startup - Steam Client launched with: '/home/samega7cattac/.local/share/Steam/ubuntu12_32/steam'
10/31 21:29:46 Init: Installing breakpad exception handler for appid(steam)/version(1698260427)/tid(11555)
[2023-10-31 21:29:46] Loading cached metrics from disk (/home/samega7cattac/.local/share/Steam/package/steam_client_metrics.bin)
[2023-10-31 21:29:46] Using the following download hosts for Public, Realm steamglobal
[2023-10-31 21:29:46] 1. https://client-update.akamai.steamstatic.com, /, Realm 'steamglobal', weight was 1000, source = 'update_hosts_cached.vdf'
[2023-10-31 21:29:46] 2. https://cdn.cloudflare.steamstatic.com, /client/, Realm 'steamglobal', weight was 1, source = 'update_hosts_cached.vdf'
[2023-10-31 21:29:46] 3. https://cdn.steamstatic.com, /client/, Realm 'steamglobal', weight was 1, source = 'baked in'
[2023-10-31 21:29:46] uninstalled manifest found in /home/samega7cattac/.local/share/Steam/package/steam_client_ubuntu12 (1).
[2023-10-31 21:29:46] Found pending update
[2023-10-31 21:29:46] Installing update...
[2023-10-31 21:29:46] Extracting package...
[2023-10-31 21:29:49] Installing update...
[2023-10-31 21:29:49] Cleaning up...
[2023-10-31 21:29:49] Update complete, launching Steam...
[2023-10-31 21:29:49] Shutdown
steam.sh[11437]: Restarting Steam by request...
steam.sh[11437]: Running Steam on arch rolling 64-bit
steam.sh[11437]: STEAM_RUNTIME has been set by the user to: /home/samega7cattac/.local/share/Steam/ubuntu12_32/steam-runtime
setup.sh[11655]: Steam runtime environment up-to-date!
steam.sh[11437]: Steam client's requirements are satisfied
[2023-10-31 21:29:49] Startup - updater built Oct 31 2023 02:08:17
[2023-10-31 21:29:49] Startup - Steam Client launched with: '/home/samega7cattac/.local/share/Steam/ubuntu12_32/steam'
10/31 21:29:49 Init: Installing breakpad exception handler for appid(steam)/version(1698777785)/tid(11701)
[2023-10-31 21:29:49] Loading cached metrics from disk (/home/samega7cattac/.local/share/Steam/package/steam_client_metrics.bin)
[2023-10-31 21:29:49] Using the following download hosts for Public, Realm steamglobal
[2023-10-31 21:29:49] 1. https://client-update.akamai.steamstatic.com, /, Realm 'steamglobal', weight was 1000, source = 'update_hosts_cached.vdf'
[2023-10-31 21:29:49] 2. https://cdn.cloudflare.steamstatic.com, /client/, Realm 'steamglobal', weight was 1, source = 'update_hosts_cached.vdf'
[2023-10-31 21:29:49] 3. https://cdn.steamstatic.com, /client/, Realm 'steamglobal', weight was 1, source = 'baked in'
[2023-10-31 21:29:49] Verifying installation...
[2023-10-31 21:29:49] Verification complete

Steam logging initialized: directory: /home/samega7cattac/.local/share/Steam/logs

XRRGetOutputInfo Workaround: initialized with override: 0 real: 0xeba16dc0
XRRGetCrtcInfo Workaround: initialized with override: 0 real: 0xeba15500
steamwebhelper.sh[11725]: Runtime for steamwebhelper: defaulting to /home/samega7cattac/.local/share/Steam/ubuntu12_64/steam-runtime-heavy
steamwebhelper.sh[11725]: glibc >= 2.34, partially disabling sandbox until CEF supports clone3()
CAppInfoCacheReadFromDiskThread took 30 milliseconds to initialize
/usr/share/themes/Breeze/gtk-2.0/widgets/entry:70: error: unexpected identifier 'direction', expected character '}'
/usr/share/themes/Breeze/gtk-2.0/widgets/styles:36: error: invalid string constant "combobox_entry", expected valid string constant
Steam Runtime Launch Service: starting steam-runtime-launcher-service
Steam Runtime Launch Service: steam-runtime-launcher-service is running pid 11903
bus_name=com.steampowered.PressureVessel.LaunchAlongsideSteam
Fossilize INFO: Setting autogroup scheduling.
BRefreshApplicationsInLibrary 1: 1ms
BuildCompleteAppOverviewChange: 445 apps
RegisterForAppOverview 1: 8ms
RegisterForAppOverview 2: 8ms
steam[11701]: Attempting HID Open IMU: LHR-77F442F8
steam[11701]: Lighthouse IMU HID opened
steam[11701]: LHR-77F442F8: Firmware Version 1526689969 watchman@runner-watchman 2018-05-19 FPGA 531(2.19/7/2) BL 1526603673
steam[11701]: LHR-77F442F8: Successfully fetched gyro/accelerometer range modes from the device. GyroRangeMode:1 AccelRangeMode:1
steam[11701]: Attempting HID Open Optical: LHR-77F442F8
steam[11701]: Lighthouse Optical HID opened
steam[11701]: Attempting HID Open VrController: LHR-77F442F8
steam[11701]: Lighthouse VrController HID opened
steam[11701]: LHR-77F442F8: Read config of 3728 bytes from [vid:28de, pid:2300] (LHR-77F442F8) and inflated to 11914 bytes
steam[11701]: Attempting HID Open IMU: 4DDCBE6EDE
steam[11701]: Lighthouse IMU HID opened
steam[11701]: Attempting HID Open IMU: 70C25FA277
steam[11701]: Lighthouse IMU HID opened
steam[11701]: 4DDCBE6EDE: Triggered keepalive (succeeded)
[... 295 lines cut ...]
vrcompositor-launcher.sh[12929]: === Tue Oct 31 09:30:24 PM WET 2023 ===
vrcompositor-launcher.sh[12929]: Relaunch under scout LDLP runtime.
vrcompositor-launcher.sh[12929]: exec /home/samega7cattac/.steam/root/ubuntu12_32/steam-runtime/run.sh /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/bin/linux64/vrcompositor-launcher.sh
[... cut ...]
Mono path[0] = '/home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/tools/steamvr_room_setup/linux64/steamvr_room_setup_Data/Managed'
Mono path[1] = '/home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/tools/steamvr_room_setup/linux64/steamvr_room_setup_Data/Mono'
Mono config path = '/home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/tools/steamvr_room_setup/linux64/steamvr_room_setup_Data/Mono/etc'
Preloaded 'ScreenSelector.so'
Preloaded 'libopenvr_api.so'
 INFO driver_proxy::driver::server_tracked_provider: waking up base stations
```

CertainLach, 2023-10-31:
It won't help then, steamvr preserves PATH. Maybe something from SteamVR's LD_LIBRARY_PATH breaks wine?..

CertainLach, 2023-10-31:
Huh, nvm.

Samega7Cattac, 2023-10-31:
It helped, but now it's `ERROR driver_proxy::server::driver_host: failed to wrap hmd: lens client error: io error: No such file or directory (os error 2)`

CertainLach, 2023-10-31:
Have you installed lens-server to correct path? Have you built this driver yourself, or downloaded installer from patreon?

Samega7Cattac, 2023-10-31:
Built it using https://github.com/santeri3700/vive-pro-2-on-linux#install-vive-pro-2-linux-driver-by-certainlach

CertainLach, 2023-10-31:
Lens-server should be placed at "/home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/lens-server/lens-server.exe"
And there should be opencv and LibLensDistortion in the same directory.

Samega7Cattac, 2023-10-31:
I can try the nix if you think it's from the way it was built

CertainLach, 2023-10-31:
Nix is definitely works, this is how tested binaries are built. First build might take some time, however)

Samega7Cattac, 2023-10-31:
```
$ file /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/lens-server/lens-server.exe
/home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/lens-server/lens-server.exe: PE32+ executable (console) x86-64, for MS Windows, 24 sections
$ ls /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/lens-server/
lens-server.exe  LibLensDistortion.dll  opencv_world346.dll
```
The only thing diff from that guide was that I used a nightly build without setting the date

Samega7Cattac, 2023-10-31:
I will try nix then

Samega7Cattac, 2023-10-31:
`error: flake 'git+file:///home/samega7cattac/gits/VivePro2-Linux-Driver' does not provide attribute 'packages.x86_64-linux.default' or 'defaultPackage.x86_64-linux'`
I might have fked up rust when trying to build from the guide xD

CertainLach, 2023-10-31:
Note the comment above nix build command :D
```shell
# Note: the #driver-proxy-release part of this command is not a comment, it's just github syntax highlighter is wrong
nix build --extra-experimental-features nix-command --extra-experimental-features flakes .#driver-proxy-release
```

Samega7Cattac, 2023-10-31:
@CertainLach how can I make it not compile to RAM?
Crashed my system and corrupted my disk

CertainLach, 2023-11-01:
It needs to build a lot of stuff to be compatible with older glibc, which is wanted by steam runtime. By default, it builds in /tmp, you can limit nix concurrency to overcome that, or do not mount /tmp as ramdisk.
I don't know how to make it use another directory, since I'm okay with in-ram builds (256gb ram is enough for cold build)

Samega7Cattac, 2023-11-01:
setting `TMPDIR` env var worked.
now `sewer` is missing, where can I find it?

Samega7Cattac, 2023-11-01:
used nix and then followed the rest from the previous guide and same error

Samega7Cattac, 2023-11-01:
Nop
```
$ ./install.sh 
SteamVR proxy driver for Vive Pro 2
Consider supporting developer on patreon: https://patreon.com/0lach
SteamVR at /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR
Found both original driver, and old original driver, seems like SteamVR was updated
= Moving updated original driver
= Patching real driver
# Driver requirement check
@485827..485839
-\x41\xc7\x84\x24\xac\x00\x00\x00\xd6\x00\x00\x00
+\x90\x90\x90\x90\x90\x90\x90\x90\x90\x90\x90\x90
# Error message clarification
@3835470..3835490
-\x77\x68\x69\x63\x68\x20\x69\x73\x20\x6e\x6f\x74\x20\x70\x72\x65\x73\x65\x6e\x74
+\x69\x67\x6e\x6f\x72\x69\x6e\x67\x2e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
= Overriding current driver
= Updating proxy server
= Testing proxy server
Wine found at wine
Installation finished, try to start SteamVR
```

CertainLach, 2023-11-01:
And it still prints "No such file or directory" on lens-server launch?

CertainLach, 2023-11-01:
I think you have set WINE variable to wrong value, since it definitely fails on `$WINE lens-server.exe` command

Samega7Cattac, 2023-11-01:
I have in steamVR `WINE=/usr/bin/wine %command%`

CertainLach, 2023-11-01:
May this wine be x32-only?
Does `/usr/bin/wine /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/lens-server/lens-server.exe` works? Does it exit with exit code 42? (`echo $?`)

Long-term plan is to get rid of wine dependency, but this is not an easy task. https://github.com/CertainLach/VivePro2-Linux-Driver/issues/25

Samega7Cattac, 2023-11-01:
```
$ /usr/bin/wine /home/samega7cattac/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/lens-server/lens-server.exe
002c:fixme:winediag:loader_init wine-staging 8.18 is a testing version containing experimental patches.
002c:fixme:winediag:loader_init Please mention your exact version when filing bug reports on winehq.org.
0088:fixme:hid:handle_IRP_MN_QUERY_ID Unhandled type 00000005
0088:fixme:hid:handle_IRP_MN_QUERY_ID Unhandled type 00000005
0088:fixme:hid:handle_IRP_MN_QUERY_ID Unhandled type 00000005
0088:fixme:hid:handle_IRP_MN_QUERY_ID Unhandled type 00000005
0088:fixme:wineusb:query_id Unhandled ID query type 0x5.
0088:fixme:wineusb:query_id Unhandled ID query type 0x5.
0088:fixme:wineusb:query_id Unhandled ID query type 0x5.
0088:fixme:wineusb:query_id Unhandled ID query type 0x5.
0088:fixme:wineusb:query_id Unhandled ID query type 0x5.
0088:fixme:wineusb:query_id Unhandled ID query type 0x5.
0088:fixme:wineusb:query_id Unhandled ID query type 0x5.
0088:fixme:wineusb:query_id Unhandled ID query type 0x5.
0088:fixme:wineusb:query_id Unhandled ID query type 0x5.
 INFO lens_server: hello from lens server
 INFO lens_server: dll path: "Z:\\home\\samega7cattac\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\LibLensDistortion.dll"

```

CertainLach, 2023-11-01:
Maybe LD_LIBRARY_PATH set by steamvr breaks library search path for wine?
What about adding `LD_DEBUG=libs` to SteamVR launch args, I think there might be some library search errors before "No such file or directory"

Samega7Cattac, 2023-11-01:
[steam_out.log](https://github.com/CertainLach/VivePro2-Linux-Driver/files/13231037/steam_out.log)

tek, 2023-11-01:
my experience is that you need `wine64` in the global environment's path – I think the local shell env isn't inherited by the script that starts the lens server

CertainLach, 2023-11-01:
There is no library being loaded between "using lens server at ..." and "failed to wrap hmd", so wine is not starting at all, so it can't find wine at /usr/bin/wine64. Are you using flatpak/similar sandboxing-based something for steam?..

> my experience is that you need wine64 in the global environment's path – I think the local shell env isn't inherited by the script that starts the lens server
He already put path to wine in environment variable, so PATH doesn't matter anymore.

Samega7Cattac, 2023-11-01:
No, straight from official arch linux repo

smalchow, 2023-11-08:
Hi,

i am experiencing a similar problem, i installed the driver from the patreon page.

i checked for lens-server.exe, it is there and running it is like this

[image]

when i creak it, the return value is obviously 0.

i am runnning on Nobara Linux (basically Fedora on Speed) with new (proprietary) AMD drivers (tried the OSS ones as well.
SteamVR Version 2.0.10
Kernel: 6.5.9-201.fsync.fc38.x86_64

Can i do something to help narrow down the problem ?

And @CertainLach  - thank you so much for working on this, i will keep on contributing to your patreon

CertainLach, 2023-11-08:
I need to retest my driver on other distribution, as in my (NixOS) steam works a bit differently, and I experience no issues with my driver.

smalchow, 2023-11-09:
Can i help in any way ?

smalchow, 2023-11-09:
I downgraded SteamVR to 1.27.5 - the error remains the same

CertainLach, 2023-11-11:
> I downgraded SteamVR to 1.27.5 - the error remains the same

Some changes are occured in steam client itself

CertainLach, 2023-11-11:
New steam client defenitely messes up PATH and LD_LIBRARY_PATH of launched applications, so globally-installed wine is not being able to start on arch linux. I will try to use proton, as it is seems to work.

CertainLach, 2023-11-11:
Figured out what they did, added proton support on master branch.
However, I need to fix compatibility with old ubuntu host system (And ubuntu is old by definition, I don't like non-rolling distributions D:) somehow, currently it will fail here with glibc version error.
Patreon release is coming after I fix glibc compatibility issue.

CertainLach, 2023-11-11:
Well, this time I think it is possible to drop support for older distributions.
Required glibc version is 2.34, and this is
- Fedora 35+
- Ubuntu 22.04+
- Debian 12(bookworm)+
- NixOS 22.05+
- Manjaro/ArchLinux - yes

(Based on https://repology.org/project/glibc/versions)

CertainLach, 2023-11-11:
v0.7.0 is released.

closed 2023-11-11 by CertainLach

Samega7Cattac, 2023-11-18:
Seems that issue got fixed but still doesn't work on me
Now the error is completely different.
In the UI it says `SteamVR Fail: A key component of SteamVR isn't working properly.` and following the link goes to a error `VRApplicationError_IPCFailed`.
Here's the new logs
[steam_out1.log](https://github.com/CertainLach/VivePro2-Linux-Driver/files/13399950/steam_out1.log)

NOTE: I removed the custom command of the SteamVR when launching

CertainLach, 2023-11-18:
Which SteamVR version do you have? IPCFailed (==Error 105) sounds like a SteamVR error itself.
You can try to dismiss it, but it looks to me like you have some system misconfiguration.

`[tracing-subscriber] Unable to write an event to the Writer for this Subscriber! Error: failed to write whole buffer` line in logs, means that the process has died too early. Please upload a full system report generated from SteamVR, as new version of the driver stores its logs there.
QSslSocket errors are also concerning, it can't find the openssl library (`qt.network.ssl: QSslSocket: cannot resolve`).
If you're using SteamVR Beta (And I think you do), you should also use Steam client beta

On master branch, the compositor should start even if lens server has died, so I don't think this driver startup may even result in that error.

Samega7Cattac, 2023-11-18:
It says `(301)` (idk if that's what u referring to)
[image]
I enabled the beta client and same issue.
I'm on SteamVR 2.1.4 rn, how do I downgrade to 2.1.2?

Samega7Cattac, 2023-11-18:
Tried 1.27 and it works for now