CertainLach/VivePro2-Linux-Driver #27: Unable to use 120hz (VP2)

issacdowling, 2023-12-03:
If I attempt to use the highest (5) resolution at 120hz, it claims to fail to connect to the display (though it works fine at 90 due to having patched DSC), and the lower (1) resolution at 120 just runs at 90. 

vrcompositor.txt says 
`Trying to match desired rate of 90.000000Hz.`
Despite being set to 120 by picking resolution 1 in the vrconfig. The same happens if I change other options directly in `vive-hid/lib.rs`. 

Kernel: 6.6.3 (but applies to all that I've tried)
SteamVR: 2.1.8 (but applies to any others that I've tried too - 1.27.5 and the beta)
AMD GPU

(Originally I made this part of a previous issue, but I thought it better so separate out)

CertainLach, 2023-12-03:
It works in KDE Wayland for me.
If xrandr shows the correct information, then I'm not sure why SteamVR thinks it is 90fps in your case.

issacdowling, 2023-12-03:
Could it be related to the fact that I get a -496 (failed to connect to display) ONLY when I use resolution 5, despite 4 being fine with DSC?

issacdowling, 2023-12-03:
In the SteamVR settings, there is a place where there _should_ be refresh rate options, but it just says refresh rate 90, with no other choices

issacdowling, 2023-12-03:
Would it be possible for me to hardcode those values within this driver?

issacdowling, 2023-12-03:
Is this the kind of thing you know how to do, or would I have to look into it independently? I'm fine doing a weird patch since I'm never gonna switch away from 120hz. Either way, you've been a great help so far, I really appreciate it

CertainLach, 2023-12-04:
I have no idea, but I will look if one of the patreon members also has this issue.

As for dirty hacks... You can try to find `default.vrsettings` for `steamvr.preferredRefreshRate` config value and write your own refresh rate here, I think this is where steamvr gets default value from.

issacdowling, 2023-12-04:
Found that setting, but whether in my default.vrsettings or regular steamvr.vrsettings, it doesn't seem to do anything. Would this be at all related to the kernel patches? (They seem to work, as DSC is fine, but just maybe an avenue for me to investigate)

CertainLach, 2023-12-04:
Did it still print "Trying to match desired rate of 90.000000Hz."? Looking at the assembly, it should try to match 90Hz then.
This is not related to the kernel patches, if xrandr is able to correctly report 120fps

issacdowling, 2023-12-04:
No, actually. SteamVR _shows_ 90hz, and it feels like 90hz, but vrcompositor.txt says it's `Trying to match desired rate of 120.000000Hz`

However, looking further into that file, it then says 
`Updated HMD Prop_DisplayFrequency_Float to 90.000000`
and 
`HMD driver recommended: 2448x1224 90.0Hz`

The displays are clearly running at 120hz as opposed to 90, as I can see some weirdness (repeated frames or more ghosting, something like that), and SteamVR mentions targeting 120, but then that it gets set to 90 anyway

CertainLach, 2024-01-21:
Can you provide EDID for yours mode 1?
It looks to me like SteamVR sees frame rate as 90Hz in EDID for you.

It should be possible to explicitly override Prop_DisplayFrequency_Float in this driver, but I think SteamVR should know better, as it discovers this information from EDID.

CertainLach, 2024-01-21:
You can try using this branch, here I added ability to explicitly provide frame rate.
https://github.com/CertainLach/VivePro2-Linux-Driver/pull/31

issacdowling, 2024-01-21:
Setting Refresh Rate to 120 in the SteamVR settings menu still does not make things render at 120hz,
```
Sun Jan 21 2024 18:18:13.242558 [Info] - HMD driver recommended: 2448x1224 90.0Hz HiddenArea(0.00%) = 539 MP/sec
```
```
Sun Jan 21 2024 18:18:23.241141 [Info] - Request to change refresh to 120.000000Hz (hmdsupport:0 graphicsdevicesupport:0)
```
These two lines seem relevant in `vrcompositor.txt`

issacdowling, 2024-01-21:
Those lines were from when I changed from 90 to 120.
When restarting, already set to 120, it just says
```
HMD driver recommended: 2448x1224 90.0Hz HiddenArea(0.00%) = 539 MP/sec
```
With the only mention of 120hz being
```
Sun Jan 21 2024 18:28:15.186983 [Info] - Tried to find direct display through Wayland: 0x5639d8649cb0
Sun Jan 21 2024 18:28:15.186999 [Info] - Trying to match desired rate of 120.000000Hz.
Sun Jan 21 2024 18:28:15.187008 [Info] - 12 modes on display:
Sun Jan 21 2024 18:28:15.187017 [Info] -  - 0: 2448x1224@120.047005Hz
Sun Jan 21 2024 18:28:15.187027 [Info] -  - 1: 1920x1200@120.047005Hz
Sun Jan 21 2024 18:28:15.187035 [Info] -  - 2: 1920x1080@120.047005Hz
Sun Jan 21 2024 18:28:15.187046 [Info] -  - 3: 1600x1200@120.047005Hz
Sun Jan 21 2024 18:28:15.187054 [Info] -  - 4: 1680x1050@120.047005Hz
Sun Jan 21 2024 18:28:15.187063 [Info] -  - 5: 1280x1024@120.047005Hz
Sun Jan 21 2024 18:28:15.187071 [Info] -  - 6: 1440x900@120.047005Hz
Sun Jan 21 2024 18:28:15.187080 [Info] -  - 7: 1280x800@120.047005Hz
Sun Jan 21 2024 18:28:15.187088 [Info] -  - 8: 1280x720@120.047005Hz
Sun Jan 21 2024 18:28:15.187096 [Info] -  - 9: 1024x768@120.047005Hz
Sun Jan 21 2024 18:28:15.187105 [Info] -  - 10: 800x600@120.047005Hz
Sun Jan 21 2024 18:28:15.187113 [Info] -  - 11: 640x480@120.047005Hz
Sun Jan 21 2024 18:28:15.187122 [Info] - Selected mode 0.
```
However, SteamVR still only renders at 90

issacdowling, 2024-01-21:
> Can you provide EDID for yours mode 1? It looks to me like SteamVR sees frame rate as 90Hz in EDID for you.
> 
> It should be possible to explicitly override Prop_DisplayFrequency_Float in this driver, but I think SteamVR should know better, as it discovers this information from EDID.

I'm not super sure how I should provide the edid, so here's edid-decode's output with res 1 selected
```
edid-decode (hex):

00 ff ff ff ff ff ff 00 22 d2 04 aa 00 00 00 00
00 1e 01 04 a5 00 00 78 00 b8 60 a9 54 4c a9 27
0c 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01
01 01 01 01 01 01 00 00 00 10 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 fc 00 56 49 56
45 20 50 72 6f 20 32 0a 20 20 00 00 00 10 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 10
00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 df

70 20 79 07 00 22 00 14 1c 97 06 80 8f 09 63 00
31 80 03 00 c7 04 bb 00 a7 80 01 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 94 90

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.4
  Vendor & Product Identification:
    Manufacturer: HVR
    Model: 43524
    Made in: 2020
  Basic Display Parameters & Features:
    Digital display
    Bits per primary color channel: 8
    DisplayPort interface
    Image size is variable
    Gamma: 2.20
    Supported color formats: RGB 4:4:4
    First detailed timing does not include the native pixel format and preferred refresh rate
  Color Characteristics:
    Red  : 0.6621, 0.3310
    Green: 0.2988, 0.6601
    Blue : 0.1533, 0.0488
    White: 0.3125, 0.3281
  Established Timings I & II: none
  Standard Timings: none
  Detailed Timing Descriptors:
    Dummy Descriptor:
    Display Product Name: 'VIVE Pro 2'
    Dummy Descriptor:
    Dummy Descriptor:
  Extension blocks: 1
Checksum: 0xdf

----------------

Block 1, DisplayID Extension Block:
  Version: 2.0
  Extension Count: 0
  Display Product Primary Use Case: Head-mounted Virtual Reality (VR) display
  Video Timing Modes Type 7 - Detailed Timings Data Block:
    DTD:  2448x1224  120.046662 Hz   1:1    169.506 kHz    431.901000 MHz (aspect 1:1, no 3D stereo, preferred)
               Hfront   50 Hsync   4 Hback   46 Hpol P
               Vfront  168 Vsync   2 Vback   18 Vpol P
  Checksum: 0x94
Checksum: 0x90

```

joni999, 2024-04-20:
Hi,
I have the same issue. Using mode 1 SteamVR doesn't want to output 120hz
[image]

I do have the 120 Hz toggle in the SteamVR Settings as seen above, but it doesn't influence the output refresh rate (of course I always restart after attempting to switch Refesh rate)

vrcompositor log messages looks very similar to issacdowling's report.
[vrcompositor.txt.mode1.240420.log](https://github.com/CertainLach/VivePro2-Linux-Driver/files/15048280/vrcompositor.txt.mode1.240420.log)

I'm on KDE X11 + Up to date Arch Linux on Wayland I have the same issue.

GPU: RTX 3090ti
CPU: Intel i7 13700k
SteamVR Stable 2.4.4

Everything else works fine so far. I was able to play a few hours of Beat Saber without issue (but in 90 Hz). Thank you very much for making this possible!

EDID (to get it I had to enable "AllowHMD" in xorg.conf, because the Nvidia driver completely hides the HMD by default) :
```
edid-decode (hex):

00 ff ff ff ff ff ff 00 22 d2 04 aa 00 00 00 00
00 1e 01 04 a5 00 00 78 00 b8 60 a9 54 4c a9 27
0c 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01
01 01 01 01 01 01 00 00 00 10 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 fc 00 56 49 56
45 20 50 72 6f 20 32 0a 20 20 00 00 00 10 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 10
00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 df

70 20 79 07 00 22 00 50 48 98 06 80 8f 09 63 00
31 80 03 00 c7 04 93 02 7f 82 01 00 1c 97 06 00
8f 09 63 00 31 80 03 00 c7 04 bb 00 a7 80 01 00
16 e4 07 00 bf 0c 77 00 3b 80 03 00 5f 06 41 00
2d 80 01 00 ba a5 09 00 5f 0e 63 00 31 80 03 00
2b 07 15 00 0b 80 01 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 4d 90

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.4
  Vendor & Product Identification:
    Manufacturer: HVR
    Model: 43524
    Made in: 2020
  Basic Display Parameters & Features:
    Digital display
    Bits per primary color channel: 8
    DisplayPort interface
    Image size is variable
    Gamma: 2.20
    Supported color formats: RGB 4:4:4
    First detailed timing does not include the native pixel format and preferred refresh rate
  Color Characteristics:
    Red  : 0.6621, 0.3310
    Green: 0.2988, 0.6601
    Blue : 0.1533, 0.0488
    White: 0.3125, 0.3281
  Established Timings I & II: none
  Standard Timings: none
  Detailed Timing Descriptors:
    Dummy Descriptor:
    Display Product Name: 'VIVE Pro 2'
    Dummy Descriptor:
    Dummy Descriptor:
  Extension blocks: 1
Checksum: 0xdf

----------------

Block 1, DisplayID Extension Block:
  Version: 2.0
  Extension Count: 0
  Display Product Primary Use Case: Head-mounted Virtual Reality (VR) display
  Video Timing Modes Type 7 - Detailed Timings Data Block:
    DTD:  2448x1224   90.033772 Hz   1:1    169.624 kHz    432.201000 MHz (aspect 1:1, no 3D stereo, preferred)
               Hfront   50 Hsync   4 Hback   46 Hpol P
               Vfront  640 Vsync   2 Vback   18 Vpol P
    DTD:  2448x1224  120.046662 Hz   1:1    169.506 kHz    431.901000 MHz (aspect 1:1, no 3D stereo)
               Hfront   50 Hsync   4 Hback   46 Hpol P
               Vfront  168 Vsync   2 Vback   18 Vpol P
    DTD:  3264x1632   90.000021 Hz   1:1    152.820 kHz    517.143000 MHz (aspect 1:1, no 3D stereo)
               Hfront   60 Hsync   4 Hback   56 Hpol P
               Vfront   46 Vsync   2 Vback   18 Vpol P
    DTD:  3680x1836   90.022696 Hz   1:1    167.262 kHz    632.251000 MHz (aspect 1:1, no 3D stereo)
               Hfront   50 Hsync   4 Hback   46 Hpol P
               Vfront   12 Vsync   2 Vback    8 Vpol P
  Checksum: 0x4d
Checksum: 0x90

----------------

Preferred Video Timing if Block 0 and DisplayID Blocks are parsed:
  DTD:  2448x1224   90.033772 Hz   1:1    169.624 kHz    432.201000 MHz (aspect 1:1, no 3D stereo, preferred)
             Hfront   50 Hsync   4 Hback   46 Hpol P
             Vfront  640 Vsync   2 Vback   18 Vpol P

----------------

No Native Video Resolution was defined.

----------------

edid-decode SHA: 84ddf9155376 2021-10-03 10:37:45

Failures:

Block 0, Base EDID:
  Standard Timings: Missing preferred timing.
EDID:
  DisplayID: Missing DisplayID Product Identification Data Block.
  DisplayID: Missing DisplayID Display Parameters Data Block.
  DisplayID: Missing DisplayID Display Interface Features Data Block.

EDID conformity: FAIL

```

Any clues?

joni999, 2024-04-20:
Regarding this:

> You can try using this branch, here I added ability to explicitly provide frame rate. #31

Do I have to configure anything special to explicitly provide the frame rate?

joni999, 2024-05-09:
@CertainLach could you please share a vrcompositor log for us to compare?