CertainLach/VivePro2-Linux-Driver #29: Windows build

mdovgialo, 2024-01-17:
Is it possible to build this driver for Windows, so I don't need to run Vive Console and it's secondary compositor? Make VP2 more SteamVR native? It could free up to 2 gigabytes of VRAM which Vive Consoles secondary compositor is using.

CertainLach, 2024-01-17:
It is possible, in the past I was building this driver for windows just to exclude linux-specific variables, and then I always used cross-platform libs to do everything... But I haven't tested windows support in a long time, and not sure if it still works.

issacdowling, 2024-01-17:
+1 to this, it would help my debugging on Linux (I'm having issues I think related to vsync_to_photon latency), as it would let me isolate driver issues from Linux SteamVR issues, and would also make Windows use better.

mdovgialo, 2024-01-17:
> It is possible, in the past I was building this driver for windows just to exclude linux-specific variables, and then I always used cross-platform libs to do everything... But I haven't tested windows support in a long time, and not sure if it still works.

I would love to try the windows build, if you could compile and share it.

issacdowling, 2024-01-17:
I did get as far as compiling for Windows, and copying things over to the right places, then patching the Windows binary from Linux, but the patch fails, I guess because it's obviously a different binary for Windows, so the stuff that needs patching is elsewhere

```
SteamVR proxy driver for Vive Pro 2
Consider supporting developer on patreon: https://patreon.com/0lach
SteamVR at /home/user/VivePro2-Linux-Driver/SteamVR
Found both original driver, and old original driver, seems like SteamVR was updated
= Moving updated original driver
= Patching real driver
# Driver requirement check
source pattern not found, file is already patched?
# Error message clarification
@2226238..2226258
-\x77\x68\x69\x63\x68\x20\x69\x73\x20\x6e\x6f\x74\x20\x70\x72\x65\x73\x65\x6e\x74
+\x69\x67\x6e\x6f\x72\x69\x6e\x67\x2e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
one or more rules failed
= Overriding current driver
= Updating proxy server
Installation finished, try to start SteamVR
```
(obviously, launching it doesn't work)

CertainLach, 2024-01-17:
Upload windows driver binary somewhere, I'll make a patch for it.

issacdowling, 2024-01-18:
They don't let me attach .dll files directly, but github issues allow zips to be attached.

Where on Linux, there's just a linux64 folder, on Windows, there's win32 and win64, both with driver_lighthouse.dll files. I presume only win64 is used, but maybe win32 is needed somewhere too since I doubt they're actually building SteamVR for use with 32-bit CPUs, so maybe there's some 32-bit Windows library needed or something?

I'll send both.
[win32 - driver_lighthouse.zip](https://github.com/CertainLach/VivePro2-Linux-Driver/files/13970692/win32.-.driver_lighthouse.zip)
[win64 - driver_lighthouse.zip](https://github.com/CertainLach/VivePro2-Linux-Driver/files/13970693/win64.-.driver_lighthouse.zip)

CertainLach, 2024-01-19:
Yes, win32 version is for x32-bit CPUs and for Steam itself
However, steam will work just fine without patching this driver, I'm not sure why it even uses those drivers in the client.

I don't quite like assembly the compiler generated for this function, the patch may be broken very soon, but this is good enough for now.

```
# Driver requirement check
-(?x-u)
-\x4c\x8b\x00\x48\x8b\xc8\x41\xff\x50\x10\x48\x85\xc0\x75\x39

+(?x)
+\x4c\x8b\x00\x48\x8b\xc8\x41\xff\x50\x10\x48\x85\xc0\xeb\x39

# Error message clarification
-(?x-u)
-\x77\x68\x69\x63\x68\x20\x69\x73\x20\x6e\x6f\x74\x20\x70\x72\x65\x73\x65\x6e\x74

+(?x)
+\x69\x67\x6e\x6f\x72\x69\x6e\x67\x2e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
```

mdovgialo, 2024-01-19:
I guess I would need to use `sewer` on linux to patch the lighthouse.dll and put compiled windows version and patched lighthouse.dll to steamvr distrubution folder?

I don't know where to even start to compile this driver for Windows. @issacdowling could you please share the compiled binaries?

CertainLach, 2024-01-19:
Sewer can be built for windows.
https://github.com/CertainLach/sewer

issacdowling, 2024-01-19:
Sadly, headset not detected, which I guess makes sense if it's trying to run lens server using wine, or any other Linux things like that

issacdowling, 2024-01-20:
```
Sat Jan 20 2024 13:28:27.822 [Info] - Unable to load driver lighthouse from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\lighthouse\bin\win64\driver_lighthouse.dll (193)
Sat Jan 20 2024 13:28:27.822 [Error] - ASSERT: "DriverManager lighthouse load error 193u" at c:\buildslave\steamvr_rel_win64\build\src\common\vrcommon\drivermanager.cpp:439.
```

```
Sat Jan 20 2024 13:28:27.980 [Error] - Unable to load driver lighthouse because of error VRInitError_Init_FileNotFound(103). Skipping.
```
This only happens with the patched driver. Maybe I'm missing some dependencies for running? I did build it on Linux for windows, rather than building from Windows

CertainLach, 2024-01-20:
Did you build it using mingw?
Then it will want windows build of pthread/possibly other things.

Better build it on windows using msvc, it shouldn't be hard with Rust.