CertainLach/VivePro2-Linux-Driver #32: Does not work on Fedora + patched 6.7.5 kernel

perillamint, 2024-02-22:
Hello, I'm currently trying to use VivePro2 Linux driver on Fedora 39, but unable to make it work.

## System specification
OS: Fedora Linux 39 (KDE Plasma) x86_64
Motherboard: AsRock X670E PG Lightning
Kernel: Tested on Vanilla 6.7.5 + patch and Fedora 6.7.5
CPU: AMD Ryzen 9 7950X (32) @ 5.881GHz 
GPU: AMD ATI Radeon RX 6900 XT

Tested version: 8982075490c242bfb291cf997c07ade37357e558, with following patch to make it compile

```
diff --git a/crates/vive-hid/src/lib.rs b/crates/vive-hid/src/lib.rs
index 060e7d8..2ce5c70 100644
--- a/crates/vive-hid/src/lib.rs
+++ b/crates/vive-hid/src/lib.rs
@@ -340,8 +340,8 @@ impl ViveDevice {
 
 #[test]
 fn test() -> Result<()> {
-       let dev = ViveDevice::open_first()?;
-       dbg!(dev.read_config()?);
-       dev.set_mode(1)?;
+//     let dev = ViveDevice::open_first()?;
+//     dbg!(dev.read_config()?);
+//     dev.set_mode(1)?;
        Ok(())
 }
```

Connection: Plugged linkbox to graphics card using bundled DP to miniDP cable.

## Expected behavior
HMD work as normal

## Actual behavior
* Right after SteamVR launch OR DP cable replug, display freezes
* SteamVR says critical component is crashed OR failed to open display.
* HMD LED stays red

## Note
Due to the kernel log, I sent the Vive Pro 2 to repair, but service center says HMD is perfectly fine.

## Relevant Logs
```
[   86.065721] usb 1-2.1.4.4: SerialNumber: 3645AAE3CC
[   86.093782] hid-generic 0003:28DE:2101.0012: hiddev100,hidraw15: USB HID v1.11 Device [Valve Software Watchman Dongle] on usb-0000:15:00.0-2.1.4.4/input0
[   87.024782] usb 1-2.1.3: new full-speed USB device number 12 using xhci_hcd
[   87.277379] usb 1-2.1.3: New USB device found, idVendor=28de, idProduct=2300, bcdDevice= 2.00
[   87.277384] usb 1-2.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   87.277385] usb 1-2.1.3: Product: LHR
[   87.277386] usb 1-2.1.3: Manufacturer: Valve
[   87.277387] usb 1-2.1.3: SerialNumber: LHR-46D3D1B2
[   87.338602] hid-generic 0003:28DE:2300.0013: hiddev101,hidraw16: USB HID v1.11 Device [Valve LHR] on usb-0000:15:00.0-2.1.3/input0
[   87.339227] hid-generic 0003:28DE:2300.0014: hiddev102,hidraw17: USB HID v1.11 Device [Valve LHR] on usb-0000:15:00.0-2.1.3/input1
[   87.339737] hid-generic 0003:28DE:2300.0015: hiddev103,hidraw18: USB HID v1.11 Device [Valve LHR] on usb-0000:15:00.0-2.1.3/input2
[   96.673916] ------------[ cut here ]------------
[   96.673919] WARNING: CPU: 11 PID: 731 at drivers/gpu/drm/amd/amdgpu/../display/dc/link/protocols/link_dp_training_8b_10b.c:260 perform_8b_10b_clock_recovery_sequence+0x1eb/0x260 [amdgpu]
[   96.674111] Modules linked in: uinput rfcomm snd_seq_dummy snd_hrtimer wireguard curve25519_x86_64 libcurve25519_generic ip6_udp_tunnel udp_tunnel nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib ip_set qrtr bnep sunrpc binfmt_misc snd_seq_midi snd_seq_midi_event pktcdvd intel_rapl_msr intel_rapl_common vfat snd_hda_codec_realtek fat snd_hda_codec_generic edac_mce_amd ledtrig_audio kvm_amd snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi kvm snd_usb_audio snd_hda_codec btusb uvcvideo btrtl snd_hda_core btintel snd_usbmidi_lib uvc videobuf2_vmalloc btbcm snd_ump videobuf2_memops btmtk videobuf2_v4l2 snd_rawmidi snd_hwdep snd_seq videobuf2_common snd_seq_device bluetooth snd_pcm irqbypass videodev snd_timer snd mc wmi_bmof rapl pcspkr k10temp i2c_piix4 rfkill soundcore razeraccessory(OE) joydev gpio_amdpt gpio_generic nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_masq nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4
[   96.674163]  nf_tables nfnetlink loop zram dm_crypt hid_logitech_hidpp amdgpu amdxcp i2c_algo_bit drm_ttm_helper crct10dif_pclmul ttm crc32_pclmul crc32c_intel polyval_clmulni polyval_generic drm_exec gpu_sched drm_suballoc_helper nvme ghash_clmulni_intel drm_buddy sha512_ssse3 drm_display_helper nvme_core ccp sha256_ssse3 sha1_ssse3 sp5100_tco r8169 nvme_auth cec realtek video wmi hid_logitech_dj nct6775 nct6775_core hwmon_vid i2c_dev fuse
[   96.674192] CPU: 11 PID: 731 Comm: kworker/11:1H Tainted: G           OE      6.7.5 #4
[   96.674194] Hardware name: ASRock X670E PG Lightning/X670E PG Lightning, BIOS 2.02 11/17/2023
[   96.674196] Workqueue: events_highpri dm_irq_work_func [amdgpu]
[   96.674408] RIP: 0010:perform_8b_10b_clock_recovery_sequence+0x1eb/0x260 [amdgpu]
[   96.674570] Code: 0f b6 44 24 1c 83 e2 0f 83 e0 0f 38 c2 0f 85 f8 fe ff ff 41 83 c7 01 41 83 ff 04 41 0f 96 c6 e9 f0 fe ff ff 41 83 fd 64 75 9a <0f> 0b 48 8b 85 d8 01 00 00 b9 64 00 00 00 48 c7 c2 00 2e f2 c0 48
[   96.674572] RSP: 0018:ffffbc67013afb90 EFLAGS: 00010246
[   96.674573] RAX: 0000000000000038 RBX: ffffbc67013afc40 RCX: 000000000000000e
[   96.674574] RDX: 0000000000000020 RSI: ffffbc67013afca4 RDI: ffffbc67013afc40
[   96.674575] RBP: ffffa04fbd171800 R08: 0000000000000000 R09: 0000000000000000
[   96.674576] R10: fffffff0ffffffff R11: 0000000000000004 R12: 0000000000000000
[   96.674577] R13: 0000000000000064 R14: 0000000000000001 R15: 0000000000000000
[   96.674578] FS:  0000000000000000(0000) GS:ffffa05e9dcc0000(0000) knlGS:0000000000000000
[   96.674579] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   96.674580] CR2: 00007f7a64e3a000 CR3: 0000000c70422000 CR4: 0000000000f50ef0
[   96.674581] PKRU: 55555554
[   96.674581] Call Trace:
[   96.674583]  <TASK>
[   96.674584]  ? perform_8b_10b_clock_recovery_sequence+0x1eb/0x260 [amdgpu]
[   96.674723]  ? __warn+0x81/0x130
[   96.674728]  ? perform_8b_10b_clock_recovery_sequence+0x1eb/0x260 [amdgpu]
[   96.674863]  ? report_bug+0x171/0x1a0
[   96.674868]  ? handle_bug+0x3c/0x80
[   96.674870]  ? exc_invalid_op+0x17/0x70
[   96.674872]  ? asm_exc_invalid_op+0x1a/0x20
[   96.674876]  ? perform_8b_10b_clock_recovery_sequence+0x1eb/0x260 [amdgpu]
[   96.675008]  dp_perform_8b_10b_link_training+0x54/0x180 [amdgpu]
[   96.675147]  dp_perform_link_training+0x19f/0x4c0 [amdgpu]
[   96.675364]  ? drm_dp_dpcd_access+0x101/0x130 [drm_display_helper]
[   96.675379]  dp_verify_link_cap_with_retries+0x20f/0x440 [amdgpu]
[   96.675584]  link_detect+0x452/0x500 [amdgpu]
[   96.675731]  handle_hpd_irq_helper+0xf9/0x170 [amdgpu]
[   96.675894]  process_one_work+0x171/0x340
[   96.675898]  worker_thread+0x27b/0x3a0
[   96.675901]  ? __pfx_worker_thread+0x10/0x10
[   96.675902]  kthread+0xe5/0x120
[   96.675904]  ? __pfx_kthread+0x10/0x10
[   96.675906]  ret_from_fork+0x31/0x50
[   96.675909]  ? __pfx_kthread+0x10/0x10
[   96.675910]  ret_from_fork_asm+0x1b/0x30
[   96.675914]  </TASK>
[   96.675915] ---[ end trace 0000000000000000 ]---
[   96.675917] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
[  100.172931] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
[  103.670379] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
[  107.167580] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
[  110.659608] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
[  116.830150] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
[  120.302726] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
[  123.726669] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
[  127.196719] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
[  130.655389] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
[  134.143429] amdgpu 0000:03:00.0: [drm] *ERROR* perform_8b_10b_clock_recovery_sequence: Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue
```

perillamint, 2024-02-22:
I suspect a hardware fault (because it failed to train DP), but I don't have any extra hardware (other graphics card OR another set of Vive Pro 2) to bisect the fault location...

CertainLach, 2024-02-22:
> Another notable thing is, UEFI firmware freezes when I try to boot the system with initialized Vive Pro 2

This is somewhat ok, my motherboard tries to render its UI on HMD with huge resolution too.

CertainLach, 2024-02-22:
> Link Training Error, could not get CR after 100 tries. Possibly voltage swing issue

Did you changed your display mode to 4/5 in the config file? It requires gpu in the special mode, which doesn't work OOB, because I haven't managed to finish submitting this patch to the upstream: https://lore.kernel.org/linux-kernel/20220220151940.58327-1-iam@lach.pw/

Modes 0-3 should work, and then it is possible hardware failure. Can you test if your HMD works under windows?

perillamint, 2024-02-23:
@CertainLach No, I stayed in 0-3, also tried 4, but result was same.

I think I can try it on Windows after I plug in extra SSD in my system (this weekend maybe)

perillamint, 2024-02-25:
@CertainLach Tested it on Windows 11, With latest Radeon Adrenaline driver.

The headset LED switches to green, but desktop freezes (I guess the reason is same as Linux, AMD DC freaked out of PHY issue)

Now, it is clear: I need to physically visit them WITH my graphics card and I also suspects they didn't properly tested LinkBox

CertainLach, 2024-02-25:
I don't think the reason is in the graphics card.
You can try to reseat all the cables, including link cable on the headset side.

perillamint, 2024-03-11:
Dropping an update here: The service technician requested me to submit the log using a Windows driver. I did that today and will post here after update.