CertainLach/VivePro2-Linux-Driver #33: Nobara 39 - Kernel 6.8.5-201 - Lens distortion helper is failed to launch

ArchangelJTW, 2024-04-21:
Error:
`Lens distortion helper is failed to launch, HMD image most probaly will be distorted and unusable.
Error: lens protocol error: io error: failed to fill whole buffer

Make sure you have any recent version of proton installed.`

I have Proton Experimental, 9.0 beta, and 6.3 installed.

I was previously having an issue with wine not being found in path.  I investigated the code and found how it was attempting to attain the path, so I put WINE onto my global path (apparently you can also pass it as a launch option on steam).  Regardless, I got past that first problem but have ran into the problem above.  

Currently looking into it, help appreciated.  If there are additional logs or other information that would be helpful, let me know, and I will provide them.

joni999, 2024-04-28:
I tried the SteamVR beta as well and have the same issue.
I found this in the /home/joni/.steam/steam/logs/vrstartup-linux.txt log:
```
fsync: up and running.
^[[32m INFO^[[0m ^[[2mlens_server^[[0m^[[2m:^[[0m hello from lens server
^[[32m INFO^[[0m ^[[2mlens_server^[[0m^[[2m:^[[0m dll path: "Z:\\home\\joni\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\LibLensDistortion.dll"
00d4:err:module:import_dll Library libvkd3d-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
00d4:err:module:import_dll Library libvkd3d-shader-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
00d4:err:module:import_dll Library wined3d.dll (which is needed by L"C:\\windows\\system32\\dxgi.dll") not found
00d4:err:module:import_dll Library dxgi.dll (which is needed by L"C:\\windows\\system32\\d3d11.dll") not found
00d4:err:module:import_dll Library libvkd3d-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
00d4:err:module:import_dll Library libvkd3d-shader-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
00d4:err:module:import_dll Library wined3d.dll (which is needed by L"C:\\windows\\system32\\d3d11.dll") not found
00d4:err:module:import_dll Library d3d11.dll (which is needed by L"Z:\\home\\joni\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\opencv_world346.dll") not found
00d4:err:module:import_dll Library opencv_world346.dll (which is needed by L"Z:\\home\\joni\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\LibLensDistortion.dll") not found
^[[31mERROR^[[0m ^[[2mlens_server^[[0m^[[2m:^[[0m ^[[3merror^[[0m^[[2m=^[[0mfailed to load library
Error: failed to load library

Caused by:
    0: LoadLibraryExW failed
    1: Module not found. (os error 126)
```
Maybe those library errors are causing the lens_server to die prematurely.

By the way: 
Have you tried running the "regular" SteamVR version? It works without this error for me. I don't have to use temp_v1.27.5

Full logfile:
[vrstartup-linux.beta_errors.txt](https://github.com/CertainLach/VivePro2-Linux-Driver/files/15141961/vrstartup-linux.beta_errors.txt)

CertainLach, 2024-04-28:
I think your wine installation is somehow broken
> import_dll Library d3d11.dll (which is needed by L"Z:\\home\\joni\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\opencv_world346.dll") not found

d3d11.dll is installed in proton by default, or you're using external wine, and not proton?

CertainLach, 2024-04-28:
Looks like there was some changes in wine, breaking existing prefixes, in the past:
https://bugzilla.redhat.com/show_bug.cgi?id=2107290

Try to drop SteamVR wineprefix: `rm -rf ~/.local/share/Steam/steamapps/compatdata/250820`

joni999, 2024-04-28:
> Try to use proton (remove WINE environment variable from SteamVR

I don't think I have that set anywhere - Where would that normally be?
My SteamVR launch options are empty and I'm not forcing a compatibility tool in the compatibility Tab of the SteamVR properties.

I have proton installed:

[image]

And I removed the prefix:
`[joni@linuxjoni04 logs]$ rm -rf ~/.local/share/Steam/steamapps/compatdata/250820`

After running SteamVR Beta again the proton prefix directory is re-created, but the errors in the logfile also seem to remain the same.

```
wine: created the configuration directory '/home/joni/.local/share/Steam/steamapps/compatdata/250820/pfx'
fsync: up and running.
004c:err:ole:StdMarshalImpl_MarshalInterface Failed to create ifstub, hr 0x80004002
004c:err:ole:CoMarshalInterface Failed to marshal the interface {6d5140c1-7436-11ce-8034-00aa006009fa}, hr 0x80004002
004c:err:ole:apartment_get_local_server_stream Failed: 0x80004002
0058:err:ole:com_get_class_object class {e22ad333-b25f-460c-83d0-0581107395c9} not registered
0058:err:ole:com_get_class_object no class object {e22ad333-b25f-460c-83d0-0581107395c9} could be created for context 0x1
0058:err:tabtip:add_uia_event_handler Failed to create IUIAutomation interface, hr 0x80040154
wine: configuration in L"/home/joni/.local/share/Steam/steamapps/compatdata/250820/pfx" has been updated.
^[[32m INFO^[[0m ^[[2mlens_server^[[0m^[[2m:^[[0m hello from lens server
^[[32m INFO^[[0m ^[[2mlens_server^[[0m^[[2m:^[[0m dll path: "Z:\\home\\joni\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\LibLensDistortion.dll"
0170:err:module:import_dll Library libvkd3d-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
0170:err:module:import_dll Library libvkd3d-shader-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
0170:err:module:import_dll Library wined3d.dll (which is needed by L"C:\\windows\\system32\\dxgi.dll") not found
0170:err:module:import_dll Library dxgi.dll (which is needed by L"C:\\windows\\system32\\d3d11.dll") not found
0170:err:module:import_dll Library libvkd3d-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
0170:err:module:import_dll Library libvkd3d-shader-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
0170:err:module:import_dll Library wined3d.dll (which is needed by L"C:\\windows\\system32\\d3d11.dll") not found
0170:err:module:import_dll Library d3d11.dll (which is needed by L"Z:\\home\\joni\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\opencv_world346.dll") not found
0170:err:module:import_dll Library opencv_world346.dll (which is needed by L"Z:\\home\\joni\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\LibLensDistortion.dll") not found
^[[31mERROR^[[0m ^[[2mlens_server^[[0m^[[2m:^[[0m ^[[3merror^[[0m^[[2m=^[[0mfailed to load library
Error: failed to load library

Caused by:
    0: LoadLibraryExW failed
    1: Module not found. (os error 126)
Gtk-Message: 16:20:22.010: Failed to load module "xapp-gtk3-module"
Gtk-Message: 16:20:22.010: Failed to load module "canberra-gtk-module"
```

Thank you for helping us here!

joni999, 2024-05-01:
I think my issues before were completely unrelated... sorry!
I switched my /home filesystem from XFS to EXT4 and that cleared many of my SteamVR problems. including this one. (Maybe you could check that as well @ArchangelJTW)

The LibLensDistortion.exe even seems to start up now (is it normal. that there are two windows for it?) -

But I get "SteamVR Fail" error now as visible in the screenshot. And the HMD doesn't seem to connect anymore.
I also completely reinstalled SteamVR beta again and deleted the proton prefix - this time I even used the Patreon downloaded installer for the driver.

[image]

I found a pretty close description of my issue in another thread but it doesn't seem like it was resolved:
https://github.com/CertainLach/VivePro2-Linux-Driver/issues/24#issuecomment-1817466739

I also have those 
`qt.network.ssl:` errors
 as well as:
`[tracing-subscriber] Unable to write an event to the Writer for this Subscriber! Error: failed to write whole buffer`

The guy in the other thread didn't provide a system report. Does mine maybe help in any way?
[SteamVR-2024-05-01-PM_03_18_18.txt](https://github.com/CertainLach/VivePro2-Linux-Driver/files/15177444/SteamVR-2024-05-01-PM_03_18_18.txt)

I'm on Steam Client Beta version 1714520927 and SteamVR Beta 2.5.3

CertainLach, 2024-05-01:
That's strange. Lens server should not launch in its own window, it should run in console mode, and proxy driver talks with it using stdin/stdout
If it has launched in separate window, then proxy driver won't be able to communicate with it.

joni999, 2024-05-01:
I think I got it.
Can someone try with and without Proton 9 Beta installed? I also had to remove some remains of it manually after uninstalling for it to work (otherwise gives a "wine not found" error):
`[joni@linuxjoni04 logs]$ rm -r /home/joni/.local/share/Steam/steamapps/common/Proton\ 9.0\ \(Beta\)/`

I just uninstalled it, and I think it works now.

grep in logs directory to see that it actually pulls proton 8 instead of 9:

[image]

CertainLach, 2024-05-01:
Huh, so it's broken on proton 9 for you?
It works for me

joni999, 2024-05-01:
Yep and even reinstalling Proton 9 doesn't change it.
With Proton 9 the lens server always starts in a separate window for me. No idea why.
But uninstalling Proton 9 is a valid workaround for me. Maybe @ArchangelJTW's issue is similar to mine or maybe I'm just the odd man out. Thank you for the assist!

ArchangelJTW, 2024-05-05:
File system settings haven't made a difference for me, I'm leaving mine on BTRFS for now.  I've been able to play steamVR fine with the current setup, I tried regular stable steamvr version and it worked still, beta version does not work.  I have proton 9 installed.  

If I remove my WINE path definitions the lens server stops working, the lens server only works with wine for me.  It seems it either cannot find proton or cannot use it, reinstalls don't seem to help.  The `failed to fill whole buffer` error almost seems like it might be a proton error but I am unsure.

tek, 2024-05-12:
after this week's SteamVR update (2.5) I'm getting the same symptoms – the mentioned error popup and an upside down image.

Also these errors:

```
 INFO lens_server: hello from lens server
 INFO lens_server: dll path: "Z:\\home\\tek\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\LibLensDistortion.dll"
0178:err:module:import_dll Library libvkd3d-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
0178:err:module:import_dll Library libvkd3d-shader-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
0178:err:module:import_dll Library wined3d.dll (which is needed by L"C:\\windows\\system32\\dxgi.dll") not found
0178:err:module:import_dll Library dxgi.dll (which is needed by L"C:\\windows\\system32\\d3d11.dll") not found
0178:err:module:import_dll Library libvkd3d-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
0178:err:module:import_dll Library libvkd3d-shader-1.dll (which is needed by L"C:\\windows\\system32\\wined3d.dll") not found
0178:err:module:import_dll Library wined3d.dll (which is needed by L"C:\\windows\\system32\\d3d11.dll") not found
0178:err:module:import_dll Library d3d11.dll (which is needed by L"Z:\\home\\tek\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\opencv_world346.dll") not found
0178:err:module:import_dll Library opencv_world346.dll (which is needed by L"Z:\\home\\tek\\.local\\share\\Steam\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\LibLensDistortion.dll") not found
ERROR lens_server: error=failed to load library
Error: failed to load library

Caused by:
    0: LoadLibraryExW failed
    1: Module not found. (os error 126)
```

WorldofShadows, 2024-06-08:
Hey there!
First of all: Thank you CertainLach for your hard work regarding this :)
I'm completely new to this and just trying to switch over to Arch from W10 and tried to get my VP2 running with this driver.
I sadly get similar results as OP.
The lensserver can't start because it can't find proton or wine on the system. Steam was installed using pacman, proton-experimental and proton 8.0 are installed through steam, wine-staging is installed through yay.
System is as fresh as it could be, kernel is custom compiled 6.9.3 with all of your patches applied.

I subscribed to your patreon to support you and to try the binary in case i did something wrong at compiling the driver myself, but it leads to the same result. I can't compile it via nix, as this fails at detecting the HMD(something with eid).

I tried adding WINE to /etc/enviroment adding it to .profile,  and adding it to the launch arguments for SteamVR, but neither worked for me, the Lensserver can't start when triggered through SteamVR.
When launching it standalone with wine64 it starts normally, but isn't recognized by SteamVR/the driver proxy.

Log entry:
```
Sat Jun 08 2024 20:05:00.703945 [Info] - lighthouse: [32m INFO[0m [2mlens_client[0m[2m:[0m using lens server at "/home/wolf/.local/share/Steam/steamapps/common/SteamVR/drivers/lighthouse/bin/linux64/lens-server/lens-server.exe"
Sat Jun 08 2024 20:05:00.703995 [Info] - lighthouse: [32m INFO[0m [2mlens_client[0m[2m:[0m trying /home/wolf/.local/share/Steam/steamapps/common/Proton 8.0/proton as proton
Sat Jun 08 2024 20:05:00.704305 [Info] - lighthouse: [32m INFO[0m [2mlens_client[0m[2m:[0m trying wine64 as wine
Sat Jun 08 2024 20:05:00.704555 [Info] - lighthouse: [32m INFO[0m [2mlens_client[0m[2m:[0m trying wine as wine
Sat Jun 08 2024 20:05:02.512011 [Info] - lighthouse: [32m INFO[0m [2mdriver_proxy::server::driver_host[0m[2m:[0m zenity finished: exit status: 0
Sat Jun 08 2024 20:05:02.512068 [Info] - lighthouse: []
Sat Jun 08 2024 20:05:02.512083 [Info] - lighthouse: []
Sat Jun 08 2024 20:05:02.512104 [Info] - lighthouse: [31mERROR[0m [2mdriver_proxy::server::driver_host[0m[2m:[0m lens server start failed: can't find wine in PATH

```
Also prominent in the logs is: 
`[tracing-subscriber] Unable to write an event to the Writer for this Subscriber! Error: failed to write whole buffer`

I'm at a loss here.

Other than that when set to 0 VR loads, but everything is upside down. When set to something other than 0 i get static on both displays.

System: Ryzen 9 5950X / 64 gigs of RAM / RTX 3080TI
Arch with 6.9.3 custom kernel with your VP2 Patches inside.
Nvidia Driver: 555.42.02 Beta, drm and fdev settings are set in modprobe.

Can i provide any logs to sort this out/fix this or do you have any idea whats going on or what i can try?
Did Valve change something to break this?

Ty :)

Bixilon, 2024-07-11:
So, I am in a similar state.

When starting SteamVR through steam, it can not find or load any dlls in wine. Steam is messing something up.

When just starting SteamVR directly (without involving steam) and using proton, I get the same result (failed to load xx). When using `export WINE=/bin/wine` to use the system (debian sid) wine installation, I get further. 
`[tracing-subscriber] Unable to write an event to the Writer for this Subscriber! Error: failed to write whole buffer`  is spammed, and it broken. Either it fails with 

```
ERROR lens_server: error=failed to read request
Error: failed to read request

Caused by:
    0: io error: failed to fill whole buffer
    1: failed to fill whole buffer
```

Or errors with `Error: io error: No such file or directory (os error 2)` (when starting through steam).

It certainly seems to do something:
```
Using breakpad crash handler
Setting breakpad minidump AppID = 250820
Forcing breakpad minidump interfaces to load
Looking up breakpad interfaces from steamclient
Calling BreakpadMiniDumpSystemInit
07/11 18:52:41 minidumps folder is set to /tmp/dumps
07/11 18:52:41 Init: Installing breakpad exception handler for appid(250820)/version(1.27.5)/tid(405318)
factory call
 INFO driver_proxy::factory: https://patreon.com/0lach
[tracing-subscriber] Unable to write an event to the Writer for this Subscriber! Error: failed to write whole buffer

 INFO driver_proxy::factory: requested interface: "IServerTrackedDeviceProvider_004"
[tracing-subscriber] Unable to write an event to the Writer for this Subscriber! Error: failed to write whole buffer

 INFO driver_proxy::driver::server_tracked_provider: intializing server tracker provider
```

I tried it with the latest steam vr release (something 2.7.x) and with the old 1.27.5 version, both fail. Also detecting wine/proton automatically is broken (probably due to steam changes).

@CertainLach can you please help with that?

alexjc, 2024-10-16:
I managed to avoid Steam updates until today, but getting the latest update today also broke this repository with the same issue. The device is basically not usable on Linux anymore...

ArchangelJTW, 2024-10-16:
> Hey there!
> First of all: Thank you CertainLach for your hard work regarding this :)
> I'm completely new to this and just trying to switch over to Arch from W10 and tried to get my VP2 running with [...]

Just to update, I now and still have this issue to this day.

CertainLach, 2026-09-30:
On master I have dropped proton and lens-server usage in favor of [champagne](https://delta.rocks/blog/lach-stuff/champagne), it should no longer fail randomly

closed 2026-09-30 by CertainLach