CertainLach/VivePro2-Linux-Driver #34: Status of DSC patches?

Supreeeme, 2024-05-01:
Hi, the DSC patch is also needed for the Bigscreen Beyond, so thanks a lot for it! I was wondering, what's the status of it getting upstreamed?

CertainLach, 2024-05-01:
I need to redo it properly (In current form, it is somewhat ad-hoc and too specific, I.e those timings should only be applied to IV timings, but not others, and necessary information is not properly preserved between kernel and amdgpu dc core yet), I'm tracking some refactorings to be finished upstream.

thetaepsilon, 2024-10-08:
Any update on said refactors a few months later? In particular, if you know of any versions where relevant work was done / merged, I think that'd be great information to share if you're able. Would also be great if you know what's left and could briefly describe it. No pressure in either case.