CertainLach/VivePro2-Linux-Driver #35: GLIBC Version ERROR

Songcc87, 2024-05-22:
I used nix build, However, the result of each compilation will show "Version GLIBC not found". How to change the GLIBC version that nix build depends on.
[image]

CertainLach, 2024-05-22:
Glibc 2.33 is 3 years old, 2.34 is 6 months newer.
Nix build of this driver requires 2.34+, and will work on this list of distributions (Based on https://repology.org/project/glibc/versions):

- Manjaro/ArchLinux - yes
- NixOS 22.05+
- Fedora 35+
- Debian 12 (Bookworm)
- Ubuntu 22.04+

Do yourself a favor, upgrade your system, it will make your life easier. Older system version may also miss other performance improvements in SteamVR itself.

Songcc87, 2024-05-23:
"GLIBC_2.36 not found" error will also be reported on ubuntu22.04, and it will only start normally on ubuntu24.04.