CertainLach/VivePro2-Linux-Driver #39: Fedora 40, kernel 6.9.9-200 (release) patches 0002 and 0003 do not apply.

Gregregorovich, 2024-07-19:
Running Fedora 40, patching the latest kernel released to the package manager (6.9.9-200) does not apply at least the second patch.

Hardware (presumably it's relevant?) :
* R9 7900X (kernel driver `amdgpu`, has integrated Raphael graphics)
* AMD RX 5600 XT (kernel driver `amdgpu`, second x16 slot, PCIe Gen IV x4 - Linux GPU, what I'm trying to get the VP2 to work on, default GPU for KDE 6 on Wayland)
* (RTX 3080 Ti - kernel driver `vfio-pci`, primary slot, PCIe Gen IV x16, Windows KVM GPU)

Output of `fedpkg --release f40 local` is as follows:
```
Not downloading unused kernel-abi-stablelists-6.6.0.tar.bz2
Not downloading unused kernel-kabi-dw-6.6.0.tar.bz2
Not downloading already downloaded linux-6.9.9.tar.xz
Not downloading already downloaded kernel-abi-stablelists-6.9.9.tar.xz
Not downloading already downloaded kernel-kabi-dw-6.9.9.tar.xz

setting SOURCE_DATE_EPOCH=1720656000
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.raxI6p
+ umask 022
+ cd /home/jarvis/kernel
kernel.spec:1730: Start of prep stage
kernel.spec:1732: Sanity checks
+ patch_command='git --work-tree=. apply'
kernel.spec:1789: Untar kernel tarball
+ cd /home/jarvis/kernel
+ rm -rf kernel-6.9.9
+ /usr/bin/mkdir -p kernel-6.9.9
+ cd kernel-6.9.9
+ /usr/lib/rpm/rpmuncompress -x /home/jarvis/kernel/linux-6.9.9.tar.xz
+ STATUS=0
+ '[' 0 -ne 0 ']'
+ rm -rf /home/jarvis/kernel/kernel-6.9.9-SPECPARTS
+ /usr/bin/mkdir -p /home/jarvis/kernel/kernel-6.9.9-SPECPARTS
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ mv linux-6.9.9 linux-6.9.9-200.fc40.x86_64
+ cd linux-6.9.9-200.fc40.x86_64
+ cp -a /home/jarvis/kernel/Makefile.rhelver .
kernel.spec:1796: Start of patch applications
+ ApplyOptionalPatch patch-6.9-redhat.patch
+ local patch=patch-6.9-redhat.patch
+ shift
kernel.spec:1779: ApplyOptionalPatch:
+ '[' '!' -f /home/jarvis/kernel/patch-6.9-redhat.patch ']'
++ wc -l /home/jarvis/kernel/patch-6.9-redhat.patch
++ awk '{print $1}'
+ local C=1775
+ '[' 1775 -gt 9 ']'
+ ApplyPatch patch-6.9-redhat.patch
+ local patch=patch-6.9-redhat.patch
+ shift
+ '[' '!' -f /home/jarvis/kernel/patch-6.9-redhat.patch ']'
+ case "$patch" in
+ git --work-tree=. apply
+ ApplyOptionalPatch 0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ local patch=0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ shift
kernel.spec:1779: ApplyOptionalPatch:
+ '[' '!' -f /home/jarvis/kernel/0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch ']'
++ wc -l /home/jarvis/kernel/0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
++ awk '{print $1}'
+ local C=30
+ '[' 30 -gt 9 ']'
+ ApplyPatch 0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ local patch=0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ shift
+ '[' '!' -f /home/jarvis/kernel/0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch ']'
Patch9999: 0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ case "$patch" in
+ git --work-tree=. apply
+ ApplyOptionalPatch 0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch
+ local patch=0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch
+ shift
kernel.spec:1779: ApplyOptionalPatch:
+ '[' '!' -f /home/jarvis/kernel/0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch ']'
++ wc -l /home/jarvis/kernel/0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch
++ awk '{print $1}'
+ local C=128
+ '[' 128 -gt 9 ']'
+ ApplyPatch 0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch
+ local patch=0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch
+ shift
+ '[' '!' -f /home/jarvis/kernel/0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch ']'
Patch10000: 0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch
+ case "$patch" in
+ git --work-tree=. apply
error: patch failed: drivers/gpu/drm/drm_edid.c:6376
error: drivers/gpu/drm/drm_edid.c: patch does not apply
error: patch failed: include/drm/drm_connector.h:721
error: include/drm/drm_connector.h: patch does not apply
error: patch failed: include/drm/drm_displayid.h:131
error: include/drm/drm_displayid.h: patch does not apply
error: Bad exit status from /var/tmp/rpm-tmp.raxI6p (%prep)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.raxI6p (%prep)
```

Commenting out patch 0002-[...] and patch 0003 also fails to apply.

```
[...]
Patch9999: 0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ case "$patch" in
+ git --work-tree=. apply
+ ApplyOptionalPatch 0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+ local patch=0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+ shift
kernel.spec:1779: ApplyOptionalPatch:
+ '[' '!' -f /home/jarvis/kernel/0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch ']'
++ wc -l /home/jarvis/kernel/0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
++ awk '{print $1}'
+ local C=59
+ '[' 59 -gt 9 ']'
+ ApplyPatch 0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+ local patch=0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+ shift
+ '[' '!' -f /home/jarvis/kernel/0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch ']'
Patch10001: 0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+ case "$patch" in
+ git --work-tree=. apply
error: patch failed: drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm_helpers.c:87
error: drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm_helpers.c: patch does not apply
error: Bad exit status from /var/tmp/rpm-tmp.QxaViC (%prep)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.QxaViC (%prep)
```

Gregregorovich, 2024-07-19:
As to how I tried to compiled the kernel, I followed the following guides:
[https://forum.level1techs.com/t/compile-fedora-kernel-the-fedora-way/149242](https://forum.level1techs.com/t/compile-fedora-kernel-the-fedora-way/149242)
[https://odysee.com/@mumblingdrunkard:7/20210405-fedora-patch-compile-and-install-kernel:0](https://odysee.com/@mumblingdrunkard:7/20210405-fedora-patch-compile-and-install-kernel:0)
except as I want to run the current release kernel, instead of `git checkout origin/f[0-9]*` I instead run `git checkout [commit-for-current-release]`, which can be found from clicking on the link in the stable version column of [https://src.fedoraproject.org/rpms/kernel](https://src.fedoraproject.org/rpms/kernel) .

As stated above, I compiled with `fedpkg --release f[current-release-number] local` as `git branch -u [commit-for-current-release]` won't work as the commit is not a branch.

RononDex, 2024-08-04:
You can find a fixed 002 patch here https://github.com/santeri3700/vive-pro-2-on-linux/tree/master/kernel

Gregregorovich, 2024-08-09:
Fedora 40 released kernel 6.10.3-200 4 days ago, the release prior to that was krnel 6.9.12-200.

The fixed 0002 patch is supposedly for kernel 6.10.2 (on Arch, presumably), and still doesn't apply. Presumably fedora's kernel is sufficiently different from arch? Output is as follows:

```
fedpkg --release f40 local
Not downloading unused kernel-abi-stablelists-6.6.0.tar.bz2
Not downloading unused kernel-kabi-dw-6.6.0.tar.bz2
Downloading linux-6.10.3.tar.xz from https://src.fedoraproject.org/repo/pkgs
######################################################################## 100.0%
Downloading kernel-abi-stablelists-6.10.3.tar.xz from https://src.fedoraproject.org/repo/pkgs
######################################################################## 100.0%
Downloading kernel-kabi-dw-6.10.3.tar.xz from https://src.fedoraproject.org/repo/pkgs
######################################################################## 100.0%

setting SOURCE_DATE_EPOCH=1722816000
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.baNIyR
+ umask 022
+ cd /home/jarvis/kernel
kernel.spec:1748: Start of prep stage
kernel.spec:1750: Sanity checks
+ patch_command='git --work-tree=. apply'
kernel.spec:1807: Untar kernel tarball
+ cd /home/jarvis/kernel
+ rm -rf kernel-6.10.3
+ /usr/bin/mkdir -p kernel-6.10.3
+ cd kernel-6.10.3
+ /usr/lib/rpm/rpmuncompress -x /home/jarvis/kernel/linux-6.10.3.tar.xz
+ STATUS=0
+ '[' 0 -ne 0 ']'
+ rm -rf /home/jarvis/kernel/kernel-6.10.3-SPECPARTS
+ /usr/bin/mkdir -p /home/jarvis/kernel/kernel-6.10.3-SPECPARTS
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ mv linux-6.10.3 linux-6.10.3-200.fc40.x86_64
+ cd linux-6.10.3-200.fc40.x86_64
+ cp -a /home/jarvis/kernel/Makefile.rhelver .
kernel.spec:1814: Start of patch applications
+ ApplyOptionalPatch patch-6.10-redhat.patch
+ local patch=patch-6.10-redhat.patch
+ shift
kernel.spec:1797: ApplyOptionalPatch:
+ '[' '!' -f /home/jarvis/kernel/patch-6.10-redhat.patch ']'
++ wc -l /home/jarvis/kernel/patch-6.10-redhat.patch
++ awk '{print $1}'
+ local C=1970
+ '[' 1970 -gt 9 ']'
+ ApplyPatch patch-6.10-redhat.patch
+ local patch=patch-6.10-redhat.patch
+ shift
+ '[' '!' -f /home/jarvis/kernel/patch-6.10-redhat.patch ']'
+ case "$patch" in
+ git --work-tree=. apply
+ ApplyOptionalPatch 0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ local patch=0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ shift
kernel.spec:1797: ApplyOptionalPatch:
+ '[' '!' -f /home/jarvis/kernel/0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch ']'
++ wc -l /home/jarvis/kernel/0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
++ awk '{print $1}'
+ local C=30
+ '[' 30 -gt 9 ']'
+ ApplyPatch 0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ local patch=0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ shift
+ '[' '!' -f /home/jarvis/kernel/0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch ']'
Patch9999: 0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ case "$patch" in
+ git --work-tree=. apply
+ ApplyOptionalPatch 0002-drm-edid-parse-DRM-VESA-dsc-bpp-target_fix.patch
+ local patch=0002-drm-edid-parse-DRM-VESA-dsc-bpp-target_fix.patch
+ shift
kernel.spec:1797: ApplyOptionalPatch:
+ '[' '!' -f /home/jarvis/kernel/0002-drm-edid-parse-DRM-VESA-dsc-bpp-target_fix.patch ']'
++ wc -l /home/jarvis/kernel/0002-drm-edid-parse-DRM-VESA-dsc-bpp-target_fix.patch
++ awk '{print $1}'
+ local C=108
+ '[' 108 -gt 9 ']'
+ ApplyPatch 0002-drm-edid-parse-DRM-VESA-dsc-bpp-target_fix.patch
+ local patch=0002-drm-edid-parse-DRM-VESA-dsc-bpp-target_fix.patch
+ shift
+ '[' '!' -f /home/jarvis/kernel/0002-drm-edid-parse-DRM-VESA-dsc-bpp-target_fix.patch ']'
Patch10000: 0002-drm-edid-parse-DRM-VESA-dsc-bpp-target_fix.patch
+ case "$patch" in
+ git --work-tree=. apply
error: patch failed: drivers/gpu/drm/drm_edid.c:6353
error: drivers/gpu/drm/drm_edid.c: patch does not apply
error: patch failed: include/drm/drm_connector.h:720
error: include/drm/drm_connector.h: patch does not apply
error: include/drm/drm_displayid.h: No such file or directory
error: Bad exit status from /var/tmp/rpm-tmp.baNIyR (%prep)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.baNIyR (%prep)
```

And patch 0003 is still not applying anyway with 0002 commented out (though maybe once 0002 is patched 0003 will magically work?)

```
[...]
Patch9999: 0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ case "$patch" in
+ git --work-tree=. apply
+ ApplyOptionalPatch 0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+ local patch=0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+ shift
kernel.spec:1797: ApplyOptionalPatch:
+ '[' '!' -f /home/jarvis/kernel/0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch ']'
++ wc -l /home/jarvis/kernel/0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
++ awk '{print $1}'
+ local C=59
+ '[' 59 -gt 9 ']'
+ ApplyPatch 0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+ local patch=0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+ shift
+ '[' '!' -f /home/jarvis/kernel/0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch ']'
Patch10001: 0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+ case "$patch" in
+ git --work-tree=. apply
error: patch failed: drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm_helpers.c:87
error: drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm_helpers.c: patch does not apply
error: Bad exit status from /var/tmp/rpm-tmp.Ge6EP7 (%prep)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.Ge6EP7 (%prep)
```

Gregregorovich, 2024-08-09:
Just applying patch 0001 does change what happens to SteamVR: Steam (and SteamVR) no longer detect that there is an HMD plugged in; no icon in the top right of Steam, no headset (or controller/lighthouse). Tried multiple USB 3.0 controllers via no hubs; two different controllers direct-to-CPU USB 3.x 10Gb ports (see 7900X block diagrams) and one USB 3.x 5Gb port from the chipset. Motherboard is the MSI X670-P WiFi for reference.

<details>
<summary>Exceprts from `lsusb -tv` (collapsed spoiler)</summary>

USB 3.x 10Gb CPU No. 1
```
[...]
/:  Bus 007.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 002: Dev 006, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 0bb4:0343 HTC (High Tech Computer Corp.)
        |__ Port 002: Dev 007, If 0, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
        |__ Port 002: Dev 007, If 1, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
        |__ Port 002: Dev 007, If 2, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
        |__ Port 002: Dev 007, If 3, Class=Application Specific Interface, Driver=[none], 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
/:  Bus 008.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 002: Dev 004, If 0, Class=Hub, Driver=hub/1p, 5000M
        ID 0bb4:0343 HTC (High Tech Computer Corp.)
[...]
```

USB 3.x 10Gb CPU No. 2
```
/:  Bus 009.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    [...]
    |__ Port 002: Dev 016, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 0bb4:0343 HTC (High Tech Computer Corp.)
        |__ Port 002: Dev 017, If 0, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
        |__ Port 002: Dev 017, If 1, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
        |__ Port 002: Dev 017, If 2, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
        |__ Port 002: Dev 017, If 3, Class=Application Specific Interface, Driver=[none], 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
/:  Bus 010.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 002: Dev 008, If 0, Class=Hub, Driver=hub/1p, 5000M
        ID 0bb4:0343 HTC (High Tech Computer Corp.)
```

USB 3.x 5Gb Chipset
```
/:  Bus 005.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    [...]
    |__ Port 005: Dev 023, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 0bb4:0343 HTC (High Tech Computer Corp.)
        |__ Port 002: Dev 024, If 0, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
        |__ Port 002: Dev 024, If 1, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
        |__ Port 002: Dev 024, If 2, Class=Vendor Specific Class, Driver=btusb, 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
        |__ Port 002: Dev 024, If 3, Class=Application Specific Interface, Driver=[none], 12M
            ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
    [...]
/:  Bus 006.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/5p, 20000M/x2
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 005: Dev 004, If 0, Class=Hub, Driver=hub/1p, 5000M
        ID 0bb4:0343 HTC (High Tech Computer Corp.)

```
</details>

RononDex, 2024-08-12:
I think the important bit is `include/drm/drm_displayid.h: No such file or directory`

You can find the proper path this file is under on your kernel src tree and update the patch (just edit it in a text editor)

Gregregorovich, 2024-08-12:
Aha! Thanks <3 . That makes sense. Now I know where to start I'll look into it some time this week, and maybe I'll finally get it working on Fedora :)

Gregregorovich, 2024-12-18:
FWIW I've written a patch file for F40 (haven't upgraded to F41 yet). I've been meaning to fork this and put my patch there, but I haven't done much testing for it or added support for the BigScreen Beyond yet (which should just require a few additional kernel lines).

`kernel.spec.patch` contains as follows:

```
--- kernel.spec.old	2024-11-10 16:41:26.110148883 +0000
+++ kernel.spec	2024-11-09 01:28:19.723086738 +0000
@@ -1018,6 +1018,10 @@
 Patch1: patch-%{patchversion}-redhat.patch
 %endif
 
+Patch10001: 0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+Patch10002: 0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch
+Patch10003: 0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+
 # empty final patch to facilitate testing of kernel patches
 Patch999999: linux-kernel-test.patch
 
@@ -1862,6 +1866,10 @@
 
 ApplyOptionalPatch linux-kernel-test.patch
 
+ApplyOptionalPatch 0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch
+ApplyOptionalPatch 0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch
+ApplyOptionalPatch 0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch
+
 %{log_msg "End of patch applications"}
 # END OF PATCH APPLICATIONS
 

```

Gregregorovich, 2024-12-18:
> adding in from F41, I'm seeing similar errors except `include/drm/drm_displayid.h: No such file or directory` does not appear because the file seems to be there.
> 
> have we tried the headset with just patch 1?

I'm having issues with what appears to be the driver itself, but I have a BigScreen Beyond coming now which doesn't apparently need a driver - just this kernel patch and a bit more kernel patching to mark it as a non-display IIRC - so I'm not sure, but patch 1 on its own with the custom driver doesn't seem to do anything, but with patches 1, 2 and 3 I get driver errors in the log.

november781, 2024-12-24:
> FWIW I've written a patch file for F40 and edited the patches to support the different format for Fedora's kernels (haven't upgraded to F41 yet). I've been meaning to fork this and put my patches there, but I haven't done much testing for it with multiple kernel versions or added support for the BigScreen Beyond yet (which should just require a few additional kernel lines).
> 
> [0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch.txt](https://github.com/user-attachments/files/18188814/0001-drm-edid-Add-Vive-Cosmos-Vive-Pro-2-to-non-desktop-l.patch.txt)
> 
> [0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch.txt](https://github.com/user-attachments/files/18188820/0002-drm-edid-parse-DRM-VESA-dsc-bpp-target.patch.txt)
> 
> [0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch.txt](https://github.com/user-attachments/files/18188827/0003-drm-amd-use-fixed-dsc-bits-per-pixel-from-edid.patch.txt)
> 
> [kernel.spec.patch.txt](https://github.com/user-attachments/files/18188829/kernel.spec.patch.txt)
> 
> I guess I could just issue a patch file to the patch files...

Ill give this a try over the weekend to see if I can get it working.

november781, 2024-12-25:
> FWIW I've written a patch file for F40 and edited the patches to support the different format for Fedora's kernels (haven't upgraded to F41 yet). I've been meaning to fork this and put my patches [...]

Tested these a bit early. They do compile, load, install, and boot, but steamwebhelper is broken on the patched kernel.