CertainLach/VivePro2-Linux-Driver #46: SteamVR is upsidedown

KaziiTheAvali, 2024-11-25:
This  seems to be a reoccuring issue as https://www.reddit.com/r/virtualreality_linux/comments/1foahw1/help_needed_with_vive_pro_2_steamvr_vive_pro_2/ aswell has this isue. the display and everything in it is upsidedown. making it impossible to play any games and even nausating. for me i get this error even though proton is installed in enabled 
[image]
[image]
system things
[image]

iondc, 2024-12-07:
just made a github account to say this i did a temporary fix. go to bin/driver-proxy/src/driver/hmd.rs

go to around line 131 it should say
`let inverse = self.lens.matrix_needs_inversion()?;`

i commented it out and changed it to

`let inverse = false;`

didnt fix the lens distortion helper and things do look kinda fuzzy but they aint upside down anymore. now i just gotta figure out why it never renders the floor

CertainLach, 2024-12-07:
You don't see the floor, because you see the wrong side of textures, because you have inverted matrix

Upside-down texture and distortions would be only fixed by having working distortion helper. I'm not sure why it doesn't find proton for some people, but you can try setting `WINE=/path/to/wine %command%` as SteamVR command line.

iondc, 2024-12-10:
i can say for a fact proton is working. i wrote a very barebones c program that opens a win32 message box:

```
#include <windows.h>
#include <winuser.h>

int main(int argc, char **argv){
    MessageBox(NULL, "my original code said something funny", "proton checks out", MB_OK | MB_ICONINFORMATION);
    return 0;
}
```

compiled with `x86_64-w64-mingw32-gcc` and replaced lens-server.exe with it. no special launch command, steamvr seems like it wanna use proton hotfix

and of course it shows the message box just fine, and when i click ok i get the same failed to fill whole buffer dialog i always got. imma play around with the starting code and the pipes, i dont know rust though so i guess now a good time to learn. or maybe im missing dlls or something like that other issue

also tried `WINE=/usr/bin/wine %command%` same outcome

CertainLach, 2024-12-11:
I wonder if lens-server is not being ran as console app for you, I have seen that happen, but it didn't for me?

Does adding
#![windows_subsystem = "console"]
As the first line of https://github.com/CertainLach/VivePro2-Linux-Driver/blob/master/bin/lens-server/src/main.rs makes anything change for you?

lens-server is not supposed to be started as a dedicated window, it is a console app talking over stdio.

iondc, 2024-12-11:
i just tried that, it didnt change anything. but i saw some comments about missing dlls from #33 so i copied vkd3d-1.dll and vkd3d-shader-1.dll over from another prefix into drive_c/windows/system32 and it works now! everythings back upside down so imma undo my patch on driver-proxy/src/driver/hmd.rs, textures stopped being upside down too :)

CertainLach, 2024-12-11:
Huh, those aren't direct dependencies of LibLensDistortion, perhaps your wine prefix was broken for some reason?

iondc, 2024-12-11:
i think opencv_world needed those. was definitely something wrong with the prefix

KaziiTheAvali, 2024-12-12:
> i just tried that, it didnt change anything. but i saw some comments about missing dlls from #33 so i copied vkd3d-1.dll and vkd3d-shader-1.dll over from another prefix into drive_c/windows/system32 [...]

are you able to give me your system32 or profile. i have done this and it is still upsidedown for me

KaziiTheAvali, 2024-12-12:
never mind. i it ended up that there was more stuff missing in my system32 and just copying the whole system32 from a nother game (i think h3) worked

KaziiTheAvali, 2024-12-12:
ofc i still struggle with updating devices. hope that doesnt come back to bite me later on.

CertainLach, 2024-12-12:
I wonder why is yours prefixes so borked, and how can I fix this at the driver side, both proton and default nixos wine installation works just fine for me.

You can get debug logs for dll loading by setting
`WINEDEBUG=+loaddll,+warn`
environment variable.

kboshi, 2025-01-10:
here is how you fix the problem install proton and then it updates and then just uninstall proton and for some apparent reason it fixes the issue I have no clue why just did.

HexSegfaultCat, 2025-01-10:
I also managed to get it working. For some reason when you run `proton runinprefix start.exe ...` if prefix does not exist then it doesn't create all the DLLs in `system32`, but when you run `proton run` then it properly creates the prefix and then the Lens Server works.

Here's the steps I did to create the prefix:

```sh
# Proton prefix directory
STEAMVR_COMPAT_DIR="${HOME}/.steam/steam/steamapps/compatdata/250820"

# Required environment variables for running Proton outside of Steam
export STEAM_COMPAT_CLIENT_INSTALL_PATH="${HOME}/.steam/steam"
export STEAM_COMPAT_DATA_PATH="${STEAMVR_COMPAT_DIR}"

# Remove current Proton prefix and create an empty directory (required for next step)
rm -r "${STEAMVR_COMPAT_DIR}"
mkdir -p "${STEAMVR_COMPAT_DIR}"

# Use `run` command to create prefix in `STEAMVR_COMPAT_DIR`
"${STEAM_COMPAT_CLIENT_INSTALL_PATH}/steamapps/common/Proton - Experimental/proton" run
```

Then, in launch options for SteamVR set `PROTON_VERSION="- Experimental" %command%` if you have multiple Proton versions installed to make sure the Lens Server uses proper version (I belive it chooses 9.0 instead of Experimental by default):

And also one more important thing - make sure you do not have `PROTON_LOG=1` set in the launch options - the server communicates by sending data to stdout and this causes timeouts as it is redirected to a log file.

paraxial, 2025-07-20:
(First of all -- thanks @CertainLach for your work on this driver -- to even get this close is great.)

Jumping onto this issue, as I believe that I'm still experiencing it. (Let me know if it seems different enough to warrant a separate issue.)

To be difficult, I am running Ubuntu 24.04.2 LTS.

`uname -r`: 6.16.0-rc1-g60ba94338047-dirty

I've: 
- patched the kernel; 
- installed the drivers from Patreon (0.7.0); 
- modified HexSegfaultCat's script a little for my atypical steam install (STEAMVR_COMPAT_DIR="/mnt/games/SteamLibrary/steamapps/compatdata/250820");
- checked that I can run the lens-server manually:
```
❯ wine lens-server.exe
 INFO lens_server: hello from lens server
 INFO lens_server: dll path: "Z:\\home\\paraxial\\.steam\\debian-installation\\steamapps\\common\\SteamVR\\drivers\\lighthouse\\bin\\linux64\\lens-server\\LibLensDistortion.dll"
```
- Attempted pulling the system32 from my Beat Saber installation and putting it into the SteamVR compatdata

However, I'm still caught on:
[image]

Trying to add `WINE=/usr/bin/wine PROTON_VERSION="- Experimental" %command%` results in:
[image]
(which seems weird, since, of course, `which wine`  gives me `/usr/bin/wine`)

I'm at a loss for what to try next, and don't have a lot of experience with this kind of tinkering.  What information can I provide?

HexSegfaultCat, 2025-07-23:
@paraxial 

This looks like a different issue. Could you look at vrserver log file (`Steam/logs/vrserver.txt`)? Look for lines with prefix `lens_client`. I checked the code and it seems like the error "can't find wine in PATH" can only occur when Proton directory doesn't exist (then it falls back to using wine from `PATH`).

Now I don't know why it can't find wine's binaries, but I might have an idea why it can't find Proton. In your comment you mentioned 2 storage directories for Steam (home dir and `/mnt`) - you need to make sure Proton directories are in the same parent dir as SteamVR (meaning if your SteamVR is at `$HOME/Steam/steamapps/common/SteamVR` then Proton needs to be at `$HOME/Steam/steamapps/common/Proton 9.0`).

So remove `WINE` from launch options (when `WINE` is set it ignores Proton and only tries to use the specified wine binary), but keep `PROTON_VERSION`, make sure `Proton - Experimental` is next to `SteamVR` directory and try to run SteamVR.

CertainLach, 2026-09-30:
On master I have dropped proton and lens-server usage in favor of [champagne](https://delta.rocks/blog/lach-stuff/champagne), it should no longer fail randomly

closed 2026-09-30 by CertainLach