CertainLach/VivePro2-Linux-Driver #47: Error: Hid(HidApiError { message: "No HID devices with requested VID/PID found in the system.")

jdubrow, 2024-12-06:
> uname -a
Linux callipeia 6.12.0-rc6+ #13 SMP PREEMPT_DYNAMIC Fri Dec  6 16:58:18 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

> lsusb  |grep -i vive
Bus 001 Device 037: ID 0bb4:0306 HTC (High Tech Computer Corp.) Vive Hub Bluetooth 4.1 (Broadcom BCM920703)
Bus 001 Device 039: ID 0bb4:030b HTC (High Tech Computer Corp.) VIVE Pro Mutimedia Audio
Bus 001 Device 040: ID 0bb4:0342 HTC (High Tech Computer Corp.) VIVE Pro 2
Bus 002 Device 014: ID 0bb4:030c HTC (High Tech Computer Corp.) VIVE Pro Multimedia Camera

> sudo nix log /nix/store/.....-vivepro2-driver-proxy-0.1.0.drv

Failures:

---- test stdout ----
Error: Hid(HidApiError { message: "No HID devices with requested VID/PID found in the system." })

failures:
    test

test result: FAILED. 0 passed; 1 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.00s

error: test failed, to rerun pass `-p vive-hid --lib`

I tried a number of items (including adjusting hidtest to match the vendor/product id, but I am unable to get this to compile.

jdubrow, 2024-12-07:
76928.096336] hid-generic 0003:0BB4:0342.0023: hiddev0,hidraw2: USB HID v1.11 Device [HTC VIVE Pro 2] on usb-0000:00:14.0-6.1.4.1/input0
[76928.370436] hid-generic 0003:28DE:2101.0024: hiddev1,hidraw3: USB HID v1.11 Device [Valve Software Watchman Dongle] on usb-0000:00:14.0-6.1.4.3/input0
[76928.647271] hid-generic 0003:28DE:2101.0025: hiddev2,hidraw8: USB HID v1.11 Device [Valve Software Watchman Dongle] on usb-0000:00:14.0-6.1.4.4/input0
[76930.928922] hid-generic 0003:28DE:2300.0026: hiddev5,hidraw9: USB HID v1.11 Device [Valve LHR] on usb-0000:00:14.0-6.1.3/input0
[76930.929846] hid-generic 0003:28DE:2300.0027: hiddev6,hidraw10: USB HID v1.11 Device [Valve LHR] on usb-0000:00:14.0-6.1.3/input1
[76930.930489] hid-generic 0003:28DE:2300.0028: hiddev7,hidraw11: USB HID v1.11 Device [Valve LHR] on usb-0000:00:14.0-6.1.3/input2

is this likely why?  The usb #'s come up id's are coming up as 0000:00 in dmesg.  lsusb is fine though.  Weird?

iondc, 2024-12-10:
is your headset turned on? i struggled building it at first but the test passed after i plugged in my headset and turned it on

jdubrow, 2024-12-10:
Yes.  Though the light is only red.  I could only ever get it to turn green when i tried it in my Macbook Pro.   It wouldn't work on my linux machine or my windows machine.

CertainLach, 2024-12-10:
This test in vive-hid crate needs to be marked as ignored, it wants access to HID devices, but you don't have that in Nix sandbox.

jdubrow, 2024-12-10:
I tried bypassing it (setting doCheck = false and it went through, but then the SteamVR kept saying "Couldn't find driver for HID".   In the end I couldn't get the display working on a windows or Mac [mac worked the best], so I returned the system.  Sad, because I really wanted it to work.

closed 2024-12-10 by jdubrow

CertainLach, 2024-12-10:
It is possible you had missing udev rules for headset device permissions