CertainLach/VivePro2-Linux-Driver #48: Half-Life: Alyx crashes if i load chapter 2 in any way

iondc, 2024-12-13:
hey im real sorry cause i dont know for sure if the drivers causing this or if its something else wrong with my machine. but i never successfully loaded chapter 2 in half life alyx in any way. it either crashes steamvr (and i get the oh no! message) or my whole pc freezes up. i just wanna ask if anybody got any further in half life alyx while playing it with this driver. i dont have any other headsets and i dont have a copy of windows to compare it to.

i tried both native and proton. i gotta leave for work in 8 minutes so i cant do much more testing rn but later imma try the novr mod. havent found crash logs yet so i dont got any to post. but as i keep trying things imma keep yall posted, and imma close this issue if i find out its nothing wrong with the driver.

CertainLach, 2024-12-13:
Are you using latest linux kernel?
Various whole pc freezing problems are usually amdgpu bugs, and there were many fixed in the past.
HL Alyx works fine for me.

iondc, 2024-12-14:
im on 6.8.0-49 generic, default for ubuntu 24.04 lts. downloading the packaged 6.11 kernel image rn, if that dont work imma build 6.12 from source

iondc, 2024-12-17:
imma close this, definitely not a problem with your driver. runs better on the default 6.8, i hadnt got around to running my built 6.12.5 yet. i also tried the amdvlk driver, and with one launch i got alot of lag but i loaded a little further into chapter 2 and saw a map kinda thing inside the train until just the game crashed. every launch after that the lag smoothed out but it went back to crashing my whole pc.

closed 2024-12-17 by iondc

iondc, 2024-12-23:
imma take a look at issue https://gitlab.freedesktop.org/drm/amd/-/issues/1974 especially [this comment](https://gitlab.freedesktop.org/drm/amd/-/issues/1974#note_2589801)