CertainLach/VivePro2-Linux-Driver #54: SteamVR 2.9+: every second frame black

Bixilon, 2025-05-06:
As the title sais, it is not happening in the default steamvr overlay where you see the sky. Only in games.

CertainLach, 2025-05-26:
On which display mode does this happen? DSC-only 4/5 with kernel patch?

Otherwise, it shouldn't be caused by the driver.

CertainLach, 2025-05-26:
If that is happening in modes 0-3 then it is probably SteamVR/Intel GPU issue. SteamVR 2.10.4 works fine for me

Bixilon, 2025-05-26:
Did you launch a game? The overlay menu with the sky works fine. It first happens when launching any game. On what mode are you?

It is also happening with the latest beta of SteamVR (2.11.1)

CertainLach, 2025-05-26:
I am using mode 5, but I have also checked that on mode 3

I see no problems in HL Alyx/Beat Saber