CertainLach/VivePro2-Linux-Driver #57: Headset IPD

Beyley, 2025-09-18:
I've been using this repository as a loose guide to implementing a native Vive Pro 2 driver for Monado (to be able to use with both libsurvive and the official Valve tracking blob), one thing which has caught me up is the IPD reading. It reports 15mm over the Lighthouse protocol, but over it's own protocol, reports 4 integers. An [earlier version of this repo](https://github.com/CertainLach/VivePro2-Linux-Driver/blob/35f365601ed86c2013c5c2d6cce7ecf84d9aec58/crates/vive-hid/src/lib.rs#L228) did have `mfg-r-ipdadc` read out, but never properly parsed it out into a millimeter or meter reading. I've found 3 more values, `mfg-r-ipdmin`, `mfg-r-ipdmax`, and `mfg-r-ipdlap`, which seem roughly related to the IPD slider, but I've been unable to figure out the algorithm to convert this into a real world measurement.

Have you happened to make progress on this? Or are you also stuck on the same issue.

Here are some values my code is reading out from the HMD:
```
DEBUG [vp2_hid_open] IPD: 309 (min: 499, max: 3581, lap: 3922)
```

CertainLach, 2025-09-18:
> ftr in case it helps, I reverse engineered the display distortion, at least the strengthen high order model, which I have working 1:1 with LibLensDistortion, you can find it here

Is your headset using this model? LibLensDistorion has much scarier code paths, and one used for me is opencv-based

CertainLach, 2025-09-18:
For basic code path I got my wine replacement to work, but I don't have enough patience to make it run opencv, thus wine.
https://github.com/CertainLach/chimera

Beyley, 2025-09-18:
> > ftr in case it helps, I reverse engineered the display distortion, at least the strengthen high order model, which I have working 1:1 with LibLensDistortion, you can find it here
> 
> Is your headset using this model? LibLensDistorion has much scarier code paths, and one used for me is opencv-based

They actually all use OpenCV, most just for matrix multiplication though. Which distortion algorithm does your HMD use? Me and all the testers that have used my driver have had `strengthen_high_order`, so it's been no issue.

CertainLach, 2025-10-16:
Just noticed that I did reverse it at some point for my unfinished configurator.

Adc is converted to not-millimeters as `not_millimeters = (70 - adc / 400)`

Not-millimeters because the final value then needs to be processed somehow to get the actual millimeters, but I didn't bother with that, because not_millimeters are roughly the same range (Physical IPD - 57-72, not_millimeters - 60-70), and also I don't see any difference with software IPD adjucement, is it even used by SteamVR itself?

Beyley, 2025-10-16:
> I don't see any difference with software IPD adjucement, is it even used by SteamVR itself?

It's used to adjust the distance between the virtual cameras. If this doesn't match the user's physical IOD (distance between centers of eyeballs) then there will be a slight mismatch in the "scale" of the world, without effecting the movement speed, which can cause disorientation or nausea for some people.

It being off by ~2mm isn't awful, and will strongly delay the issue, but it's still better for it to be exact to the user's actual IOD.

CertainLach, 2025-10-16:
I don't understand how it should affect the virtual cameras? They have fixed ipd

CertainLach, 2025-10-16:
Oh, nvm, you meant the inside game cameras

CertainLach, 2025-10-16:
Ok, makes sense then. I have expected some difference in how the optics should work, but now I see the issue.

Beyley, 2025-10-16:
> I have expected some difference in how the optics should work.

VR optics are a bit special, where the position of the lenses relative to the user's eyes doesn't actually effect anything but clarity, distortion correction, and FOV. It doesn't change the user's actual view "feel" at all, given it makes everything parallel.

You can see this effect by putting a static image on the HMD displays and moving the HMD laterally while keeping your face stationary.

CertainLach, 2025-10-16:
I don't remember that much, I just have huge pile of broken code and wireshark usb pcaps stored on my nas.

CertainLach, 2026-10-01:
@Beyley I have reversed how the headset handles IPD changes, turns out it reports such things on ViveHid device. Latest version of the driver handles IPD correctly.

closed 2026-10-01 by CertainLach